Sloeber/arduino-eclipse-plugin #208: Arduino Nanos with CH340G chipset for USB/Serial are not listed on Mac OS-X Yosemite

kigster, 2015-01-02:
I briefly looked at how the plugin gets list of available Serial ports, and I *think* it's going through RXTX library, which then goes through native ObjectiveC API, in the file SerialImp.c. The API seems to be using a kernel function:

```c
  kernResult = IOServiceGetMatchingServices(masterPort, classesToMatch, serialIterator);
```

For whatever reason this does not include the ports created by Nano with the above mentioned chipset.  This chipset does not currently have a Yosemite specific / signed driver, so this may be why the OS-X Api is not properly registering it as a serial port.  But it is available under /dev:

```bash
crw-rw-rw-   1 root  wheel      17,   4 Jan  1 16:40 tty.wchusbserial144140
crw-rw-rw-   1 root  wheel      17,   5 Jan  1 16:40 cu.wchusbserial144140
```
Is there a higher level call inside the plugin itself that can be modified to include these ports in the list of Serial ports?

jantje, 2015-01-02:
Some comments
1) The Arduino eclipse plugin lets you enter anything for serial port in the project properties or the create sketch wizard. There is no test on existence nor on format. During testing I mostly type "fff" or "ggg" as serial port. 
Apart from resetting the board via the serial port if needed the plugin simply doesn't use the port. If the resetting fails the plugin still tries to upload.
2) The serial monitor is a different story. There you can only select the found ports.
3) The latest version of the plugin no longer uses rxtx. This port has been done by  amorellgarcia and I haven't had a need to go and look into the code in detail. Thanks for the good work  amorellgarcia.
4) The rxtx library has "internal logic to filter ports". On linux there is a work around http://eclipse.baeyens.it/Arduino%20eclipse%20plugin%20FAQ.html#LinuxNoSerialPorts but I'm not sure this works on OS-X

My advice: upgrade to the latest nightly build and see whether you still have the issue.

kigster, 2015-01-02:
Actually I am using nightly build of 2.4 :) If this serial port filtering logic exists in java, I would be happy to add support for this chipset myself and send a PR.

kigster, 2015-03-02:
Any update on this?  Or pointers where in the source that logic lives, so that I can try to patch it?

jantje, 2015-03-02:
Have you tried the latest version?

kigster, 2015-03-02:
Not the latest, but at the time I logged this ticket I was on the HEAD.

jantje, 2015-03-02:
Och; I was not aware so I thought you'd check out the latest version.
The code you are looking for is 
in project it.baeyens.arduino.common 
in package it.baeyens.arduino.arduino 
in class Serial
the method list

Good luck

jantje, 2015-04-07:
is there an update?

jantje, 2015-05-05:
as I do not have the hardware (boards and osx) I'll mark this as a known limitation.
If you ever get to fix this please post

jantje, 2015-12-07:
If anyone can confirm the fix is working fine that would be great :-)

kigster, 2015-12-07:
I can, will try today.

jantje, 2015-12-07:
Txs
If you want to use the "update" functionality, pleas wait untill tomorrow. This change will only be caught up by the nightly build.

jantje, 2015-12-17:
@kigster did you find the time needed to test?

kigster, 2015-12-21:
Hey – I tested this using the nightly build. 

On the positive note – they do show up in the list: both in the project's configuration, as well as the Serial console as shown below.

On the downside – it takes exactly 15 seconds to bring up each dialog. Whatever logic is scanning for ports is taking this long, which is definitely not ideal.

Hope this helps!

[image]

[image]

alexandrezia, 2015-12-21:
I'm using the nightly build as well,
And it's taking less then a second to show up all my serial ports.
Perhaps the regular expression:
```
portNames = SerialPortList.getPortNames("/dev/", Pattern.compile("tty.*"));
```
Should be narrowed a bit to exclude the device that's causing the delay.

jantje, 2015-12-21:
I see no delay here as well.
@alexandrezia 
Any clue on a better regular expression?

kigster, 2015-12-21:
Do you have the WSCH driver loaded, when it takes less than a second?

kigster, 2015-12-21:
Pattern can be optimized as follows:
```java
...compile("^cu\..+");
```

alexandrezia, 2015-12-21:
Yes, always loaded

jantje, 2015-12-21:
Please advice on the actions I should take.

alexandrezia, 2015-12-21:
I did some tests and found out that bluetooth serial connections could slow down serial port detection because these connections exists but are not actually active, takes some time.
So this is the most optimal regular expression I could found:
```
Pattern.compile("^cu\\..*(serial|usb).*"));
```
As it excludes bluetooth connections, it turned out to be real fast.

alexandrezia, 2015-12-21:
Pull request:  https://github.com/jantje/arduino-eclipse-plugin/pull/335

jantje, 2015-12-21:
merged :-)
txs.

kigster, 2015-12-24:
Awesome, thanks! – the delay is now gone.

jantje, 2015-12-24:
good than we can close this.
Thanks to all of you.
Jantje

closed 2015-12-24 by jantje

TrippyLighting, 2016-01-03:
@kigster would you be willing to share that dark theme in that can be seen in the screenshot other up in this thread... and also possibly how to install it ?
Sorry for hijacking this thread!

wimjongman, 2016-01-03:
preferences: General/Appearance - Theme: Dark

TrippyLighting, 2016-01-04:
Doh ;-) Thanks!