Sloeber/arduino-eclipse-plugin #209: uploading to a Micro gives error on com port COM

jantje, 2015-01-10:
from issue https://github.com/arduino/Arduino/issues/2422#issuecomment-69455515

I'm not sure if this is related to the new toolchain or not, but while everything works fine with the Uno and the Pro Mini, I'm having issues uploading to a Micro which seems not resetting the COM.

Launching R:/Tools/ArduinoIDE/hardware/tools/avr/bin/avrdude -CR:/Tools/ArduinoIDE/hardware/tools/avr/etc/avrdude.conf -patmega32u4 -cavr109 -P COM4 -b57600 -D -Uflash:w:R:\Arduino\workspace\temp/Release/temp.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM4": Impossibile trovare il file specificato.
avrdude: ser_send(): write error: sorry no info avail

The issue doesn't occur when using the Arduino IDE 1.5.8 and I'm currently unable to verify if it's unrelated to the new toolchain, but somebody else might do a quick test.

I'm trying to upload a simple blink firmware.

rlogiacco, 2015-01-10:
I'm not anymore in front of my PC, but I believe there might be another change that could have caused this: I did run an update of the plugin which reported a new version was available.
Could it be related?

jantje, 2015-01-10:
IMHO Nothing changed in that area recently. Note that updates to the website will also result in a new plugin version when using the nightly build. And I did a lot of those lately. ;-)

rlogiacco, 2015-01-12:
Ok, then why do I get that message? :-)
```
Starting upload
using arduino loader

Launching R:/Tools/ArduinoIDE/hardware/tools/avr/bin/avrdude -CR:/Tools/ArduinoIDE/hardware/tools/avr/etc/avrdude.conf -patmega32u4 -cavr109 -P COM4 -b57600 -D -Uflash:w:R:\Arduino\workspace\temp/Release/temp.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM4": Impossibile trovare il file specificato.

avrdude: ser_send(): write error: sorry no info avail
R:/Tools/ArduinoIDE/hardware/tools/avr/bin/avrdude finished
upload done
```

As you can see the provided port parameter is `COM4`, but you say the error message reports a wrong port...

jantje, 2015-01-12:
the command states -P COM4 but the error message states \\.\COM4.
So the com port on the error message is clearly wrong.
Can you put on verbose logging in arduino IDE and see what the upload command looks like there?

rlogiacco, 2015-01-12:
The upload command looks quite different on the Arduino IDE as it points to COM5...
Please forgive the Italian translations and the quite long paste, but I thought it could be helpful :-D

```
Forzo il reset della porta tramite una apertura/chiusura a 1200bpsCOM4
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM4, } => {}
PORTS {COM3, COM4, } / {COM3, COM5, } => {COM5, }
Found upload port: COM5
R:\Tools\ArduinoIDE/hardware/tools/avr/bin/avrdude -CR:\Tools\ArduinoIDE/hardware/tools/avr/etc/avrdude.conf -v -v -v -v -patmega32u4 -cavr109 -PCOM5 -b57600 -D -Uflash:w:C:\Users\RLOGIA~1\AppData\Local\Temp\build151623416228946608.tmp/sketch_jan12a.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Jan  2 2015 at 20:40:00
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "R:\Tools\ArduinoIDE/hardware/tools/avr/etc/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader
```

It seems Eclipse IDE doesn't try to reset the board before attempting the upload. Because I'm using Arduino IDE 1.5.8 with the new toolchain I tried to compare the `boards.txt` files which do differ indeed, but not on the Arduino Leonardo and Micro sections, which is where the error occurs.

I double checked with an original Uno R3, a Uno R3 clone and a Mega 2560 clone, they all worked nicely, but they do not need a reset procedure.

rlogiacco, 2015-01-12:
After a little more investigation I believe this might have something to do with the new toolchain :)

When I select Uno or Mega boards the Eclipse project explorer panel displays
[image]

But when I select a Micro as board I get
[image]

Doesn't this remind you something?

jantje, 2015-01-12:
What I see is that the micro has these settings
>micro.upload.use_1200bps_touch=true
micro.upload.wait_for_upload_port=true

This means that the micro should be reset and that a different port should be used.
As you can see from the verbose arduino log you see COM4 is reset and becomes COM5. then upload to COM5 is done.
The Plugin tries to upload to com4. Which is wrong.
So to me the upload is clearly a plugin issue.
Please look at the error log (that is a view in eclipse) whether you get any hints on what the plugin is trying to do and what fails.

The includes not being there indeed rings a bell but..... although don't jump to early conclusions. There are several "normal" reasons why the includes would not be there. The includes are generated by the indexer. If the indexer didn't finish the includes won't be there. Switching board on a project must relaunch the indexer, so you may have checked to early.
Also the indexer issue does not affect the build nor the upload.

rlogiacco, 2015-01-12:
The indexer doesn't seem to pick the libraries, I gave it plenty of time and clean/rebuild/reindex didn't help.
Nonetheless I checked the error log and what I got was the following:
```
null
it.baeyens.core
Warning
Tue Jan 13 00:08:58 CET 2015
Leonardo upload port is not disappearing after reset and 202 checks
```

Forcing reindexing produces the following in the same error log Eclipse view:
```
null
org.eclipse.cdt.core
Info
Tue Jan 13 00:11:54 CET 2015
Indexed 'temp' (27 sources, 29 headers) in 0,326 sec: 1.621 declarations; 2.386 references; 62 unresolved inclusions; 27 syntax errors; 613 unresolved names (13%)
```

Along with that I get multiple warnings and compilation errors in the editor but compilation in the console view completes nicely with just some warnings.

While this is not critical to me (I have other boards for development) I believe this might be something we should fix.

If I just switch back the Arduino Eclipse configuration to ArduinoIDE 1.5.5 all the editor errors disappear and the libraries get linked correctly (the indexer successfully finds the header files), but the upload still fails with the same error.

My impression is we are looking at two separate issues.

jantje, 2015-01-13:
>My impression is we are looking at two separate issues.  

I agree on this one.
>Leonardo upload port is not disappearing after reset and 202 checks

This means the reset is not successful and as such com4 is used instead of com5 during upload.
The "reset and search for new port" has shown to be "less stable". Some people have contributed changes but the number of boards and OS'es is huge.
I tried the micro on my system (linux redhat). Just like the leonardo I have a system configuration problem which makes that the com port is not available :-(
But the indexer and compilation work fine.
The next coming months I won't have access to a windows system. I'll add some debugging info to the upload so we can have a better understanding of the upload issue.

for the indexer issue I propose to do the same as with the similar issue we had. That is turn the "allocate console view on" in the project properties->C/C++ general->pre processor include paths, macro's etc->providers->Arduino compiler settings

And run that command you see at the top of the console on the command line in the arduino folder

Send both outputs (console and command line) to me.

jantje, 2015-01-13:
I have made a fix that may fix your upload issue but it is actually targeted at being more verbose about the upload so we have a better understanding what goes wrong.
Basically the plugin will tell you what it is trying to do and what fails in the console window.

rlogiacco, 2015-01-13:
I've updated the Arduino Eclipse plugin to the latest nightly build `2.4.0.201501130228` along with the core Eclipse CDT and it seems the indexer issue has gone away: includes are correctly displayed while editor errors and warnings are gone.

On the other hand I still cannot upload to my micro and I'm getting the same output in the console view:
```
Starting upload
using arduino loader

Launching R:/Tools/ArduinoIDE.old/hardware/tools/avr/bin/avrdude -CR:/Tools/ArduinoIDE.old/hardware/tools/avr/etc/avrdude.conf -patmega32u4 -cavr109 -P COM4 -b57600 -D -Uflash:w:R:\Arduino\workspace\temp/Release/temp.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM4": Impossibile trovare il file specificato.

avrdude: ser_send(): write error: sorry no info avail
R:/Tools/ArduinoIDE.old/hardware/tools/avr/bin/avrdude finished
upload done
```

The error log contains only two items related to the plugin:
```eclipse.buildId=4.4.1.M20140925-0400
java.version=1.8.0_25
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=it_IT
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product

org.eclipse.cdt.core
Error
Tue Jan 13 22:02:54 CET 2015
Exception trying register listener it.baeyens.arduino.languageSettingsProvider

java.lang.NullPointerException
	at it.baeyens.arduino.toolchain.ArduinoLanguageProvider.getCompilerCommand(ArduinoLanguageProvider.java:144)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.resolveCommand(AbstractBuiltinSpecsDetector.java:308)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.calculateEnvHash(AbstractBuiltinSpecsDetector.java:422)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.validateEnvironment(AbstractBuiltinSpecsDetector.java:466)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.execute(AbstractBuiltinSpecsDetector.java:481)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.registerListener(AbstractBuiltinSpecsDetector.java:375)
	at org.eclipse.cdt.internal.core.language.settings.providers.LanguageSettingsProvidersSerializer.reRegisterListeners(LanguageSettingsProvidersSerializer.java:1375)
	at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.setCurrentDescription(XmlProjectDescriptionStorage.java:386)
	at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.setLoaddedDescriptionOnLoad(XmlProjectDescriptionStorage.java:357)
	at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.getProjectDescription(XmlProjectDescriptionStorage.java:256)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescriptionInternal(CProjectDescriptionManager.java:437)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:419)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:413)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:406)
	at org.eclipse.cdt.core.resources.RefreshScopeManager.loadSettings(RefreshScopeManager.java:480)
	at org.eclipse.cdt.core.resources.RefreshScopeManager.loadSettings(RefreshScopeManager.java:461)
	at org.eclipse.cdt.core.resources.RefreshScopeManager.<init>(RefreshScopeManager.java:108)
	at org.eclipse.cdt.core.resources.RefreshScopeManager.getInstance(RefreshScopeManager.java:90)
	at org.eclipse.cdt.internal.core.BuildRunnerHelper.refreshProject(BuildRunnerHelper.java:360)
	at org.eclipse.cdt.managedbuilder.core.ExternalBuildRunner.invokeExternalBuild(ExternalBuildRunner.java:133)
	at org.eclipse.cdt.managedbuilder.core.ExternalBuildRunner.invokeBuild(ExternalBuildRunner.java:72)
	at org.eclipse.cdt.managedbuilder.internal.core.CommonBuilder.build(CommonBuilder.java:753)
	at org.eclipse.cdt.managedbuilder.internal.core.CommonBuilder.build(CommonBuilder.java:510)
	at org.eclipse.cdt.managedbuilder.internal.core.CommonBuilder.build(CommonBuilder.java:459)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:734)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:206)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:246)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:299)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:302)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:263)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:401)
	at org.eclipse.core.internal.resources.Project$1.run(Project.java:618)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
	at org.eclipse.core.internal.resources.Project.internalBuild(Project.java:597)
	at org.eclipse.core.internal.resources.Project.build(Project.java:114)
	at it.baeyens.arduino.actions.JobHandler.run(BuildHandler.java:44)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)
```
[... cut ...]
eclipse.buildId=4.4.1.M20140925-0400
java.version=1.8.0_25
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=it_IT
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product

it.baeyens.core
Warning
Tue Jan 13 22:04:14 CET 2015
Leonardo upload port is not disappearing after reset and 202 checks
```

I wonder if I've missed something or if the update doesn't contain your changes yet.

jantje, 2015-01-14:
The update doesn't contain the changes yet. 
The build is done at around 3:00 CET and takes time.
So when upgrading tomorrow you should get a newer version that is more verbose on the upload.

rlogiacco, 2015-01-15:
Update installed, no joy. As you can see I tried a different USB port just to exclude other possible source issues: COM7 is on the back of my desktop.

```
Starting upload
using arduino loader
Starting reset using 1200bps touch process
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
PORTS { COM7,} / { COM7,} => {}
Comport is not behaving as expected
Using comport COM7 from now onwards
Ending reset using 1200bps touch process

Launching R:/Tools/ArduinoIDE.old/hardware/tools/avr/bin/avrdude -CR:/Tools/ArduinoIDE.old/hardware/tools/avr/etc/avrdude.conf -patmega32u4 -cavr109 -P COM7 -b57600 -D -Uflash:w:R:\Arduino\workspace\temp/Release/temp.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM7": Impossibile trovare il file specificato.

avrdude: ser_send(): write error: sorry no info avail
R:/Tools/ArduinoIDE.old/hardware/tools/avr/bin/avrdude finished
upload done
```

At first glance it seems the plugin fails quite sooner than the ide. If you want me to do a small change in the source code and rebuild, keep in mind I should be able to it without needed a nightly build.

jantje, 2015-01-15:
can you provide the output with com4?

It is very easy to do local development. See the readme.md here https://github.com/jantje/arduino-eclipse-plugin

jantje, 2015-01-15:
From what I understand the touch to the serial port to reset the arduino is not working properly.
I have been comparing the code of the Arduino IDE with the plugin and the only thing I found so far is that in the plugin there is a delay of 100ms which is not there in the Arduino IDE
https://github.com/arduino/Arduino/blob/8e0a311e871a3feb505f18771a6fd58abf5048cd/app/src/processing/app/Serial.java look for touchPort
https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/communication/ArduinoSerial.java look for reset_Arduino_by_baud_rate

It may be a good test to remove the Thread.sleep(openTime);

If you can't do a local test give me a shout and I'll do it and you can see tomorrow.

rlogiacco, 2015-01-15:
You pointed me to the right direction, but apparently the issue was related to a too short delay between the checks. Please note the Arduino IDE waits up to 10 seconds for the COM port to show up, sleeping for 250ms between each try and adding an additional 300ms wait after the 1200 baud reset request on non MacOS platforms: this latter delay seems to be needed to avoid a subsequent reset which would invalidate the 1200 baud reset.

It takes 3 seconds for my original Micro to show up the upload COM port on my very performing desktop PC: I believe 2 seconds were not enough on many machines and even 5 seconds (the current setup) might not be enough on slower machines.

Can we put those information into a preference option or, in case that would be too much hassle, provide a way to override the defaults by using something like eclipse environment variables?

I'm not really proficient with Eclipse, but I might be able to do something myself if you can point me in the right direction :smile:

rlogiacco, 2015-01-15:
Last note: I played a bit with the parameters once I had it working and it takes about 2800ms for my Micro to reset and show up. We can revert back to 100ms sleep (both per check and after 1200 baud reset request) to save a little less than half a second in the upload process, but we need to boost the number of retries up from the current 20 to about 50 in order to accommodate slower computers.

jantje, 2015-01-16:
> Please note the Arduino IDE waits up to 10 seconds for the COM port to show up, sleeping for 250ms between each try and adding an additional 300ms wait after the 1200 baud reset request on non MacOS [...]

If that is what arduino does now; That should be the same for the plugin.
So todo
1) Set delay=250
2) set trials=40
3) add 300ms wait after the 1200 baud reset request on non MacOS platforms

I'll do it this evening

rlogiacco, 2015-01-16:
Ok. delay it's already at 250 and we have the 300ms wait, but I didn't know how to check the platform. Number of trials needs to be bumped up to 40.

jantje, 2015-01-16:
Untested .... scary ....

jantje, 2015-04-07:
works fine

closed 2015-04-07 by jantje

MatthewKleinsmith, 2016-10-27:
TL;DR: How do I modify the duration the Arduino IDE will wait for the Arduino Micro to reconnect?

I'm trying to upload the Blink sketch to my Arduino Micro from Ubuntu 14.04. I've tried with the Arduino IDE 1.0.5, 1.6.12, and 1.06. My problem seems to be similar to what I think was discussed here: The Arduino IDE gives up too soon when waiting for the Arduino Micro to reconnect to the USB port. Here's my upload log:

> Binary sketch size: 4,696 bytes (of a 28,672 byte maximum)
> **Forcing reset using 1200bps open/close on port /dev/ttyACM0**
> PORTS {**/dev/ttyACM0**, /dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> PORTS {/dev/ttyS1, /dev/ttyS0, } / {/dev/ttyS1, /dev/ttyS0, } => {}
> processing.app.debug.RunnerException: **Couldn’t find a Leonardo on the selected port.** Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.
> 	at processing.app.debug.AvrdudeUploader.uploadViaBootloader(AvrdudeUploader.java:153)
> 	at processing.app.debug.AvrdudeUploader.uploadUsingPreferences(AvrdudeUploader.java:67)
> 	at processing.app.Sketch.upload(Sketch.java:1666)
> 	at processing.app.Sketch.exportApplet(Sketch.java:1622)
> 	at processing.app.Sketch.exportApplet(Sketch.java:1594)
> 	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2382)
> 	at java.lang.Thread.run(Thread.java:745)

How do I modify the duration the Arduino IDE will wait for the Arduino Micro to reconnect?