Sloeber/arduino-eclipse-plugin #213: search functionality of the new website does not work with firefox in linux

jantje, 2015-01-17:
I can type text but when I press enter or the looking glass nothing seems to happen.
Though I do feel like my system hung for a while the first time I clicked. That I'm not sure.

rlogiacco, 2015-01-18:
No, I never succeed in having it working. The plan was to link Google Site Search, but I don't like the output format. I can put something together as a demo

jantje, 2015-01-18:
many sites use google as search engine so there must be a documented way to make it look good or people got used to it.
From the top of my head it is the second.

rlogiacco, 2015-02-13:
This hould temporarily fix this until I, or somebody else,finds a better solution.
So far the only better alternative I found was to use the Google APIs, but we are limited to 100 searches per day for free: I believe that will not be sufficient for the site needs.

jantje, 2015-02-14:
I agree that 100 searches will probably not be enough and with current rate of money coming in; paying google is not an option.

rlogiacco, 2015-02-14:
Exactly my same reasoning :smiley: 
If I find a way to nicely integrate the embedded Google site search widget I will submit an update, for now I believe the current implementation will be a decent workaround. main issue with the applied solution is the results are displayed in a standard Google search page, but they are still refering to our site only.

If anybody has a solution to integrate a Bootstrap 3 search input and Google search widget please feel free to step in!

closed 2015-02-20 by jantje