Sloeber/arduino-eclipse-plugin #214: images of contributors do not appear when in eclipse

jantje, 2015-01-17:
When the remind page is loaded in eclipse the html page http://eclipse.baeyens.it/remind3_0.html is loaded.
When I go to this page with my browser (firefox) the images of the contributors are shown fine.
However in eclipse the images of the contributors are not shown (a replacing icon is shown). The up arrow image is however shown correctly.

If anyone has a clue on this I'd be happy to hear about it.
If anyone does not have this problem please tell me.
Ik anyone has the same issue please tell me as well.

rlogiacco, 2015-01-18:
Last time the window opened up in my Eclipse I had 3 faces showing up and the images where there.

Should I style this page along with the others? Wouldn't it be better if we include this page within the plugin in the form of a an SWT window?

jantje, 2015-01-18:
The idea is that I can change the html to represent the latest data. Therefore each release has a different html address.
For instance People using 2.2 see they should upgrade.
If donations remain low as it is (0 Euro in December and January) I will consider adding advertisements to the page. 
I'm not sure how a SWT window makes it different from the current browser page.
Making the pages compliant with the current look is good :-)

closed 2015-04-07 by jantje