Sloeber/arduino-eclipse-plugin #215: the downoad of stable win link does not work anymore

marcelrv, 2015-01-21:
http://www.baeyens.it/eclipse/stable-win.html

the stable win link does not work anymore

 ```
Not Found

The requested URL /download/product/V2.3_win32.2014-07-12_02-06-35.tar.gz was not found on this server.```

jantje, 2015-01-22:
Indeed
But please use http://eclipse.baeyens.it/stable-win.html then it works fine.
Can you tell me where you found the http://www.baeyens.it/eclipse/ reference?
Best regards
Jantje

jantje, 2015-01-22:
Should be fixed now.

closed 2015-01-22 by jantje

TrippyLighting, 2015-01-22:
I think I'll need to update the links and references in the instructions on my blog.

marcelrv, 2015-01-22:
Hi
Arrived via google:
https://www.google.com/search?q=arduino+eclipse+&ie=utf-8&oe=utf-8
The 3rd link (for me) pointed to www.baeyens.it/eclipse/

Using the Install->Windows from the top menu bar.

marcelrv, 2015-01-22:
Indeed fixed now

jantje, 2015-01-23:
txs for the link. Unfortunately I don't know how to change google output ;-)

rlogiacco, 2015-01-27:
It should self fix in a few days when google crawler will try to re-index the site. References from other sites though will have to be updated manually.