Sloeber/arduino-eclipse-plugin #216: Mac OS can not set plugin

leikoilja, 2015-01-22:
Hello. I really love that plugin and would like so it work
I am using your arduino plugin on my mac book pro OS X 10.10.1; 
Eclipse Luna IDE for C/C++ Developers(4.4.1) for Mac OS X 64 Bit; 
Arduino ide 1.5.7; 
Late night plugin 2.4.0

But when i am trying to upload schetch on the arduino it shows me some error :/ (please see attached picture) 
[image]

jantje, 2015-01-23:
Have you verified before the upload?
The plugin does not do a verify before upload.

leikoilja, 2015-01-23:
How can I verify?

TrippyLighting, 2015-01-23:
Try the "Verify" button. This is the green button to the left of the "upload" button ;-)

leikoilja, 2015-01-23:
Ou. Sorry,  I just misunderstood the question then.  Yes,  I have tried pressinf verify,  nothing changes.

jantje, 2015-01-23:
and the compilation was successfull?

leikoilja, 2015-01-23:
oups, no it wasn't. It shows me mistake that file type is not found. But that's strange, cause i have set file type as it was written in the guide.
[image]
[image]

jantje, 2015-01-23:
The plugin is not correctly configured. It has to be *.ino and *.pde
See http://eclipse.baeyens.it/how_to.shtml#/e on how to configure correctly.
And visit http://eclipse.baeyens.it/donate.shtml to show your appreciation for the support.

leikoilja, 2015-01-23:
I will show my appreciation as soon as i could make it work :/
okey, that was stupid error, i have solved it, but now it still shows some strange things
[image]
and in the configuration plugin does not accept my Arduino.app
[image]

jantje, 2015-01-23:
I guess you have installed older versions of the arduino IDE in /Applications/Arduino.app
I say so because the plugin uses .../content/resources/java/... and this is pre Arduino IDE 1.5.7.
In Arduino IDE 1.5.7 (and later) it should be .../Contents/Resources/Java/... as you  can read here https://github.com/jantje/arduino-eclipse-plugin/pull/180
I advice to completely remove the arduino app and reinstall it.

leikoilja, 2015-01-23:
Installed 1.5.8, still the same error. 
Tried writing "/Applications/Arduino.app/Contents/Java" instead of /Applications/Arduino.app, but still it shows me the error "Value must be an existing file"

jantje, 2015-01-23:
V2.2 of the plugin won't work.
Have you reselected the board in the board properties? That is vital when changing the underlying arduino IDE

leikoilja, 2015-01-24:
Double checked, the board selected correctly. 
[image]
I am getting that error now while compiling :/
[image]
and i still have that written in eclipce preferences 
[image]

jantje, 2015-01-24:
as you can see the path is now correct set to  .../Contents/java/ instead of  .../Contents/Resources/Java/
so you made some progress.
Now all this is fine try to make a new sketch using the create sketch wizard.
I'm pretty sure this should work.

leikoilja, 2015-01-24:
Have created a new sketch, even changed the microcontoller to arduino uno. Still got the same error, while compilation. 
Googled that error, found kind a solution to that:
[image]
Have done how they advise... Got new errors :D :/
[image]

jantje, 2015-01-24:
the problem you have now is that you need to include Arduino.h in your ino files if you have no other libraries in use.

leikoilja, 2015-01-24:
Can not believe THAT... successfully compiled :O :+1: 
[image]
p.s. i think if you could add exact version of eclipse & arduino ide, which are working, to your guide, it might help a lot of people :) 
Last question, if it possible to make commands such as "digitalWrite" to be highlighted? :) 

Oh.. tried to upload on my arduino uno got
[image]

TrippyLighting, 2015-01-24:
Not sure this is still the case but in the past spaces in directory names were best avoided as Eclipse and/or the plugin encountered problems with spaces.
There is a space in ...../Eclipse workspace/.....
See if you can remove that and try again.

leikoilja, 2015-01-24:
Yes yes yes yes yes yes, FINALLY!!!!
Thank you very much!!! Now it even compiled in other way: 
[image]

Before closing topic, i will ask once more, is it possible to make arduino functon highlighted? :) 
[image]

jantje, 2015-01-25:
>is it possible to make arduino functon highlighted?
yes an no.
It is possible to highlight a function but that would be all functions.
It is not possible to highlight the arduino functions and not the other functions.

leikoilja, 2015-01-25:
how to make them highlighted? :)

jantje, 2015-01-25:
preferences->C/C++->code style->Formatter

jantje, 2015-01-26:
@leikoilja: Is everything working fine now?

leikoilja, 2015-01-26:
@jantje, yes, thank you very much for help!
The only thing I haven't really found where i can make functions such as "digitalWrite" and etc. be coloured(highlighted). Even download some additional plugin called "Eclipse colorer" for C/C++ language, but turns out that is something else... it didn't help me :D

rlogiacco, 2015-01-27:
The Eclipse editor syntax coloring is the way to go. You don't need any additional tools or plugins as Eclipse has a built-in engine.

The code formatter is for, well, formatting code and setting spacing, indentation and similar. You are looking for the syntax coloring preferences.

jantje, 2015-01-27:
indeed syntax coloring not code formatter. My Bad.

And visit http://eclipse.baeyens.it/donate.shtml to show your appreciation for the support.

closed 2015-04-07 by jantje