Sloeber/arduino-eclipse-plugin #218: "Make" problem.

beingobserver, 2015-01-29:
Hi. (OS X / Kepler / Arduino 1.5.6 r2 / V2.2)

Why i am getting this error?

4:57:43 **** Incremental Build of configuration Release for project TestProject ****
make all 
Cannot run program "make" (in directory "/Users/myComputer/Documents/workspace/TestProject/Release"): error=2, No such file or directory
14:57:43 Build Finished (took 21ms)

Thanks.

rlogiacco, 2015-01-29:
Bad combination of plugin version and Arduino IDE? Have you checked the website if that's an acceptable pair?

beingobserver, 2015-01-29:
Web site is too complicated. Can you give me a link which shows the appropriate combinations?
Thank you.

jantje, 2015-01-29:
>Web site is too complicated. Can you give me a link which shows the appropriate combinations?

Nice compliment.
Basically my thought: If install->start here->mac is to complicated for you .... stick to the arduino IDE.
No thanks
Jantje

closed 2015-01-29 by jantje