Sloeber/arduino-eclipse-plugin #219: Code for Small Blinky Teensy 3.1 Project Compiles to 42k

mizraith, 2015-01-30:
Not sure this is a bug, but looking for someone who can point me to optimization options.

When compiling a basic blinky example  (Arduino Eclipse 2.4), Arduino 1.5.8, Mac OS  X 10.9.5 for the Teensy 3.1, the resultant .hex file comes out at 42k!  The same program on the Uno is 3k.   I understand I might be able to tweak optimizations in boards.txt, etc....but don't even know where to begin.  

Any help, guidance or additional documentation would be massively appreciated.  Otherwise, you may close this as "not really an issue" I guess.

mizraith, 2015-01-30:
Okay, update.

When selecting the "Use alternative size command" compile options, the report states a program of 14984 bytes and 2452 data bytes, which is in line with what teensyduino says is used (something like 6%).   The resultant .hex file is still 42k -- but I'm too much of a novice to know why the .hex file would be nearly 3 times larger than what eclipsearduino says is the program size.

jantje, 2015-02-10:
In my test there are no significant differences between the 2. Other people reported there were no differences on their system as well.

There are many reasons why you can have differences. The only way to find out is comparing the 2 builds.
To compare the 2 you need to setup the 2 environment. As Teensy eclipse plugin is mostly hand configuration it is very likely something went wrong there and it is impossible to tell you from here what is causing the differences.
Because arduino IDE and Arduino eclipse plugin should run the exact same commands the output should be exactly the same. If it is not you can find/explain the difference by comparing the commands executed.
To see the commands you need to activate "verbose" in the arduino IDE. The eclipse plugin is always verbose.
If you do so make sure you run a "clean" before compilation in both the Arduino IDE and in the eclipse plugin. The clean in Arduino IDE is a different "Action" based on the version.

Once you found the difference in the commands you can post them here and I can help you try to figure out why the commands are different.

Good luck

jantje, 2015-04-07:
If this was an issue it should be fixed with the new teensyduino

closed 2015-04-07 by jantje

mizraith, 2015-04-24:
Thanks, Janjte.  I think it was a mistake somehow on my part with the arduino IDE and eclipse.   I eventually gave up on Eclipse and went with EmbedXcode...a much nicer project on the Mac.