Sloeber/arduino-eclipse-plugin #220: Support Arduino_STM32 ?

taoyuan, 2015-02-03:
There is a hardware [library](https://github.com/rogerclarkmelbourne/Arduino_STM32) to support STM32 based boards like the LeafLabs maple and other generic STM32F103 based boards. Is it possible to support it in arduino-eclipse-plugin.

taoyuan, 2015-02-04:
There is already boards and platform file and all necessary libs, but I can't compile using arduino eclipse plugin. It shows can't find `stm32.h` error when compiling, but it can work in arduino ide.

taoyuan, 2015-02-04:
I tried `due` and it's working in eclipse. So I compared `due`'s platform.txt and `stm32`'s platform.txt:

due's platform.txt
```
...
compiler.libsam.c.flags="-I{build.system.path}/libsam" "-I{build.system.path}/CMSIS/CMSIS/Include/" "-I{build.system.path}/CMSIS/Device/ATMEL/"
...
```

stm32's platform.txt
```
...
compiler.libs.c.flags="-I{build.system.path}/libmaple" "-I{build.system.path}/libmaple/include" "-I{build.system.path}/libmaple/stm32f1/include" "-I{build.system.path}/libmaple/usb/stm32f1" "-I{build.system.path}/libmaple/usb/usb_lib" "-I{build.system.path}/libmaple/include/libmaple"
...
```

They are both using `{build.system.path}` variable. It can be convert correctly in `sam`, not in `stm32`. So find it in plugin's source code, and found this in [`ArduinoHelpers.java`](https://github.com/jantje/arduino-eclipse-plugin/blob/50f1148b89ebc2e43958584316f6b4e698ad462a/it.baeyens.arduino.core/src/it/baeyens/arduino/tools/ArduinoHelpers.java#L567-L579):
```java
...
	if (platformFile.segment(platformFile.segmentCount() - 2).equals("avr")) {
	    var = new EnvironmentVariable(ENV_KEY_compiler_path, makeEnvironmentVar("A.RUNTIME.IDE.PATH") + "/hardware/tools/avr/bin/");
	    contribEnv.addVariable(var, confDesc);
	} else if (platformFile.segment(platformFile.segmentCount() - 2).equals("sam")) {
	    var = new EnvironmentVariable(ENV_KEY_build_system_path, makeEnvironmentVar("A.RUNTIME.IDE.PATH") + "/hardware/arduino/sam/system");
	    contribEnv.addVariable(var, confDesc);
	    var = new EnvironmentVariable(ENV_KEY_build_generic_path, makeEnvironmentVar("A.RUNTIME.IDE.PATH")
		    + "/hardware/tools/g++_arm_none_eabi/arm-none-eabi/bin");
	    contribEnv.addVariable(var, confDesc);
	} else if (platformFile.segment(platformFile.segmentCount() - 2).equals("mtk")) {
	    var = new EnvironmentVariable(ENV_KEY_build_system_path, makeEnvironmentVar("A.RUNTIME.IDE.PATH") + "/hardware/arduino/mtk/system");
	    contribEnv.addVariable(var, confDesc);
	}
...
```

The `{build.system.path}` variable seems only available in `sam` or `mtk`.

jantje, 2015-02-04:
I don't like that code and I think there is a far easier more open way.
That is: add the following to the platform.txt file
build.system.path={runtime.ide.path}/hardware/tools/avr/bin/
off course with the path mtk needs.
After the change you need to select your board again in the project properties and select apply to make the changes in the config of our project.
success

taoyuan, 2015-02-05:
Yes. Now all files founded after add the following to the platform.txt file:
```
build.system.path={runtime.platform.path}/system
```

But throws many `multiple definition of xxx` errors, like these:

```
...
/Users/taoyuan/Documents/Arduino/hardware/Arduino_STM32/STM32F1XX/cores/maple/libmaple/usb/stm32f1/usb.c:90: multiple definition of `usb_init_usblib'
/Users/taoyuan/Documents/arduino-workspace-avr/armdemo/Release/arduino.ar(usb.c.o):/Users/taoyuan/Documents/Arduino/hardware/Arduino_STM32/STM32F1XX/cores/maple/libmaple/usb/stm32f1/usb.c:90: first defined here
/Users/taoyuan/Documents/arduino-workspace-avr/armdemo/Release/arduino.ar(usb.c.o): In function `dispatch_endpt':
/Users/taoyuan/Documents/Arduino/hardware/Arduino_STM32/STM32F1XX/cores/maple/libmaple/usb/stm32f1/usb.c:187: multiple definition of `__irq_usb_lp_can_rx0'
/Users/taoyuan/Documents/arduino-workspace-avr/armdemo/Release/arduino.ar(usb.c.o):/Users/taoyuan/Documents/Arduino/hardware/Arduino_STM32/STM32F1XX/cores/maple/libmaple/usb/stm32f1/usb.c:187: first defined here
...
```

jantje, 2015-02-05:
you need to take a look at the code. you are probably missing a define.

taoyuan, 2015-02-06:
The problem is that it is ok in Arduino IDE.

taoyuan, 2015-02-06:
Here is an example code for the first `multiple definition of xxx` error.

usb.h
```c
...
void usb_init_usblib(usblib_dev *dev,
                     void (**ep_int_in)(void),
                     void (**ep_int_out)(void));

...
```

usb.c
```c
...
void usb_init_usblib(usblib_dev *dev,
                     void (**ep_int_in)(void),
                     void (**ep_int_out)(void)) {
    rcc_clk_enable(dev->clk_id);

    dev->ep_int_in = ep_int_in;
    dev->ep_int_out = ep_int_out;

    /* usb_lib/ declares both and then assumes that pFoo points to Foo
     * (even though the names don't always match), which is stupid for
     * all of the obvious reasons, but whatever.  Here we are. */
    pInformation = &Device_Info;
    pProperty = &Device_Property;
    pUser_Standard_Requests = &User_Standard_Requests;

    pInformation->ControlState = 2; /* FIXME [0.0.12] use
                                       CONTROL_STATE enumerator */
    pProperty->Init();
}
...
```

Any problem ?

jantje, 2015-02-06:
As I stated before: Feel free to add STM32 base support. In other words I'm not going to do it.
I'm wiling to support you in getting it done but you'll have to do it.
As far as I see it you have gotten so far that the compilation and linking commands are executed. However there is a coding problem. There are 2 ways to try to find out what is going one.
1) look at the code and try to find out what the problem is and based on that modify the build/linking commands
2) look at the build/linking commands used by the arduino IDE and the eclipse plugin. Compare them and spot the differences.

I recall now that I have been in contact with @nicoverduin from http://www.verelec.nl and he told me he had got this to work. He mentioned planning to do some writeup on it.

taoyuan, 2015-02-07:
If a verbal clash before, I feel very sorry about that.

I see what you mean, but ran into the problems mentioned above, did not find a solution, so I want to consult you.

By the way, this plugin is great. Thank you for your pay. On this issue, I will continue to research it.

Thank you for your help.

nicoverduin, 2015-02-07:
I wanted to use MBED in the Eclipse Arduino environment and created my own boads.tx and platform.txt file
platform.txt:
```
#
# STM32 compile variables
# --------------------- 

#compiler.path={runtime.ide.path}/hardware/tools/g++_arm_none_eabi/bin/
compiler.path=D:/gcc-arm-none-eabi/bin/
#
# platform include files
# build.family and build.target may be found in boards.txt 
#
build.includes=-I"{runtime.hardware.path}/stm32/{architecture}/cores/{build.core}/api" -I"{runtime.hardware.path}/stm32/{architecture}/cores/{build.core}/targets/hal/{build.family}/{build.target}" -I"{runtime.hardware.path}/stm32/{architecture}/cores/{build.core}/targets/cmsis/{build.family}/{build.target}" -I"{runtime.hardware.path}/stm32/{architecture}/cores/{build.core}/targets/cmsis" -I"{runtime.hardware.path}/stm32/{architecture}/cores/{build.core}/common" -I"{runtime.hardware.path}/stm32/{architecture}/cores/{build.core}/hal"
#
# C compiler flags
#
compiler.c.cmd=arm-none-eabi-gcc
compiler.c.flags=-c -g -Os -w -march=armv7-m -mthumb -mthumb-interwork -mlittle-endian  -munaligned-access -fno-exceptions 
compiler.c.flags2=-Os -fmessage-length=0 -ffunction-sections -fdata-sections -mfloat-abi=soft  -g -DSTM32L152RD -DSTM32L1XX_HD 
compiler.c.flags3=-fabi-version=6 -fno-exceptions 
#
# assembly command and flags
#
compiler.s.cmd=arm-none-eabi-gcc
compiler.s.flags=-c -g -Os -w -mthumb -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -Wa,--no-warn, -MMD -MP 
#
# C++ compiler flags
#
compiler.cpp.cmd=arm-none-eabi-g++
compiler.cpp.flags=-march=armv7-m -mthumb -mthumb-interwork -mlittle-endian -munaligned-access  
compiler.cpp.flags2=-Os -fmessage-length=0 -ffunction-sections -fdata-sections -mfloat-abi=soft  -g -DSTM32L152RD -DSTM32L1XX_HD 
compiler.cpp.flags3=-std=gnu++11 -fabi-version=6 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics
#
# linker flags. Archiver is not used with these boards
# 
compiler.c.elf.cmd=arm-none-eabi-g++
compiler.c.elf.flags=-Os -u _printf_float -u _scanf_float -specs=nano.specs -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys
#
# archiver flags (we cannot comment them out or delete them as the make file still expects an archiver
#
compiler.ar.cmd=arm-none-eabi-ar
compiler.ar.flags=rcs
#
# object copy (create HEX or BIN file)
#
compiler.objcopy.cmd=arm-none-eabi-objcopy
compiler.objcopy.eep.flags=-O --strip-all ihex -j .text -j .data -Xlinker -j .eeprom    .eeprom=0 -jstack=0x0100 -jdevice=cortex-m3 -jlscript="{runtime.hardware.path}/stm32/{architecture}/cores/{build.core}/stm32l152re_flash.ld" -jheap=0x0000 

compiler.elf2hex.flags=-O binary
compiler.elf2hex.cmd=arm-none-eabi-objcopy
#
# linker flags
#
compiler.ldflags=
#
# sizer
#
compiler.size.cmd=arm-none-eabi-size

# this can be overriden in boards.txt
build.extra_flags=-DTARGET_NUCLEO_L152RE -DTARGET_M3 -DTARGET_CORTEX_M -DTARGET_STM -DTARGET_STM32L1 -DTARGET_STM32L152RE -DTOOLCHAIN_GCC_ARM -DTOOLCHAIN_GCC -D__CORTEX_M3 -DARM_MATH_CM3 -DMBED_BUILD_TIMESTAMP=1417636368.24 -D__MBED__=1 -DTARGET_FF_ARDUINO -DTARGET_FF_MORPHO 
#
# Cortex STM32 ARM M3 compile patterns
# ------------------------------

## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" -mcpu={build.mcu} {compiler.c.flags} {build.extra_flags} {compiler.c.flags2} {includes} {build.includes}  {compiler.c.flags3} -MMD -MP {build.extra_flags} "{source_file}" -c -o "{object_file}" 

## Compile c++ files
[... 33 lines cut ...]
boards.txt
```
stm32_nucleo_l152re.name=STM32_NUCLEO_L152RE
stm32_nucleo_l152re.build.target=TARGET_NUCLEO_L152RE
stm32_nucleo_l152re.build.family=TARGET_STM
stm32_nucleo_l152re.upload.tool=bossac
stm32_nucleo_l152re.upload.protocol=sam-ba
stm32_nucleo_l152re.upload.maximum_size=524288
stm32_nucleo_l152re.upload.use_1200bps_touch=true
stm32_nucleo_l152re.upload.wait_for_upload_port=false
stm32_nucleo_l152re.upload.native_usb=false
stm32_nucleo_l152re.build.mcu=cortex-m3
stm32_nucleo_l152re.build.f_cpu=8000000L
stm32_nucleo_l152re.build.core=mbed
#stm32_nucleo_l152re.build.extra_flags=
stm32_nucleo_l152re.build.ldscript=linker_scripts/gcc/flash.ld
stm32_nucleo_l152re.build.variant=
stm32_nucleo_l152re.build.variant_system_lib=
stm32_nucleo_l152re.build.vid=
stm32_nucleo_l152re.build.pid=
stm32_nucleo_l152re.build.noarchive=true
stm32_nucleo_l152re.build.use_archiver=false
```

It worked. But the code was never really as small as it could be. Somehow I still got large binaries then for example at MBED.org compiler or the EM::Blocks IDE. Then I tried to add a second board STM32F103C8T6 (those very cheap 6US$ boards on ebay). And somehow I screwed up my environment.. Pissed as I was (as I couldn't get anything to work any more :)) I gave up and am using EM::Blocks (with project generation on MBED.org). Code is compact. Essentially it is, of course, my lack of experience in building MAKE files etc for GCC. Must learn that one day.

jantje, 2015-02-07:
@nicoverduin 
rofl Pissed = zat; pissed off= boos
;-)

jantje, 2015-05-27:
with the  new labelling I'm doing I can reopen this issue.
If you want to implement this please take ownership of this issue. Atthe time of writing I'm not planning to implement this.
Note that I started a hardware repository. I think that is a better fit.

jantje, 2016-03-25:
I close this one as now it is only a matter of making a good json file.

closed 2016-03-25 by jantje