Sloeber/arduino-eclipse-plugin #221: Support serial.port config variable for upload.pattern

anidev, 2015-02-06:
The Intel Galileo board is an Arduino-compatible board that uses the Arduino IDE. In its platform.txt, it specifies an upload.pattern that makes use of the serial.port config variable. Here is the revelant line:
`tools.izmirdl.upload.pattern=/bin/bash --verbose  --noprofile {tools.izmirdl.cmd.path} "{runtime.ide.path}/hardware/tools" {build.path}/{build.project_name}.elf {serial.port}`

The eclipse plugin doesn't seem to fill in {serial.port} at all when uploading. Instead, it just removes {serial.port} from the command entirely. The other parameters seem to be working fine.

jantje, 2015-02-07:
Normally {serial.port} should be replaced with the serial port set in the project properties.
I did some changes in this area lately in regards to being more verbose on what goes on. Can you try with the nightly and dump the console output?

anidev, 2015-02-08:
I set the serial port in the project properties, which in my case is /dev/ttyACM0.

I cloned this repo and built it locally. After installing the locally built version in Eclipse and recreating the project, I tried uploading again and got the same problem. This is what the console said:

````
Starting upload
using arduino loader
Starting reset using DTR toggle process
Flushing buffer
Toggling DTR
Continuing to use /dev/ttyACM0
Ending reset using DTR toggle process

Launching /bin/bash --verbose --noprofile /home/anirudh/Projects/AstroBotics/GalileoDev/arduino-1.5.3-Intel.1.0.4/hardware/arduino/x86/tools/izmir/clupload_linux.sh /home/anirudh/Projects/AstroBotics/GalileoDev/arduino-1.5.3-Intel.1.0.4/hardware/tools /home/anirudh/Projects/AstroBotics/GalileoDev/Workspace/Blink/Release/Blink.elf 
Output:
<output from the shell script>
bash finished
upload done
````

After Blink.elf, there should be the serial port, as per the above upload.pattern. Incidentally, the new version of the plugin asks about the uploading protocol, and the only option is Default. Does that have anything to do with it?

jantje, 2015-02-08:
Protocol is something advanced and not related. Just keep that to Default.
I suspect this is a bug related to {serial.port} being at the end of the command.
Can you check what the value is of the environment variable "A.TOOLS.IZMIRDL.UPLOAD.PATTERN"
look in "project properties->C/C++ build/Environment"

anidev, 2015-02-08:
This is what I see for the A.TOOLS.IZMIRDL.UPLOAD.PATTERN: /bin/bash --verbose  --noprofile ${A.TOOLS.IZMIRDL.CMD.PATH} "${A.RUNTIME.IDE.PATH}/hardware/tools" ${A.BUILD.PATH}/${A.BUILD.PROJECT_NAME}.elf ${A.SERIAL.PORT}

jantje, 2015-02-09:
Wierd. I had expected this to be wrong but that is what it should be.
What is the value of A.SERIAL.PORT?

anidev, 2015-02-09:
Hmm, as far as I can tell A.SERIAL.PORT doesn't seem to be set. Not sure why.

jantje, 2015-02-09:
ok I know what is causing the problem now.
The problem is in uploadUsingPreferences
This method assumes that the upload port is preceded with -P or  -port or --port=
I'm not sure yet what the best way is to work around this problem
I need to think about it

to test this hypothesis you could change the command to
tools.izmirdl.upload.pattern=/bin/bash --verbose --noprofile {tools.izmirdl.cmd.path} "{runtime.ide.path}/hardware/tools" {build.path}/{build.project_name}.elf -P {serial.port}
Do not foreget to reselect the board in project properties and select apply to make the change happen.

jantje, 2015-02-09:
I have written some code that should do the trick but I can't test now. As you have a local copy right now you can give it a try

   @Override
    public boolean uploadUsingPreferences(IFile hexFile, IProject project, boolean usingProgrammer, IProgressMonitor monitor) {
	String MComPort = "";
	String boardName = "";

	IEnvironmentVariableManager envManager = CCorePlugin.getDefault().getBuildEnvironmentManager();
	IContributedEnvironment contribEnv = envManager.getContributedEnvironment();
	ICProjectDescription prjDesc = CoreModel.getDefault().getProjectDescription(project);
	ICConfigurationDescription configurationDescription = prjDesc.getConfigurationByName(mycConf);

	try {
	    MComPort = envManager.getVariable(ArduinoConst.ENV_KEY_JANTJE_COM_PORT, configurationDescription, true).getValue();
	} catch (Exception e) {// ignore all errors
	}
	try {
	    boardName = envManager.getVariable(ArduinoConst.ENV_KEY_JANTJE_BOARD_NAME, configurationDescription, true).getValue();
	} catch (Exception e) {// ignore all errors
	}
	String NewSerialPort = ArduinoSerial.makeArduinoUploadready(myConsole.newMessageStream(), myProject, mycConf, MComPort);

	IEnvironmentVariable var = new EnvironmentVariable(ArduinoConst.ENV_KEY_SERIAL_PORT, NewSerialPort);
	contribEnv.addVariable(var, configurationDescription);
	var = new EnvironmentVariable(ArduinoConst.ENV_KEY_SERIAL_PORT_FILE, NewSerialPort.replace("/dev/", ""));
	contribEnv.addVariable(var, configurationDescription);

	String command = "";
	try {
	    command = envManager.getVariable("A.TOOLS." + myUploadTool.toUpperCase() + ".UPLOAD.PATTERN", configurationDescription, true).getValue();
	} catch (Exception e) {// ignore all errors
	}

	try {
	    GenericLocalUploader.RunConsoledCommand(myConsole, command, new SubProgressMonitor(monitor, 1));
	} catch (IOException e1) {
	    e1.printStackTrace();

	    return false;
	}
	if (boardName.startsWith("Arduino Due ")) {
	    ArduinoSerial.reset_Arduino_by_baud_rate(MComPort, 115200, 100);
	}

	return true;
    }

anidev, 2015-02-09:
I tried to compile the above code. Am I right in assuming that it should be in the class arduinoUploader?
Working off that assumption, I got a few compile errors, most of which were classes that needed to be imported. After fixing that, there were still two errors: ENV_KEY_SERIAL_PORT and ENV_KEY_SERIAL_PORT_FILE in ArduinoConst could not be resolved.

jantje, 2015-02-10:
My bad.
This needs to be added to ArduinoConst.java
```java
    public static final String ENV_KEY_SERIAL_PORT = ENV_KEY_ARDUINO_START + "SERIAL.PORT";
    public static final String ENV_KEY_SERIAL_PORT_FILE = ENV_KEY_ARDUINO_START + "SERIAL.PORT.FILE";
```

In arduinoUploader.java replace the method uploadUsingPreferences with the one below
```java
@Override
    public boolean uploadUsingPreferences(IFile hexFile, IProject project, boolean usingProgrammer, IProgressMonitor monitor) {
	String MComPort = "";
	String boardName = "";

	IEnvironmentVariableManager envManager = CCorePlugin.getDefault().getBuildEnvironmentManager();
	IContributedEnvironment contribEnv = envManager.getContributedEnvironment();
	ICProjectDescription prjDesc = CoreModel.getDefault().getProjectDescription(project);
	ICConfigurationDescription configurationDescription = prjDesc.getConfigurationByName(mycConf);

	try {
	    MComPort = envManager.getVariable(ArduinoConst.ENV_KEY_JANTJE_COM_PORT, configurationDescription, true).getValue();
	} catch (Exception e) {// ignore all errors
	}
	try {
	    boardName = envManager.getVariable(ArduinoConst.ENV_KEY_JANTJE_BOARD_NAME, configurationDescription, true).getValue();
	} catch (Exception e) {// ignore all errors
	}
	String NewSerialPort = ArduinoSerial.makeArduinoUploadready(myConsole.newMessageStream(), myProject, mycConf, MComPort);

	IEnvironmentVariable var = new EnvironmentVariable(ArduinoConst.ENV_KEY_SERIAL_PORT, NewSerialPort);
	contribEnv.addVariable(var, configurationDescription);
	var = new EnvironmentVariable(ArduinoConst.ENV_KEY_SERIAL_PORT_FILE, NewSerialPort.replace("/dev/", ""));
	contribEnv.addVariable(var, configurationDescription);

	String command = "";
	try {
	    command = envManager.getVariable("A.TOOLS." + myUploadTool.toUpperCase() + ".UPLOAD.PATTERN", configurationDescription, true).getValue();
	} catch (Exception e) {// ignore all errors
	}

	try {
	    GenericLocalUploader.RunConsoledCommand(myConsole, command, new SubProgressMonitor(monitor, 1));
	} catch (IOException e1) {
	    e1.printStackTrace();

	    return false;
	}
	if (boardName.startsWith("Arduino Due ")) {
	    ArduinoSerial.reset_Arduino_by_baud_rate(MComPort, 115200, 100);
	}

	return true;
    }
```

Github tells me eclipse added these imports to arduinoUploader.java
```java
import org.eclipse.cdt.core.CCorePlugin;
import org.eclipse.cdt.core.envvar.EnvironmentVariable;
import org.eclipse.cdt.core.envvar.IContributedEnvironment;
import org.eclipse.cdt.core.envvar.IEnvironmentVariable;
import org.eclipse.cdt.core.envvar.IEnvironmentVariableManager;
import org.eclipse.cdt.core.model.CoreModel;
import org.eclipse.cdt.core.settings.model.ICConfigurationDescription;
import org.eclipse.cdt.core.settings.model.ICProjectDescription;
```

anidev, 2015-02-10:
All right! Looks like it works now. This is the command it runs now:
`Launching /bin/bash --verbose --noprofile /home/anirudh/Projects/AstroBotics/GalileoDev/arduino-1.5.3-Intel.1.0.4/hardware/arduino/x86/tools/izmir/clupload_linux.sh /home/anirudh/Projects/AstroBotics/GalileoDev/arduino-1.5.3-Intel.1.0.4/hardware/tools /home/anirudh/Projects/AstroBotics/GalileoDev/Workspace/Blink/Release/Blink.elf /dev/ttyACM0 `

I can't actually test it with the board connected right now, but at least it's running the correct command as far as I can tell. Later today I'll actually connect the board and try uploading then, but I think it will work. Thanks!

jantje, 2015-02-10:
Great
I have no default uploader board at hand right now. So let me know if it works and if it does I'll release this code.

anidev, 2015-02-11:
Just tested on the Galileo, seems to be working perfectly.

jantje, 2015-02-11:
cool. Tomorrow the nightly should contain the fix.

closed 2015-02-11 by anidev