Sloeber/arduino-eclipse-plugin #222: Cannot use ArduinoISP programmer

jfpoilpret, 2015-02-09:
Hi, I use Arduino Eclipse on Windows (28.12.14 archive) and I tried to perform upload through the ArduinoISP programmer.

It happens that on Windows, ArduinoISP uses a special driver upon USB, hence there is no COMx port associated with it when connected:

[image]

I pointed Arduino Eclipse plugin to Arduino IDE 1.5.8, that version includes the "arduinoisp" programmer.

I set this programmer in my projects settings without any COM port:

[image]

When I try to upload my sketch, I first get an error message:

[image]

If I click OK, I get the following in the console:

    Launching D:/electronics/arduino/tools/arduino-1.5.8/hardware/tools/avr/bin/avrdude -CD:/electronics/arduino/tools/arduino-1.5.8/hardware/tools/avr/etc/avrdude.conf -patmega328p -carduinoisp -P -b19200 -D -Uflash:w:D:\electronics\projects\alarm-system\modules\ActivationModule/Release/ActivationModule.hex:i 

    Output:
    avrdude: Error: Invalid -P value: '-b19200'
             Use -P usb:bus:device

    avrdude done.  Thank you.

I also tried to put a port that I don't use (COM3), but got the same result.

Then I verified the `avrdude` command line directly in a command prompt, but I just removed the `-P` and `-b` options:

    D:/electronics/arduino/tools/arduino-1.5.8/hardware/tools/avr/bin/avrdude -CD:/electronics/arduino/tools/arduino-1.5.8/hardware/tools/avr/etc/avrdude.conf -patmega328p -carduinoisp -Uflash:w:D:\electronics\projects\alarm-system\modules\ActivationModule/Release/ActivationModule.hex:i 

Here is the result:

    D:\>D:/electronics/arduino/tools/arduino-1.5.8/hardware/tools/avr/bin/avrdude -CD:/electronics/arduino/tools/arduino-1.5.8/hardware/
    tools/avr/etc/avrdude.conf -patmega328p -carduinoisp -Uflash:w:D:\electronics\projects\alarm-system\modules\ActivationModule/Release
    /ActivationModule.hex:i
    
    avrdude: AVR device initialized and ready to accept instructions
    
    Reading | ################################################## | 100% 0.02s
    
    avrdude: Device signature = 0x1e950f
    avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
    avrdude: erasing chip
    avrdude: reading input file "D:\electronics\projects\alarm-system\modules\ActivationModule/Release/ActivationModule.hex"
    avrdude: writing flash (11004 bytes):
    
    Writing | ################################################## | 100% 13.62s
    
    avrdude: 11004 bytes of flash written
    avrdude: verifying flash memory against D:\electronics\projects\alarm-system\modules\ActivationModule/Release/ActivationModule.hex:
    avrdude: load data flash data from input file D:\electronics\projects\alarm-system\modules\ActivationModule/Release/ActivationModule
    .hex:
    avrdude: input file D:\electronics\projects\alarm-system\modules\ActivationModule/Release/ActivationModule.hex contains 11004 bytes
    avrdude: reading on-chip flash data:
    
    Reading | ################################################## | 100% 9.64s
    
    avrdude: verifying ...
    avrdude: 11004 bytes of flash verified
    
    avrdude: safemode: Fuses OK (H:05, E:DE, L:E2)
    
    avrdude done.  Thank you.

Note that I also had to remove the `-D` option otherwise the upload would fail.

Would it be possible to adapt Arduino Eclipse plugin to also cover the ArduinoISP specific case?

jantje, 2015-02-09:
It is possible. Feel free to do it :-)

jantje, 2015-05-27:
reopened as I have no objection to implementing this but at this time no intention to do so.
By using labels I can now flag this better.

razerraz, 2015-06-17:
Don't know if it's the same issue, posting here in doubt
I have a bunch of ATMega32 & AVRISP MKII programmers.
I'm able to use it with arduino java interface, using "upload with a programmer", and in eclipse when using AVR project, but not with arduino-eclipse plugin
Since it seems simple to implement (just calling avrdude with correct parameters and hex file), I'm looking for a way to do that inside arduino-eclipse project
Cheers

jantje, 2015-06-17:
@razerraz 
feel free to implement.

hasselk, 2016-02-27:
@jantje 
Hi, could you show me a hint where the options for avrdude are handled? heavent found it yet..
i have to use an ISP programmer(buspirate) and the default "-D" option does not work with this setup, i always get verification errors. I´m using the nightly version

Thanks Hassel

jantje, 2016-02-27:
you should probably start by reading these to get an understanding of how it works.
http://iloapp.baeyens.it/blog/blog?Home&post=22
http://iloapp.baeyens.it/blog/blog?Home&post=24

jantje, 2016-04-22:
I've started working on this.
Got a bit stuck on how the Arduino core team interpretes the programmers.txt content. So I asked for some clarification. http://forum.arduino.cc/index.php?topic=395666.0

jantje, 2016-04-26:
I'm not sure how good this will work. As i only have a arduino as isp there is not mutch I can test.

razerraz, 2016-04-27:
> 9081

Fantastic ! Grats !

I'm in the middle of too many other things right now, but count on me to give it
a try soon
On the other hand, I've always more issues when I upgrade your plugin with my
eclipse install mess (I don't use the standalone version and then have to fix
buildin variables to make it work, so maybe I'll need some time before be able
to test)

cheers

jantje, 2016-05-17:
@razerraz 
If you have time to test, don't forget to report the results :-)

jantje, 2016-06-22:
@razerraz
I'm still interested in your test results.

razerraz, 2016-06-22:
> 5710

What a coincidence, I try the thing today and it's working, at least with mkII
programmers.
Congrats and thanks, I'm considering now every day usage of you eclipse bundle
with students.

Bye

PS : I'll try soon another homemade uno base programmers to be sure and give my
feedback

jantje, 2016-06-22:
thanks for the feedback :-)

closed 2016-08-01 by jantje