Sloeber/arduino-eclipse-plugin #224: Upload using programmer

andmiller, 2015-02-16:
I have been experimenting with Arduino and the Arduino Eclipse plugin with my dated 3pi robot. I have gotten it to work in the Arduino IDE by using the Upload using programmer option, but that option is not available in the Eclipse plugin. When AVRDude runs, it is missing the -c (programmerId) parameter. I tried modifying the platform.txt file in D:\Arduino\hardware\arduino\avr to make it static, but the Eclipse plugin seemed to ignore the change. Are you using the platform.txt config file, and if so, is there a way to refresh the plugin?

jantje, 2015-02-16:
The plugin only reads platform.txt and boards.txt at the time of board selection.
When changing platform.txt or boards.txt goto project properties->arduino and select apply.
Now the plugin should be using the new values.
Best regards
Jantje

andmiller, 2015-02-16:
That answers my question about loading the boards.txt and platform.txt. Here is the additions to the platform.txt to get this resolved:
tools.avrdudeISP.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude
tools.avrdudeISP.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf

tools.avrdudeISP.upload.params.verbose=-v
tools.avrdudeISP.upload.params.quiet=-q -q
tools.avrdudeISP.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} "-Uflash:w:{build.path}/{build.project_name}.hex:i"

Then in boards.txt I have the following entry for the 3pi board:

#######3pi

3pi.name=3Pi Atmega 168
3pi.upload.tool=avrdudeISP
3pi.upload.protocol=avrispmkii
3pi.build.f_cpu=20000000L
3pi.build.core=arduino:arduino
3pi.build.variant=standard
3pi.build.upload.maximum_size=16384
3pi.build.mcu=atmega168

Hope this is helpful to others, thanks for your quick response, great work...

closed 2015-02-16 by andmiller

jantje, 2015-02-16:
you're welcome