Sloeber/arduino-eclipse-plugin #225: Eclipse showing errors

andmiller, 2015-02-17:
I know this was discussed here: http://baeyens.it/eclipse/faq.shtml#/troubleshooting, but I am using the newest download (Product bundle), and have updated to the latest version of the plugin, and the options mentioned in the FAQs no longer exist in Eclipse. I am having the same issue as mentioned in 'Eclipse tells me there are bugs in my code but it compiles ok.'. Is there a fix or work-around for this in the newest version? Below is a simple test program for some of the Pololu 3pi libraries:
[image]

jantje, 2015-02-17:
First of all: There is definitely something wrong.
Secondly: The problem matches the problem description of  'Eclipse tells me there are bugs in my code but it compiles ok.' :-)

but I'm a bit surprised when I look at the image.
I see that pinMode, DigitalWrite, HIGH and LOW are recognized. So Arduino.h is parsed properly. It looks like the urangutan libraries are not parsed properly.
I see that arduino is part of the project so this looks like V2
I see the arduino icon so this is the product version.
I do not see a Arduino menu :-(
I do not see the Arduino tool bar :-(
So the image looks "mixed" or "weird" to me.

so here are my questions:
Have you intentionally removed the arduino menu and arduino toolbar? If so why?
How have you created the project?
How have you added the urangutan libraries to the project?
How do you start a build?
Which arduino IDE is underlying?
Have you compiled with all warnings and if so does the compile generate warnings?

Best regards
jantje

andmiller, 2015-02-17:
Sorry, took the previous picture in 'Resource' perspective.
### Answers: ###
*Have you intentionally removed the arduino menu and arduino toolbar? If so why?*
**No, Wrong perspective.**
*How have you created the project?*
**Using the Arduino plugin.**
*How have you added the urangutan libraries to the project?*
**Using the Arduino plugin.**
*How do you start a build?*
**Tried 'Verify' from Arduino plugin, and 'Build All' from Eclipse standard. Same results.**
*Which arduino IDE is underlying?*
**1.6.0**
*Have you compiled with all warnings and if so does the compile generate warnings?*
**Only warning is: 'MS-DOS style path detected', but the project compiles fine.**

New screen shot below in 'Arduino' perspective:
[image]

Also, installation details for Arduino plugin:
[image]

rlogiacco, 2015-02-17:
Just a side note that might help you out: `OrangutanLCD::clear()` is C++ notation: is it possible the .ino files are limited to being C? Do the errors go away if you convert the C++ notation to the corresponding C notation?
```
// OrangutanLCD::clear();
clear();
// OrangutanLCD::print(delayMillis);
print_long(delayMillis);
```

andmiller, 2015-02-17:
Genius! That took care of my issue, should we have an issue opened '.ino files don't support C++ notation when parsing'? Thanks for the insights rlogiacco!

andmiller, 2015-02-17:
OK, found a solution to the issue, and still be able to use C++ notation, add a language mapping:

[image]

No errors:
[image]

closed 2015-02-17 by andmiller

jantje, 2015-02-17:
are you sure you mapped ino to c++ and not to C?

rlogiacco, 2015-02-17:
Glad having been helpful :smile: