Sloeber/arduino-eclipse-plugin #227: Sometimes the include of a library is removed wihout reason

jantje, 2015-02-23:
Jantje experienced this problem and mwahahaha reported this issue on the arduino forum.

>I have an issue with the Adafruit_BMP085_U library. It works fine when I import it but if I close and reopen Eclipse (after a shutdown for instance), I get an error message saying "Adafruit_BMP085_U.h" can't be found even though the library still appears in the tree !
If I reimport it, it works again for the session but I think it also disappered once in the session itself !

Reimporting the library is a safe workaround. It seems not related to it.baeyens.cdt.refactor

rlogiacco, 2015-02-25:
Can it be related to the path length? It's a quite long name for a library and if deeply nested in a long directory structure it might break some limit... Does it occur if you rename the library to something like A_085_U? And if you rename it to AdafruitBMP085Uv2 ? The latter has the same length but no underscores...
I hardly believe it can have anything to do with the library content itself...

jantje, 2015-02-25:
@rlogiacco 
Could be related, indeed.

GoGunner, 2015-02-26:
Using XP, Juno, plugin V1 and this path 
C:\shdwd\Arduino\libraries\Adafruit_BMP085_Unified\Adafruit_BMP085_U.h
and is linked to the following folder:
E:\AVRARDV1\crap\Uno\Adafruit_BMP085_Unified
there are no problems.

jantje, 2015-02-26:
I had it again with the Time Library and  DS1307.
Both were gone.
Added them again and restarted. DS1307 was gone. Time was still there
Added DS1307 and restarted. Both were there.
I didn't see any errors that i thought were related.
I had this project for a long time and i only recently added a .h and .cpp to the root of the project. That is the only thing that I can think of that might be related.

jantje, 2016-01-20:
This should be autofixed with the new library inclusion mechanism

closed 2016-04-01 by jantje