Sloeber/arduino-eclipse-plugin #228: symbols tab missing

GoGunner, 2015-02-24:
Using Juno & plugin 2.4.0.201502210217
Previously was a V1 user so don't know if the functionality was removed or other.
Properties > C/C++ General > Paths and Symbols >  no longer add the Symbols tab on new projects.
[image]

jantje, 2015-02-24:
@GoGunner 
Only the Arduino section is under control of the arduino eclipse plugin.
Is it this you are looking for?
[image]

GoGunner, 2015-02-24:
As I can see it was at both places with the Symbols tab information being piped over to the CDT Managed Build Setting Entries [Shared], which was not editable.  I take it then that this is the new behavior and access point with the Symbol tab not being a part of it?  From the previous release I had expected it in the Symbols tab.   It doesn't seem to be the standard behavior one might expect, especially someone that is migrating over from V1 or C++.
[image]

jantje, 2015-02-24:
I'm not quite following?
I assume your problem is fixed. So I'm closing the issue. If i misunderstood please reopen.

closed 2015-02-24 by jantje