Sloeber/arduino-eclipse-plugin #230: Build fails because of java 7 constructs

jantje, 2015-03-07:
@wimjongman 
@brodykenrick 
@anyone who has a clue :-)

Due to a forced package upgrade on my system java 1.6 was removed from my system so I had to adopt the project to 1.7. ( I thought we already agreed on moving to 1.7 and I thought we already were but it seems reality is not matching my perception ;-) )
Untill now I fixed all issues and lots of warnings but the maven build now fails with the error
>Resource specification not allowed here for source level below 1.7

So maven seems to think I'm building pre java 1.7
I looked in the pom.xml files but I fail to find the reference to the java version

Please please help

Jantje

jantje, 2015-03-07:
I have run mvn -X verify and I think I see what the problem is

>  <source>1.6</source>
  <staleMillis default-value="0">${lastModGranularityMs}</staleMillis>
  <strictCompilerTarget default-value="false"/>
  <target>1.6</target>

When I look at the pom.xml (with the pom editor) I see 
>              <source>1.6</source>
              <target>1.6</target>

several times. However in the pom.xml this is not there.

So this is a default, but how do I change it? I see nothing in the pom editor

wimjongman, 2015-03-07:
Just edit the pom as plain text file.

jantje, 2015-03-07:
@wimjongman 
Thanks for your reply.
The pom.xml does not contain anything like that. So what do I put there?

jantje, 2015-03-07:
And do I need to change the pom.xml in the root or in application or in all?

wimjongman, 2015-03-07:
Hi Jantje. The root pom points to the parent pom. The parent pom contains references to java 6. If you change this to java7 (or 8. why not), you are golden.

https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.parent/pom.xml

wimjongman, 2015-03-07:
The build is really not that hard to understand (otherwise I could not do it ;). 

It just chains a bunch of modules (bundles) together. When you check out and do 

     mvn clean install

Then it looks up the modules to build. The root pom points to a so called parent pom. This one defines all the plugins it needs to do its magic (e.g. buid an eclipse product). Then this parent points to the modules (bundles) to build. 

	 <modules>
		<module>../it.baeyens.arduino.core.nl1</module>
		<module>../it.baeyens.arduino.feature</module>
		<module>../it.baeyens.arduino.common</module>
		<module>../it.baeyens.arduino.core</module>
		<module>../it.baeyens.arduino.monitor</module>
		<module>../it.baeyens.arduino.product</module>
		<module>../it.baeyens.arduino.updatesite</module>
		<module>../it.baeyens.arduino.application</module>
		<module>../it.baeyens.cdt.refactor</module>
	 </modules>

One of the modules (bundles) is the product and this will produce the actual runtime. 

When looking to change global stuff, the parent pom is the one to visit.

rlogiacco, 2015-03-07:
I can help here, unless you have already figured out...

In the parent pom you need to add or modify the `compiler-plugin`:

```
      <plugin>
        <groupId>org.apache.maven.plugins</groupId>
        <artifactId>maven-compiler-plugin</artifactId>
        <version>3.2</version>
        <configuration>
          <source>1.7</source>
          <target>1.7</target>
        </configuration>
      </plugin>
```

wimjongman, 2015-03-07:
In our case the tycho-compiler-plugin

jantje, 2015-03-07:
found it txs :-))))

closed 2015-03-07 by jantje