Sloeber/arduino-eclipse-plugin #232: Arduino v1.6.1 Path Change

lylemoffitt, 2015-03-19:
In version 1.6.1 of Arduino.app, they changed the path of the `Java` folder; instead of being at `Arduino.app/Contents/Resources/Java/`, it is now at `Arduino.app/Contents/Java/`. The plugin needs to be updated to reflect this change as it is looking for libraries and tools under the former.

lylemoffitt, 2015-03-19:
I tried overriding all the environment variables that use that path in the global preferences, but that doesn't work. The only workaround I know of is manually changing them in *every* projects' property settings.

jantje, 2015-03-19:
isn't this the same as #189?

lylemoffitt, 2015-03-23:
No, I don't think so. That seems to be about where the libraries are, which albeit related is a different issue. I arrived at this problem because the build command couldn't find `avr-g++`. That's because it was looking for it at `[...]/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++`, which no longer exists because the kind people at Arduino moved the `Java` folder up from `Resources/` to `Contents/`.

I might be in the minority affected because I have my "Private Hardware path" and "Private Library path" set to be in another directory entirely.

In any case, the last update to your nightly build (v2.4.0.201503190223) seems to have patched the issue. Though, version 1.6.0 and before are now incompatible.

Here's the console output for a build with v1.6.0 of Arduino.app installed.
```
make all 
Building file: ../.ino.cpp
Starting C++ compile
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -fno-exceptions 
  -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p 
  -DF_CPU=16000000L -DARDUINO=158 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR 
  --std=c++11 -fno-rtti    
  -I"/Volumes/MacBook_HD/Developer/Eclipse-Workspaces/Arduino/libraries/TimerOne" 
  -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI" 
  -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" 
  -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" 
  -I"/Volumes/MacBook_HD/Developer/Eclipse-Workspaces/Arduino/libraries/fast_Digital" 
  -I"/Volumes/MacBook_HD/Developer/Eclipse-Workspaces/Arduino/libraries/fast_SPI" -MMD -MP 
  -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall  
  -std=gnu++11
/bin/sh: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++: No such file or directory
make: *** [.ino.cpp.o] Error 127
```

lylemoffitt, 2015-03-23:
##### Correction: 

It would seem that the path change problem I'm talking about only applies to the Java-7 version of the release. The Java-6 version of the Arduino v1.6.1 has the same path as v1.6.0. So really the current build only works for the Java-7 version of the v1.6.1 release.

jantje, 2015-03-23:
I'm sorry. I fail to understand why this issue is different from #189
The fix I applied tests for both cases Contents/Resources/Java and Contents/Java and uses what matches the file system.
As i only test folder existence the old one may still be around?

lylemoffitt, 2015-03-24:
I'm not totally sure, but it seems to me that #189 is about finding the finding the Java folder from within the Arduino.app. But that process only applies during the creation of a project. New projects seem to work fine with whatever version of Arduino is installed at the time, but older projects don't. 

All the path settings are fixed at whatever they were when the project was created. So if open a project I created before the Java path change now, it won't work. The project doesn't update to reflect the new path. It still looks in the old path (which doesn't exist now). 

This is (i think) because the path variables are specified on a per-project basis, and they cannot be overridden by the global path settings. The way it *should* work is that the path variables are defined in the global build settings, and if need be overridden in the local project settings.

--
So to clarify, this problem exists because there are multiple projects created under different versions with different path schemes. The projects persist with whatever the path layout was when they were created, meaning that old projects are now incompatible with the new layout. Meanwhile, all new projects work fine.

jantje, 2015-03-24:
The problem is you cannot set the variables under a global setting as they change for each project.
Please see http://eclipse.baeyens.it/how_to.shtml#/n
More specifically the after update section to get things working again.

jantje, 2015-04-07:
@lylemoffitt 
Can this issue be closed?

closed 2015-04-16 by jantje