Sloeber/arduino-eclipse-plugin #233: Can't import library when a libraries directory exists

efcis, 2015-04-03:
(Following the forum thread http://forum.arduino.cc/index.php?topic=79595.msg2168896#msg2168896)

Eclipse :Service Release 2 Build id: 20140224-0627
Arduino plugin : 2.2.0.1
Arduino IDE : arduino-1.5.6-r2

I created a new project, and dezipped this GitHub project https://github.com/k3ng/k3ng_rotator_controller into this project directory. Renamed the .ino file to .cpp and edited the header file (.h) to add the function prototypes. Note that a "libraries" subdirectory already exists in this GitHub project.

Then I wanted to import the EEPROM library from "Hardware provided Libraries", but got an error message "Failed to import library Cannot create linked resource '/k3ng_rotator_controller/Libraries/EEPROM'.  The parent resource is not accessible.".

I had to remove the existing libraries directory (actually, I moved its content to my private libraries directory and delete the former - and now empty - directory) to be able to import the requested libraries.

Maybe it was not the correct way to create this project in my workspace, and the plugin get confused ?

Thanks !

jantje, 2015-04-03:
Was this on windows? And if so which version?

efcis, 2015-04-03:
Ooops ! I forgot to mention : Windows 7 SP1 - Sorry

jantje, 2015-04-03:
Is your file system setup to be case sensitive? I ask because I tried to have Libraries and libraries on my windows machine and that worked fine.
Googling a bit there still seem to be case sensitivity issues in windows.
So I wonder can you create 2 folders in a eclipse project only different in casing?

efcis, 2015-04-03:
No, I can't : If I create a folder named "a", and try to create another one named "A", I get the following error message : "A resource exists with a different case 'test_plugin/a'"

jantje, 2015-04-03:
I think that is the problem.
When I have a windows system I'll see whether I can smart this one up. I mean use libraries or give a nice error message.

jantje, 2016-01-07:
I added some code so the error message is better

closed 2016-04-01 by jantje