Sloeber/arduino-eclipse-plugin #234: add support a other board in configuration (discussion for important library change)

jantje, 2015-04-04:
Hi
Playing around with different boards, local builds and unit tests I start to feel the need to make multiple configurations each with a different board.
The first problem I bump into are the libraries. People who use the nightly build may have noticed that I did some changes in this area which I think are improvements (more arduino IDE alike) but I may have to reconsider these.
Let me explain the problem.
When using a library it can be in the hardware folder or the personal folder or the arduino folder. This location can change when you switch "architecture". In contrast to the Arduino IDE the latest stable release lets you select the folder you want to use. At the time of writing the nightly only provides library names (alphabetically sorted) and picks the correct folder for you (like the Arduino IDE). 
I added a method "reattach libraries" you can use if a board switch leads to wrong libraries.

As you can see; if you have multiple boards there is no such thing as the "correct folder"

IMHO it is not possible to have different links to folders based on the configuration.

As such I'd like some input on the following proposal
Instead of adding only 1 location per folder; do it as follows: add all (existing?) locations and use exclude from build to filter the correct folders to be used in this configuration.
To do so the libraries folder must have 3 subfolders "hardware" "arduino" and "private".
The benefit is that it allows (at least a step towards) a board per config.
The drawback is that it breaks the "downwards compatibility" and may come as a cluttered interface.
Please provide input.

jantje, 2015-04-08:
I created a new branch to try out the new way of linking libraries.
I'm still hoping for some feedback input experiences

jantje, 2015-04-08:
I found a way to switch between the boards by using a change config event
I have checked that in in the multiConfig thread.

kdschlosser, 2015-04-10:
any way to add support for the ESP8266 as a Board? someone managed to add it to the Arduino IDE

jantje, 2015-04-10:
@kdschlosser this discussion is about multiple configurations which is different from other boards.
See issue #220 for more details

jantje, 2015-04-14:
I have great results with using a change config event. As this is "as it used to be" and the discussion didn't get any traction I think I'll merge the branches.

jantje, 2015-04-16:
I have fixed the remaining issues with the multiconfiguration. That is setting it up easily (the previous remark was about switching correctly setup configs)

closed 2015-04-22 by jantje

wimjongman, 2015-04-24:
great fix!

jantje, 2015-04-24:
txs. I did some more fixes in this area. It seems pretty stable now.
I'm using google test now to do testing. I'm still learning but it works like a charm.

jantje, 2018-03-03:
oeps wrong issue