Sloeber/arduino-eclipse-plugin #235: Make is not found in path

ranm, 2015-04-06:
Hey , 
I've been trying to work on this for days now without luck. 
I'm installing the plugin on a new windows 64bit PC, existing Luna installation, Arduino version 1.5.6-r2 , installed latest V2.2 
I've set the paths for the IDE in the Arduino tab correctly. 
Created a new project , and can't build :-(

I'm getting this: 
Cannot run program "make" (in directory "C:\Users\ranm\workspace\TestArduino\Release"): CreateProcess error=2, The system cannot find the file specified

why is it that the path is not correctly updated to include the IDE. 
I have the make (it's not 1.5.7) so I don't get it. 
Any inputs?

wimjongman, 2015-04-06:
The error message you get looks quite unambiguous. Please take a look in that directory to make sure that make can be found there.

Cheers,

Wim

ranm, 2015-04-06:
Make is not found in there, that's the path to my project. 
Make should be a part of the IDE directory , but that's not included in the path where the tools seeks for it.

jantje, 2015-04-06:
see http://eclipse.baeyens.it/installAdvice.shtml
Closing as duplicate of #218
and #183

closed 2015-04-06 by jantje

ranm, 2015-04-06:
I'm sorry , but I don't get this still. 
I know about the install advice and the limitations - I read it carefully a few times in the past few days. I'm very familiar with #183 as I'm the one who reported it in the first place and if anything this looks  similar to #218 but I'm using a valid combination of plugin+eclipse+OS+IDE . 
I'm using IDE version 1.5.6-r2 which includes make. 
The issue as I see it,  is that the IDE directory is not included in the path in the first place.
Can you tell me what am I missing ? 
Thanks.

jantje, 2015-04-06:
you may have spaces in your arduino IDE path.
Something may have gone wrong during the installation process.
Something may have gone wrong later
Your path environment var may be to long
.....
Basically the error is clear and it looks like a local issue on your system.  

You can try to get it fixed and stick to this old version or ....
one day or another you will want to upgrade the arduino IDE still using the plugin.
So in that case you will have to install a make utility. So why do not do it right away?

ranm, 2015-04-08:
I agree about the issue being local on my system, but that doesn't make me fell better about it... :-)

I don't mind updating the IDE version, I just wanted to have the same setup on both my PCs. 
I just don't see how that would help, the path the build is looking at doesn't include the IDE directory at all - so changing the files in there shouldn't make a difference. 
I already tried re-installing both plugin and IDE a few times without help. 
Can you instruct me on where can I see the explicit build path to check it for errors? how much is too long? 
At this point I don't mind re-installing anything if it would help, because I'm actually stuck, but I already tried it a bit and it didn't help. I think something with this issue doesn't clear out when I uninstall the plugin. 

I won't be near this PC till Sunday so I'll be happy for tips on what to do till then. 

Anyway , I think we can at least agree this is not a duplicate to any of the issues mentioned (#183 and #218 )

jantje, 2015-04-08:
The easiest way to know the path used is to modify the platform.txt to make one of the commands equal to echo %PATH% (I guess in windows)
When doing a build the console will sow the path.

jantje, 2015-04-08:
If you are really using arduino IDE 1.5.6-r2 ,  this should not be a duplicate of  (#183 and #218 )
I can only check the content of the zip file. Maybe the installer is different.

ranm, 2015-04-19:
Changing the platform.txt (in the Arduino IDE install path) doesn't have any affect on console output. 
I just get the same error and nothing changes. 
Is there another way to see what to build path is from eclipse or change it / add the IDE path into it manually?

ranm, 2015-04-19:
I guess you'll be happy to now I found the issue - CDT was not installed on this eclipse setup . 
It might sound trivial, but I think it's worth mentioning that in the install advice that the existing eclipse installation should include CDT. I didn't see it anywhere when going over the documentation. 
I was working on a PC I got with a pre-installed eclipse package and didn't think about it till now.

jantje, 2015-04-19:
I'm happy to see you found the issue.
The error log should report issues about missing components. Can you check you see those?

As to your documentation remark not stating you need CDT:
I think this page on installing the V2 plugin clearly states you need the C++ eclipse version
http://eclipse.baeyens.it/how_to.shtml#/c

Also the components page in the install menu states so
http://eclipse.baeyens.it/getting-started.shtml

It is a complex setup and that is why I advice newbies to install the product.

ranm, 2015-04-19:
I agree , although I don't think it's clear enough for the case of an existing eclipse installation. I definitely missed that. 
Also when it says "you need the c++ version"  I think the more correct form would be "you need the c++ version or you need to install the CDT plugin on you installation"
Some people (like me) use eclipse for other purposes as well - Java, python in my case . so installing a new product is not really a preferable option in that case.

jantje, 2015-04-19:
if you feel it can bee improved feel free to create a pull request. the we site is available here https://github.com/jantje/arduino-eclipse-plugin/tree/master/Eclipse%20update%20site/WebContent

wimjongman, 2015-04-20:
There is dependency on CDT in the arduino feature [1]. It should not be
able to install if CDT is not installed.

[1]
https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.feature/feature.xml#L137

wimjongman, 2015-04-20:
It looks like that still works:

[image]

jantje, 2015-04-20:
txs for the input wim :-)

ranm, 2015-04-21:
Well it did install in my case. - downloaded fresh luna (Java version) - and was able to install without a problem. 
I installed from Eclipse marketplace , is there a difference?

jantje, 2015-04-21:
I know it was possible in the old days. 
As far as I remember the market place is pointing to the last stable version which is called 2.3 but has version number 2.2
I know some things went wrong with the version numbers so I'm mostly looking at the build date which is part of the version number and filename.
I also know that the error page of the marketplace contains lots of errors about missing cdt's
http://marketplace.eclipse.org/content/arduino-eclipse-ide
I have no clue how this is shown to the user.

nirmalyakar, 2019-06-22:
I moved eclipse to a shorter path and found the issue is resolved.(Thanks to above conversations)
Working path C:\EC\eclipse.exe

Old path C:\WIP\EC\eclipse.exe where I was getting below issue.
Cannot run program "python.exe" (in directory "C:\Users\Nirmalya\workSpaceCAurdino20181227"): CreateProcess error=2, The system cannot find the file specified
failed to upload "L/RobotControllerNodeMCU/RobotControllerNodeMCU}/Release/RobotControllerNodeMCU.hex"