Sloeber/arduino-eclipse-plugin #236: OSX: Arduino 1.6.2 path change - avr-g++ not located at the correct place

jipp, 2015-04-14:
I tried the following using the plugin version 2.2.0.1 and the nightly build (2.4.0.201504090213) of the plugin.

for OSX the Arduino IDE has changed the path.
1.6.1: /Applications/Arduino.app/Contents/Resources/Java/hardware
1.6.2 and upwards: /Applications/Arduino.app/Contents/Java/hardware

setting the preferences in Eclipse:
using the plugin with OSX with 1.6.1 works without problems
using 1.6.2 and above results in "value must be an existing file" message. The version is detected successfully - but compilation is not working anymore.

I changed the preferences per project to match the new folder structure, but compilation shows the following - the path to the avr-g++ is not correct, while the path to the libraries is OK:

21:12:51 **** Incremental Build of configuration Release for project timelapse ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"/bin/avr-g++" -c -g -Os  -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=nightly -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/EEPROM" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
/bin/sh: /bin/avr-g++: No such file or directory
make: *** [.ino.cpp.o] Error 127

21:12:52 Build Finished (took 76ms)

jipp, 2015-04-14:
it might be related to the platform.txt:

1.6.1:
compiler.path={runtime.ide.path}/hardware/tools/avr/bin/

1.6.2 and upwards:
compiler.path={runtime.tools.avr-gcc.path}/bin/

looking at the output above the 'runtime.tools.avr-gcc.path' seems to be empty.

jantje, 2015-04-15:
I never tried 1.6.2 but the 1.6.3  uses
compiler.path={runtime.tools.avr-gcc.path}/bin/
as ArduinoIDE didn't define runtime.tools.avr-gcc.path it doesn't work
I changed 
compiler.path={runtime.tools.avr-gcc.path}/bin/
to
#compiler.path={runtime.tools.avr-gcc.path}/bin/
and now it works.

jantje, 2015-04-15:
I made issue https://github.com/arduino/Arduino/issues/2982 on the arduino IDE.
Just for now you can remove the compiler.path from the platform.txt
Note that sam has the same issue.

jantje, 2015-04-15:
and teensy has not

jantje, 2015-04-22:
I removed 1.6.2 and 1.6.3 from the supported list. Basically 1.6.2 and 1.6.3 can no longer be selected from the preference page

closed 2015-04-22 by jantje