Sloeber/arduino-eclipse-plugin #239: Upload doesn't autosave

sk8nfool, 2015-04-21:
I have selected auto-save (Windows->Preferences workspace Save automatically before build) which works just fine when I build the project (modified files are saved and the build performed). If I poke the Upload the Build button the project is built but the auto-save isn't done.

sk8nfool, 2015-04-21:
Update: I re-opened Eclipse, modified a file (the project's main .ccp file) and poked the "Upload the build" button. This caused an error window to open claiming a "Problem occurred". The Details say
    An error has occurred when closing the workbench. See error log for more details.
    Invalid thread access
Again with the "see error log"!
This time I started Eclipse from the command line to see if anything was being reported - there was but I don't think it was cause by the button poke.
    Apr 21, 2015 10:18:13 AM javax.jmdns.impl.DNSIncoming readAnswer
    WARNING: There was an OPT answer. Not currently handled. Option code: 65002 data:     6395EF82DB65560A

jantje, 2015-04-21:
The "OPT answer" is related to having mac devices on your network and is not related.
the save before build is no longer needed. The code does always forces a save.
The error log can bee seen in the gui by opening the error window.

I can confirm there is a problem with the upload (using build before upload) when a file is not saved. An error pops up "Problem occured"
The error windows provides following info
org.eclipse.swt.SWTException: Invalid thread access
at org.eclipse.swt.SWT.error(SWT.java:4441)
at org.eclipse.swt.SWT.error(SWT.java:4356)
at org.eclipse.swt.SWT.error(SWT.java:4327)
at org.eclipse.swt.widgets.Display.error(Display.java:1214)
at org.eclipse.swt.widgets.Display.checkDevice(Display.java:767)
at org.eclipse.swt.widgets.Display.getFocusControl(Display.java:1811)
at org.eclipse.ui.internal.WorkbenchWindow.run(WorkbenchWindow.java:2111)
at org.eclipse.ui.internal.SaveableHelper.runProgressMonitorOperation(SaveableHelper.java:291)
at org.eclipse.ui.internal.WorkbenchPage.saveAll(WorkbenchPage.java:3779)
at org.eclipse.ui.internal.WorkbenchPage.saveAllEditors(WorkbenchPage.java:3572)
at org.eclipse.ui.internal.Workbench.saveAllEditors(Workbench.java:1367)
at org.eclipse.ui.internal.Workbench.saveAllEditors(Workbench.java:1360)
at it.baeyens.arduino.actions.UploadJobHandler.run(ArduinoUploadProjectHandler.java:41)
at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)

sk8nfool, 2015-04-21:
re: auto-save - must be a version difference because for me the auto-save doesn't happen with the upload initiated build! I'm using Eclipse 3.8 and Arduino Plugin 2.4.0.201504200215 on Ubuntu 14.04. Poking Help->check for updates says I'm good.

No MACs on my network (that I know of)! And why is the net being used at all?

Now that you've taught me to how to view the error log I can send you a copy. It is very similar to what you cited (some line number differences).

jantje, 2015-04-21:
>re: auto-save - must be a version difference

auto-save is only available in the latest version. So we are on the same version. I think we also have the same behaviour.

>No MACs on my network (that I know of)! And why is the net being used at all?

Yun

jantje, 2015-04-22:
fixed

closed 2015-04-22 by jantje