Sloeber/arduino-eclipse-plugin #241: Extra blank lines in Serial Monitor View

sk8nfool, 2015-04-21:
There are extra blank lines in the output displayed by the Serial Monitor viewer from the following program
// Do not remove the include below
#include "stutter.h"

//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
	Serial.begin(9600);
	while (!Serial)
		;   // wait until ready
	Serial.println("\n\nStutter Print sample");
	Serial.flush(); // wait for all bytes to be transmitted to the Serial Monitor
}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
	while (1)
	{
		unsigned long int t = millis();

		delay(1);

		Serial.println("ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE " + String(millis()-t));

		delay(500);
	}
}

It seems to be a little sensitive to timing but I can't swear to that.
The output in the Serial Monitor looks something like this

Stutter Print sample

ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1

ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1

ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1

ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1
ABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDEFGHIJabcdefghijABCDE 1

jantje, 2015-04-21:
Indeed there are.
I have no clue on the root cause.

sk8nfool, 2015-04-21:
Hey - thanks for the quick responses by the way. Most helpful.

I tried closing the Serial Monitor Viewer and "cat /dev/ttyACM0" to see if the extra lines are really coming from the Arduino (UNO in my case) but it returns immediately.

Is there some kind of protocol running on the serial link?

jantje, 2015-04-21:
There is nothing special as far as I know. I'm not sure can do this.
maybe tail -f can
It is better to use a real serial monitor. But you'll see arduino IDE does not have the blank line issue.

sk8nfool, 2015-04-22:
To verify if the extra lines are actually coming up the serial port and to see if some kind of protocol is being used on the channel I wrote a little tty tapper that sat between the real tty and the master side of a pseudo-terminal. something along the lines of

uno ---> tapper ---> eclipse/monitor viewer

where the tapper reads (writes) /dev/ttyACM0 and copies the data to/from a pseudo tty.

The tapper works without eclipse but the serial monitor viewer won't let me open the pseudo-terminal. ARRGH.

So I fired up strace on eclipse and captured the reads on /dev/ttyACM0. Data looks to be clean (no extra characters). Assuming I didn't screw up the strace it looks like the serial monitor is hiccuping.

FredericG-BE, 2015-06-03:
Not sure if this is helpful in any way, I am seeing exactly the same thing on a Windows machine... There are always extra newlines where newlines are expected, but were it happens seems to be random.

jantje, 2015-06-03:
@FredericG-BE 
Txs for the input. It seems like a non-os related bug in the plugin introducing blank lines.

jantje, 2016-02-22:
I found it was an issue when the serial data was split between the \r and \n.

sk8nfool, 2016-02-22:
Excellent - thanks

jantje, 2016-03-09:
confirmed

closed 2016-03-09 by jantje