Sloeber/arduino-eclipse-plugin #242: Message: "Considering about Arduino IDE compatibility" always appears after setting IDE

jipp, 2015-04-27:
Hi

using Luna or Kepler on the OSX and setting the Arduino path always results in the compatibility message:

"You are using a version of the Arduino IDE that is newer than available at the release of this plugin."

Using v2.x and nightly builds.

thanks
Wolfgang

jipp, 2015-04-27:
I'm using 1.6.1 of the Arduino IDE. Looks like in ArduinoPreferencePage.java this part needs some adoption:

	if (mArduinoIdeVersion.getStringValue().compareTo("1.6.0") > 0) {
	    if (!showError("You are using a version of the Arduino IDE that is newer than available at the release of this plugin."))
		return false;
	}

jantje, 2015-04-27:
Indeed
Txs for the input. I'm rewriting the section for better maintenance.

jantje, 2015-04-27:
The fix should be available tomorrow
Please close if you think it is satisfactory

jipp, 2015-04-28:
Hi

I tested the fix - so far so good. Now the messages make more sense - thx.
Nevertheless I have a comment. 

-when using 1.6.1 I think its not needed to give the info that this is the currently recommended version
-also I get this hint 2x in the window 

This is what I get:
Arduino IDE 1.6.1 is the currently advised version for all platforms.
Remember to add your own make as it is no longer delivered with arduino.
Advised version is 1.6.1
Please see <http://eclipse.baeyens.it/installAdvice.shtml> for more info.

thanks for the effort

Wolfgang

jipp, 2015-04-28:
maybe it would be enough to comment out line 173:
	    infoMessage = "Arduino IDE " + mArduinoIdeVersion.getStringValue() + " is the currently advised version for all platforms.";

to avoid the double message

jantje, 2015-04-28:
Yes it should. I'll change it to
infoMessage = "Arduino IDE " + mArduinoIdeVersion.getStringValue() + "works great."

jipp, 2015-04-29:
works like expected and described, so OK for me

thx

closed 2015-04-29 by jipp

jantje, 2015-04-29:
ok txs