Sloeber/arduino-eclipse-plugin #243: .settings\org.eclipse.cdt.core.prefs file gets tainted every time the project is opened again

dniklaus, 2015-04-27:
With the EclipseArduino bundle, nightly build version 2015-04-24_02-06-01 (after switching from 2015-04-09_02-06-59), I have now an issue with the following two lines changing on every project re-open action:

    environment/project/it.baeyens.arduino.core.toolChain.release.256903333/A.EXTRA.TIME.LOCAL/value=1430156500
    environment/project/it.baeyens.arduino.core.toolChain.release.256903333/A.EXTRA.TIME.UTC/value=1430149300

This brings in some sort of discomfort as long as the project is under version control.

Why has this been changed?

jantje, 2015-04-27:
This has to do with Teensy. Teensy needs these linking defines that are basically current date and time.
See here for more detail 
https://forum.pjrc.com/threads/27740-Arduino-1-6-0-any-plans-to-support-it?p=64812&viewfull=1#post64812

dniklaus, 2015-04-27:
Sad story... Maybe you want to just set these to 0, as Paul Stoffregen suggests to do beside other options. And/or make the usage of the feature switchable...

jantje, 2015-04-27:
Using 0 is not an option. Teensy would not work properly and I plan to use this feature as I already mentioned in the thread mentioned above.
Adding a switch may be an option. Default on.
Maybe there are already defines that can be used to make this static (my preferred option)
Anyway,I accept the issue as being limiting for version control (even forcing manual merges) but it is not high on my to do list.
Jantje

dniklaus, 2015-04-29:
Ok, thank you!

jantje, 2018-03-03:
Now the question is.... what is the easiest way to test  this.
Note that for this to work you can not use environment expansion for makefiles otherwise the same problem still exists

uzi18, 2018-03-03:
Copy before use, to backup file

jantje, 2018-03-04:
@uzi18  rofl I was only joking :-)

jantje, 2018-03-04:
I just tested on my working workspace and it worked. I fear it did introduce another issue so do not update unless you really need this

closed 2018-03-04 by jantje