Sloeber/arduino-eclipse-plugin #244: Error compilation on OpenSuse (Arduino IDE 1.6.1, Arduino Eclipse 2.2)

slava-viktorov, 2015-04-28:
OpenSuse 13.2

Arduino Eclipse console:
23:33:24 **** Incremental Build of configuration Release for project 1 ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"/home/slava/Загрузки/arduino-1.6.1/hardware/tools/avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=161 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR     -I"/home/slava/Загрузки/arduino-1.6.1/hardware/arduino/avr/cores/arduino" -I"/home/slava/Загрузки/arduino-1.6.1/hardware/arduino/avr/variants/eightanaloginputs" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
../.ino.cpp:8:2: error: #error the file: Blink.ino is not found in the indexer though it exists on the file system.
 #error the file: Blink.ino is not found in the indexer though it exists on the file system.
  ^
../.ino.cpp:9:2: error: #error this is probably due to a bad eclipse configuration : ino and pde are not marked as c++ file.
 #error this is probably due to a bad eclipse configuration : ino and pde are not marked as c++ file.
  ^
../.ino.cpp:10:2: error: #error please check wether *.ino and *.pde are marked as C++ source code in windows->preferences->C/C++->file types.
 #error please check wether *.ino and *.pde are marked as C++ source code in windows->preferences->C/C++->file types.
  ^
make: *** [.ino.cpp.o] Ошибка 1
subdir.mk:18: ошибка выполнения рецепта для цели «.ino.cpp.o»

23:33:25 Build Finished (took 141ms)

jantje, 2015-04-28:
please read the error message and act appropriately

closed 2015-04-28 by jantje

enricopava, 2016-01-23:
It is not possible to act appropriately as the error message says because you can't mark *.ino and *.pde as C++ source code in windows->preferences->C/C++->file types!

You must go to:
   Project->Properties->C/C++ Generals->File types,
   select "Use project settings" and add .ino and *.pde as C++ source code pressing on "New Button".
[image]