Sloeber/arduino-eclipse-plugin #245: Arduino plugin for eclipse having trouble finding avr-g++

skyviper, 2015-04-29:
I installed plug-in V2.2 for windows and having trouble compiling. I have seen similar posts for older versions but no real solution to the issue. Here is what I see:
19:10:29 **** Incremental Build of configuration Release for project My_Sketch ****
make all 
Building file: ../My_Sketch.cpp
Starting C++ compile
"/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega168 -DF_CPU=16000000L -DARDUINO=163 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR     -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" -MMD -MP -MF"My_Sketch.cpp.d" -MT"My_Sketch.cpp.d" -D__IN_ECLIPSE__=1 -x c++ "../My_Sketch.cpp"  -o  "My_Sketch.cpp.o"
/usr/bin/sh: /bin/avr-g++: No such file or directory
make: *** [My_Sketch.cpp.o] Error 127

19:10:29 Build Finished (took 218ms)

The Eclipse log is complaining about conflicts. See below:

!SESSION 2015-04-28 19:10:12.727 -----------------------------------------------
eclipse.buildId=4.3.0.M20130911-1000
java.version=1.7.0_55
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86 -product org.eclipse.epp.package.cpp.product

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:10:16.799
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.cdt.core 4 0 2015-04-28 19:10:18.765
!MESSAGE Error: Cannot run program "\bin\avr-gcc": Launching failed
!STACK 0
java.io.IOException: Cannot run program "\bin\avr-gcc": Launching failed
   at org.eclipse.cdt.utils.spawner.Spawner.exec(Spawner.java:347)
   at org.eclipse.cdt.utils.spawner.Spawner.<init>(Spawner.java:89)
   at org.eclipse.cdt.utils.spawner.ProcessFactory.exec(ProcessFactory.java:89)
   at org.eclipse.cdt.core.CommandLauncher.execute(CommandLauncher.java:194)
   at org.eclipse.cdt.internal.core.BuildRunnerHelper.build(BuildRunnerHelper.java:238)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runProgramForLanguage(AbstractBuiltinSpecsDetector.java:704)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForLanguage(AbstractBuiltinSpecsDetector.java:634)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForEachLanguage(AbstractBuiltinSpecsDetector.java:507)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector$1.runInWorkspace(AbstractBuiltinSpecsDetector.java:441)
   at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:38)
   at org.eclipse.core.internal.jobs.Worker.run(Worker.java:53)

!ENTRY org.eclipse.cdt.core 4 0 2015-04-28 19:10:18.858
!MESSAGE Error: Cannot run program "\bin\avr-g++": Launching failed
!STACK 0
java.io.IOException: Cannot run program "\bin\avr-g++": Launching failed
   at org.eclipse.cdt.utils.spawner.Spawner.exec(Spawner.java:347)
   at org.eclipse.cdt.utils.spawner.Spawner.<init>(Spawner.java:89)
   at org.eclipse.cdt.utils.spawner.ProcessFactory.exec(ProcessFactory.java:89)
   at org.eclipse.cdt.core.CommandLauncher.execute(CommandLauncher.java:194)
   at org.eclipse.cdt.internal.core.BuildRunnerHelper.build(BuildRunnerHelper.java:238)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runProgramForLanguage(AbstractBuiltinSpecsDetector.java:704)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForLanguage(AbstractBuiltinSpecsDetector.java:634)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForEachLanguage(AbstractBuiltinSpecsDetector.java:507)
   at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector$1.runInWorkspace(AbstractBuiltinSpecsDetector.java:441)
   at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:38)
   at org.eclipse.core.internal.jobs.Worker.run(Worker.java:53)

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:10:19.654
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.egit.ui 2 0 2015-04-28 19:10:20.122
!MESSAGE Warning: EGit couldn't detect the installation path "gitPrefix" of native Git. Hence EGit can't respect system level
Git settings which might be configured in ${gitPrefix}/etc/gitconfig under the native Git installation directory.
The most important of these settings is core.autocrlf. Git for Windows by default sets this parameter to true in
this system level configuration. The Git installation location can be configured on the
Team > Git > Configuration preference page's 'System Settings' tab.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.

!ENTRY org.eclipse.egit.ui 2 0 2015-04-28 19:10:20.122
!MESSAGE Warning: The environment variable HOME is not set. The following directory will be used to store the Git
user global configuration and to define the default location to store repositories: 'C:\Users\Yourarmsoff'. If this is
not correct please set the HOME environment variable and restart Eclipse. Otherwise Git for Windows and
EGit might behave differently since they see different configuration options.
This warning can be switched off on the Team > Git > Confirmations and Warnings preference page.

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:10:24.256
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler@1765f2b} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:10:27.563
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler@1765f2b} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:17:36.611
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler@1765f2b} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:17:38.795
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler@1765f2b} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:17:38.841
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler@1765f2b} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:17:39.855
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler@1765f2b} vs {it.baeyens.arduino.actions.BuildHandler}

!ENTRY org.eclipse.ui 4 4 2015-04-28 19:17:40.058
!MESSAGE Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler@1765f2b} vs {it.baeyens.arduino.actions.BuildHandler}

Can you tell me what is causing this issue?

jipp, 2015-04-29:
Hi

which Arduino IDE are you using?

I ran into a similar one:
https://github.com/jantje/arduino-eclipse-plugin/issues/236

reference to Adruino issues:
https://github.com/arduino/Arduino/issues/2982

cheers
Wolfgang

jantje, 2015-04-29:
@skyviper jipp is right. it is a 1.6.2/3 issue. 1.6.2 and 1.6.3 are not supported by the plugin.

closed 2015-04-29 by jantje