Sloeber/arduino-eclipse-plugin #246: Add perspective for Arduino development

riban-bw, 2015-05-04:
It would be advantageous to have a separate perspective for Arduino development. The extra tool bar buttons are not required for non-Arduino development. I tried to save the C/C++ perspective as Arduino perspective but the Arduino toolbar buttons become unavailable after closing and opening the new perspective!

jantje, 2015-05-04:
There is an arduino perspective avalable. Which version are you using?

riban-bw, 2015-05-04:
I am using Arduino eclipse plugin version 2.2.0.1 installed from the Eclipse Marketplace in Eclipse 4.4.2 on Windows 7 and Ubuntu 15.04.

jantje, 2015-05-04:
This version is getting so old I can't recall the perspective is already available or not. 
The Arduino perspective is available in the nightly build.
 As only 4 people donated to make a  new release ....

closed 2015-05-04 by jantje