Sloeber/arduino-eclipse-plugin #250: Switching perspective loses Arduino tools

riban-bw, 2015-05-04:
I have installed plugin version 2.4.0.201504290213 and removed the Arduino toolbar from the C/C++ perspective. (I don't know why they were there?) Switching between Arduino and C/C++ perspectives, the Arduino toolbar fails to show in the Arduino perspective. Using Eclipse 4,4,2,

jantje, 2015-05-05:
I'm not surprised about this but @wimjongman may have an explanation for this.
I'm not surprised because the Arduino eclipse plugin is a layer on top of CDT which is a layer on gcc. Similar to how the Arduino IDE is a layer on gcc. This plus the complexity of having the plugin as part of a project and a standalone plugin makes things at the least fussy.

wimjongman, 2015-05-05:
Try resetting the arduino perspective. Right click on the icon in the perspective bar en choose reset.

jantje, 2015-05-05:
txs wim :-)

riban-bw, 2015-05-05:
Resetting perspective does not restore the Arduino toolbar for me :-(.

jantje, 2015-05-05:
I think the toolbar is in the c/c++ perspective because the arduino perspective is not part of the plugin but only of the product.
As you have been able to remove the toolbar from the C/C++ perspective could you try to to add the arduino toolbar to the arduino perspective as a work around?

riban-bw, 2015-05-05:
I can configure Arduino perspective to show the Arduino toolbar.
If I close all perspectives, the Arduino toolbar remains visible. If I open the C/C++ perspecitve (which does not have the Arduino toolbar enabled), the Arduino toolbar disappears. I can then close the C/C++ perspective and open the Arduino perspective but the toolbar does not reappear. To get it back I must reconfigure perspective to remove toolbar, The reconfigure to add toolbar. A restart of Eclipse also restores the toolbar. It is as if the toolbar is not bound to the Arduino perspective and so is not being (properly) shown or hidden. My current work-around is to only use one perspective at a time. This is okay when working on small projects like a single Arduino sketch but may become irksome if working on a larger project with Arduino sketches and desktop software.

jantje, 2015-05-05:
I just tested this.
In the c/c++ perspective I selected windows->customize perspective and removed the arduino toolbar
Then I went to the arduino perspective and the arduino toolbar is gone.
I selected windows->customize perspective and the toolbar is selected. :-(
I kept is selected. I switched a couple of times between C/C++ lua/ jazz admistration/ debug and arduino and then magincally it started to work.

jantje, 2016-03-25:
I think this one is fixed in eclipse Mars

jantje, 2016-05-07:
fixed by mars

closed 2016-05-07 by jantje