Sloeber/arduino-eclipse-plugin #251: avrdude pop-up shows "no name provided"

riban-bw, 2015-05-05:
During upload of firmware to Arduino, the pop-up shows the text, "no name provided". What does this mean? Is there a way to specify what text is shown?

jantje, 2015-05-05:
can you show me an image?

riban-bw, 2015-05-05:
Screenshot attached.
[image]

jantje, 2015-05-05:
Och I see
That is probably something in the code that was not assigned a value.
I don't think there is a way to change the text without coding.

riban-bw, 2015-05-21:
UploadSketchWrapper.java:155 looks up an build environmental variable which is empty so uses default text, "no name provided". I can't figure out what this line of code is trying to do. Replacing it with:
```
monitor.beginTask("Uploading \"" + myProject.getName() + "\"", 2);
```
gives a more useful message in the dialog.

jantje, 2015-06-09:
should be fixed as requested. please reopen if fix doesn't work.

closed 2015-06-09 by jantje