Sloeber/arduino-eclipse-plugin #252: Can't adjust project (Arduino) properties

riban-bw, 2015-05-05:
I have a project that I have been working on for a few days. Today I selected Project->Properties->Arduino and was presented with an error, "Could Not Accept Changes", "The currently displayed page contains invalid values." The only thing I can remember changing is the USB serial interface. I have tried setting com port value in the .prefs file with no luck. (I think that may be a red herring.) I have updated to the latest nightly (2.4.0.201505050212) but no luck. I have restarted Eclipse (a few times). Any ideas how to rescue my project?

riban-bw, 2015-05-05:
The project won't build. There is something very wrong yet I can't remember doing anything to break it!

jantje, 2015-05-05:
I have seen the "The currently displayed page contains invalid values." error a couple of times. I have made some bug fixes lately that improved the situation. Seems there is still an issue somewhere.
I noticed that when I closed eclipse and reopened it works fine. Not sure why that won't work for you. Maybe poking around in the . pref file caused that. (I never poke around in eclipse files directly)
What I would do in your case is create a new project in the same workspace and drag and drop your code from the "bad project" to the "new project". Make sure you get a copy of the files and not a link.
Continue working from the new project.
As you are on the latest version you should be able to copy the libraries folder over from the "bad project" as well. Selecting the windows->reattach libraries should fix the build path in the new project. (never tried it though)

riban-bw, 2015-05-05:
Ah! Security update for .NET (KB3037581) applied today. I wonder if that has caused problems. I am still able to compile C++ programs using cygwin or mingw compiler.

riban-bw, 2015-05-06:
I did try to create a new project but that isn't working either!
1. New Arduino sketch.
2. Enter name
3. Wizard won't go any further. Stuck at Project Name dialog.

jantje, 2015-05-06:
something seems seriously wrong on your system.
Do you feel brave enough to do a debugging session?

riban-bw, 2015-05-06:
I am brave but am working on a company laptop without admin rights so have limited functionality, e.g. I can not run application installers. (I am using Eclipse Portable.)

jantje, 2015-05-06:
There is no need to install anything. It would be easier if you could install git but it is not needed.
Basically if you download the tarbal of the project (=workspace) https://github.com/jantje/arduino-eclipse-plugin/archive/master.zip and open the workspace with eclipse (a java development version) you should be able to debug and find what is causing the errors.

riban-bw, 2015-05-06:
Hmm! I might have fixed it (though not sure how)! I downloaded the Product and created a new workspace. This allowed me to create a new project which I imported the files from the old project. I mad a couple of code tweaks* and saved this project. I was then able to open the same project from my original copy of Eclipse with your plugin and things were working. I can't explain why so much was not working before, even when I removed the project or created a new workspace. It would be useful to figure out what went wrong. I will play a bit more to see whether I can reproduce the problem and debug it. If I haven't managed within a couple of days, I will close this issue as resolved.
*The project I was using had a main.cpp file which is not part of the current wizard. I can't see that would cause the problems I experienced.

jantje, 2015-05-06:
txs for the feedback

riban-bw, 2015-05-09:
I haven't had time to investigate further. It seems okay now so I will close this issue and if there is recurrence, open another. Cheers!

riban-bw, 2015-05-13:
This has recurred. I don't understand your instructions on debugging. I downloaded master.zip and opened it as a workspace. How do I now debug what is happening with eclipse / plugin when I try to use the plugin, e.g. create a new project?

riban-bw, 2015-05-13:
Okay, I have a debug version of the application working. Clicking on "New Sketch" toolbar yields following debug output:

!ENTRY org.eclipse.ui 4 0 2015-05-13 20:37:21.184
!MESSAGE Unhandled event loop exception
!STACK 0
org.eclipse.e4.core.di.InjectionException: java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.getSerialPortNames()[Ljava/lang/String;
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:62)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:247)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:229)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:132)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:149)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:499)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:508)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:210)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:825)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.handleWidgetSelection(HandledContributionItem.java:701)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.access$6(HandledContributionItem.java:685)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem$4.handleEvent(HandledContributionItem.java:613)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4353)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1061)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4172)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3761)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$9.run(PartRenderingEngine.java:1151)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:332)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1032)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:148)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:636)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:332)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:579)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:150)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:135)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:380)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:235)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:648)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:603)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1465)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1438)
Caused by: java.lang.UnsatisfiedLinkError: jssc.SerialNativeInterface.getSerialPortNames()[Ljava/lang/String;
	at jssc.SerialNativeInterface.getSerialPortNames(Native Method)
	at jssc.SerialPortList.getWindowsPortNames(SerialPortList.java:305)
	at jssc.SerialPortList.getPortNames(SerialPortList.java:294)
	at jssc.SerialPortList.getPortNames(SerialPortList.java:178)
	at it.baeyens.arduino.arduino.Serial.list(Serial.java:70)
	at it.baeyens.arduino.common.Common.listComPorts(Common.java:449)
	at it.baeyens.arduino.ui.ArduinoSelectionPage.draw(ArduinoSelectionPage.java:240)
	at it.baeyens.arduino.ui.NewArduinoSketchWizardBoardPage.createControl(NewArduinoSketchWizardBoardPage.java:49)
	at org.eclipse.jface.wizard.Wizard.createPageControls(Wizard.java:178)
	at org.eclipse.jface.wizard.WizardDialog.createPageControls(WizardDialog.java:759)
	at org.eclipse.jface.wizard.WizardDialog.createContents(WizardDialog.java:622)
	at org.eclipse.jface.window.Window.create(Window.java:435)
	at org.eclipse.jface.dialogs.Dialog.create(Dialog.java:1101)
	at org.eclipse.jface.window.Window.open(Window.java:797)
	at it.baeyens.arduino.actions.NewSketchHandler.execute(NewSketchHandler.java:22)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:294)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:90)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	... 38 more

jantje, 2015-05-13:
It looks like jssc has a problem on your system.
Do you have multiple jssc's installed?
Do you have a special usb device that may cause this problem (when connected)?

riban-bw, 2015-05-13:
Same result with or without USB devices plugged. I can't find jssc installation. Any idea where to look?

jantje, 2015-05-13:
Sorry for the confusion but this is out of my comfort zone.
jssc seems to be part of the arduio eclipse plugin at https://github.com/jantje/arduino-eclipse-plugin/tree/master/it.baeyens.arduino.common/lib
if it is on your system it should be as a jssc.jar file.
It seems version 2.8.0 (from jan 2014) is the latest version. This version is used by the plugin.

I'm a bit confused for another reason.
Initially there was no error logged. However a stack trace as you provided above should be logged in the error view. So it may be that something is wrong with the debugging environment.
How did you setup the debug configuration?

riban-bw, 2015-05-13:
To debug, I started Eclipse, imported the plugin workspace to a new workspace (because the downloaded workspace did not show any files), debug run the application which creates a new workspace, close debug application, copied my DUT workspace to the newly created workspace, debug run the application again, clicked on "New Sketch" toolbar, copied output from console window.

I have now figured out what the issue is. jssc.SerialNativeInterface.getSerialPortNames() extracts dll files to the directory [user.home]/.jssc/windows. On my laptop [user.home] is a network share that is mirrored locally. The .jssc folder was read-only hence jssc was failing. I'm not sure why this is the case and I am experimenting with the online / offline / sync features - it seems this file changes to read-onl when I disconnect from my company network. A work-around may be to set the Java user.home variable for Eclipse. I am struggling to figure out how to do that. It should be possible to set this in the configuration file but my attempts have failed.

riban-bw, 2015-05-14:
I have implemented a work-around by adding -Duser.home=D:\bin\EclipsePortable\Data\user.home to the eclipse.ini file.
I suggest the plugin is updated to avoid failing if com ports can not be detected but instead maybe warn on the page that allows com port selection. The current implementation stops the plugin from working if jssc.SerialNativeInterface.getSerialPortNames() fails which is extreme. Maybe wrapping calls to getSerialPortNames() in a try / catch exception handler?
I appreciate my workflow may not be common but it demonstrates an issue that (maybe) could be better handled.
Cheers

jantje, 2015-05-14:
Txs very much for the debugging and finding the root cause. :-)
I agree this is a special case and I also agree it should be "better handled". The try catch exception handler is probably the best way to go.

Just thinking. If the folder write rights is the root cause .... and the plugin and arduino ide use the same jss (of which I am pretty sure)... and arduino ide does not set the user.home variable (which I assume) ... the problem should be reproducible in the arduino IDE as well.
Could you do a test on this?
If it doesn't fail there I'll have a look in the Arduino IDE code how they did it (The code in the plugin is a old copied from the Arduino IDE).

jantje, 2015-05-14:
Thinking again. In the Arduino IDE it probably shows as an empty port list only.

riban-bw, 2015-05-14:
Arduino IDE does not appear to extract DLLs to same folder. (I can't figure out where it does.)
If you want to test any solution you implement, you could replace jSSC-2.8_x86.dll with a different (non DLL) file with same name. This won't give the same error but will error (initialising DLL) so should allow testing workflow. I think the error probably did appear in the Error Log but that is not part of the Arduino perspective so I would not have noticed. (I am unable to test now because the folder is not going read-only!) I do see the error message in the Error Log if I change the DLL fiile to a non-DLL so that the plugin cannot load it.

jantje, 2015-05-14:
you're assuming I'm using windows ;-) I'm not :-)

jantje, 2015-05-14:
I seem to be unable to reproduce in linux
Can you give it a test tomorrow?

riban-bw, 2015-05-16:
Excellent that you not using Windows :-). I prefer not to but am constrained at times. It is easier to test on *nix by changing the permission of ~/.jssc/llinux (on a linux machine - not sure what on OS X). My Windows laptop is at the clinic being fixed so I am using my Ubuntu laptop. I can reproduce by making the directory not accessible to the user but there was no nightly build this morning so your fix isn't available via normal update mechanism. I will test on both machines (Ubuntu & Windows 7) as soon as I can. Cheers!

jantje, 2015-05-16:
I fixed this 2 days ago so the change is in the 2015_05_15 version.

riban-bw, 2015-05-16:
Sorry - that didn't work :-(. I grabbed the latest version of the plugin from github and it still bombs out at the same point. (Testing on Ubuntu.) Maybe the try / catch wrapper does not catch exceptions thrown within the jssc.jar which should be handled by jssc. Maybe jssc may not catch errors when it calls jssc.SerialNativeInterface.openPort. I am just having a look at the jssc code to see whether this needs fixing there.

riban-bw, 2015-05-16:
jssc checks whether the user.home folder has write permission and if so, uses it else uses the temp folder however this does not cover (the corner case) [home]/.jssc exists but is read-only. jssc attempts to extract the native serial libs with error checking so does not throw an error when it is initiated and tries to extract this native lib. Subsequent use of the library assumes this lib exists. Is there a way to check whether jssc initiates correctly and avoid using it if it does not?

riban-bw, 2015-05-16:
I have reviewed jssc source and see that there is no indication available as to whether native library has been loaded therefore there seems no simple fix / workaround within this plugin. I have reported this issue to jssc project thus: https://github.com/scream3r/java-simple-serial-connector/issues/72.

riban-bw, 2015-05-16:
WAIT!!! Change Serial.java:73 from
`} catch (Exception e) {`
to
`} catch (Error e) {`
and your code will work. However, the result is an error popup which is a bit worrying that the program has had a fatal error (which it hasn't) and the serial port listbox is populated with "config error" rather than being empty. I would suggest not showing the error popup but instead changing the form that shows the com port selector to show a warning, maybe with a link to this issue (like you have in the error message). This is less alarming and interrupts workflow less. Anyway, keep the fix (with possible enhancements) but catch error instead of (or as well as) exception. (FYI: This is because the exception raised if of type Error rather than type Exception. There is a third type called RuntimeException.)

jantje, 2015-05-16:
The code is currently made with the idea that errors are shown in a dialogbox to the end users.
An error is anything that can not be worked around by the plugin.
Warning are only logged in the "error log"
I think it is kind of cool that the list box contains the error info.
As this is a very rare case I don't want to spend to much effort on it as well.
So it looks right to me.
If you think you can improve feel free to make a pull request.

riban-bw, 2015-05-16:
I will test on Windows and Linux and close issue if now resolved in nightly / git. Cheers!

riban-bw, 2015-05-17:
Tested on Ubuntu okay. Can't test on Windows - my laptop is broken. I am satisfied this is likely to fix the problem so closing the issue. Thanks for your work on this.

closed 2015-05-17 by riban-bw