Sloeber/arduino-eclipse-plugin #253: Using programmer "USBTinyISP"

karunadheera, 2015-05-11:
Hi there,
Cannot continue in project settings without selection of a Serial Port. USBTinyISP has no serial port because it is not a USB to Serial converter. I have this working fine on Arduino IDE. I really like when my favourite comprehensive IDE (Eclipse) can be used to write Arduino Code.

How to make this plugin work with USBTinyISP?

Thanks,
PK.

[image]

jantje, 2015-05-11:
Just fill in "something".
It should work fine

jantje, 2015-05-14:
did that fix it?

riban-bw, 2015-05-31:
This issue is also described in issue #264 and fixed by pull request #268 so should be able to close this.

jantje, 2015-05-31:
@riban-bw 
indeed

closed 2015-05-31 by jantje