Sloeber/arduino-eclipse-plugin #254: Plugin generates incorrect ARDUINO macro value for 1.6.X versions of IDE

george-hawkins, 2015-05-14:
With the switch to 1.6.X the Arduino developers changed the format of the `ARDUINO` macro.

Previously e.g. 1.5.8 was mapped to 158, i.e. one digit per component of the version number, but this meant they couldn't easily encode e.g. 1.5.10.

So they moved to two digits for the minor and patch components of the version number, so 1.6.0 became 10600 (the current committed revision is 10605).

You can see this switch in format in commit [`b9e186e`](https://github.com/arduino/Arduino/commit/b9e186e45aa281b6441bd25e0733c5668adb2b41#diff-46c56e55e554cf2f19c1bad072dabd2a).

Here in `BaseNoGui.java` you can see they define `REVISION`, this is picked up by [`PreferencesData.java`](https://github.com/arduino/Arduino/blob/9b58812d1dbe93e110d6c1a62377e2923e35132d/arduino-core/src/processing/app/PreferencesData.java#L59) that writes it out to the IDE preferences as `runtime.ide.version` which in turn is picked up by the [`platform.txt`](https://github.com/arduino/Arduino/blob/4d029de000e2c696026ba8be074d395f7310daa5/hardware/arduino/avr/platform.txt#L56) files as the value used for the `ARDUINO` macro.

This all means that at the moment when I use the 1.6.1 Arduino IDE the IDE specifies `-DARDUINO=10601` while the Eclipse plugin specifies a different value, i.e. `-DARDUINO=161`.

jantje, 2015-05-14:
Would life not be easier if the value was simply put in the platform.txt file?
I'm not using these defines but someone will sooner or later. So I'll mod it

george-hawkins, 2015-05-14:
It might indeed be simpler if things were setup as you suggest.

The `ARDUINO` macro is quite important, even if you're not using it directly. It's value is checked by various include files in the standard Arduino libraries and by lot's of third party libraries.

PS thanks for all your work on the Arduino plugin for Eclipse. It's a super product - I'd go mad without it :smiley:

jantje, 2015-05-16:
Did the fix fix it?

george-hawkins, 2015-05-17:
Cool - I now see `-DARDUINO=10601` :smiley:

closed 2015-05-17 by george-hawkins

jantje, 2015-05-18:
txs for testin