Sloeber/arduino-eclipse-plugin #255: Com port should not be associated with project

riban-bw, 2015-05-16:
The com port used for firmware upload and serial console is associated with the project. I think it would be better if this were a global setting. On some platforms, the com port changes regularly, e.g. USB serial ports change identifier depending on how they are enumerated by the OS USB driver. This is an element of the plugin that is likely to affect all projects and should be extracted to the global settings (and preferably added to the menu and toolbar). This would also reduce the risk of issues such as https://github.com/jantje/arduino-eclipse-plugin/issues/252 where issues with the com port stop much of the plugin from working. Separating the com port handling may also allow more dynamic handling of the com port.

jantje, 2015-05-16:
I miss your point or I disagree ;-)

First of all the comport of the project is not linked to the serial monitor. When you open the serial monitor you select the com port to use. The listing there is the same as the listing you get in the new sketch wizard and the project properties. The only difference is that in the serial port you can not "type" the serial port.

Secondly if you are working on a multi arduino project (I mean multiple projects at the same time -like a client arduino project and a server arduino project) having a com port per project is the way to go.

I know teensy and other boards do not use the com port but then you fill in some garbage and all is well.
In these cases having a com port linked to the project is no "draw back"

I'm also more and more in yun where the com port is fixed (it is the computer name of the linux on yun). Again a port per project is easy (not needing to swap the port when swapping project) and avoids errors.

Issue #252 is a completely different nature. Basically there is a system problem/bug and all com port activity has become impossible making the plugin basically useless. Having a port selection at the global level or at the project level would hardly make a difference in functionality available when 252 manifests itself.

Having a option for a project wide versus a workspace wide com port would mean lots of change. I doubt I would accept a pull request for this because I think it would also be confusing to the end user.

There is a long lasting proposal on the remind page for a change called "changing the port on the toolbar". The way I see this is that you see the com port of the current selected com project and you can change the com port for current selected project by selecting another com port from a combo edit box.
The advantage would be that you do not need to build as you know only the com port has changed.
The change has 4 votes untill now. Not really a driver for change for me.

riban-bw, 2015-05-16:
I understand what you say. I am finding it rather irksome to change the port setting for every project whenever the serial port changes id. I will close this feature request (on the basis that it is unlikely to be implemented) and add my opinion to adding serial port to toolbar. Cheers!

closed 2015-05-16 by riban-bw

scottmberry, 2020-10-20:
I was just searching for an option to at least easily set the com port or have an "automatic" option like VSCode does with projectio. Using a VM the com port constantly changes and I also change which Arduino types I am programming with the same project. Having these options handy right away in the IDE would be so helpful.

jantje, 2020-10-21:
> an "automatic" option like VSCode does with projectio. 

I don't know what this is nor how it works. If you can elaborate please do so in a new issue.