Sloeber/arduino-eclipse-plugin #256: Invalid library error

maxritter, 2015-05-16:
Hey,

in the newer versions of your Eclipse Arduino plugin, I receive the following error when building my Teensy 3.1 project under Arduino 1.6.3:

"The library {Library name} is not valid for this board".

Whats the criteria for this error ? The libraries affected are modified or created by myself, but they work fine. When I use an older version of your plugin (currently 09.04.2015), the same project compiles fine. 

I would be glad if you can tell me how I can avoid this issue so that I can use the newest versions of your greate plugin again.

Thanks,
Max

jantje, 2015-05-16:
If it is your own developed library it should be in your private library folder and it should work fine.

maxritter, 2015-05-17:
Thanks, that fixed it !

closed 2015-05-17 by maxritter

tomtasjohan, 2015-06-13:
Same issue here. Is the "private library" folder not the Library folder in my workspace project?

[image]

mikaeltulldahl, 2016-03-10:
I got the same error using this library [UartEvent ](https://github.com/duff2013/UartEvent ) on a Teensy 3.1. the fix Jantje mentioned works, to move the library to the project folder. However I'm still curious what triggers the error

jantje, 2016-03-10:
this issue has been closed. Please open your own issue.