Sloeber/arduino-eclipse-plugin #257: crash when add serial monitor

braindevices, 2015-05-17:
installed directly from eclipse market, arduino IDE 1.5.8 (running fine)
rxtx test was successful. 

    $ dpkg -s librxtx-java
    Package: librxtx-java
    Status: install ok installed
    Priority: optional
    Section: java
    Installed-Size: 928
    Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
    Architecture: amd64
    Source: rxtx
    Version: 2.2pre2-10
    Provides: lib-rxtx-java
    Depends: libc6 (>= 2.4)
    Conflicts: lib-rxtx-java

However during test there will be:

    java.lang.UnsatisfiedLinkError: Can't load library: ~/opt/arduino-1.5.8/lib/librxtxSerial64.so

When adding serial monitor:
    Stable Library
    =========================================
    Native lib Version = RXTX-2.1-7
    Java lib Version   = RXTX-2.1-7
    RXTX Warning:  Removing stale lock file. /var/lock/LCK..ttyACM0
    # A fatal error has been detected by the Java Runtime Environment:
    #
    #  SIGSEGV (0xb) at pc=0x00007f10c0f16462, pid=9874, tid=139709933221632
    #
    # JRE version: OpenJDK Runtime Environment (7.0_79-b14) (build 1.7.0_79-b14)
    # Java VM: OpenJDK 64-Bit Server VM (24.79-b02 mixed mode linux-amd64 compressed oops)
    # Derivative: IcedTea 2.5.5
    # Distribution: Ubuntu 12.04 LTS, package 7u79-2.5.5-0ubuntu0.12.04.1
    # Problematic frame:
    # C  [librxtxSerialx86_64.so+0x6462]  read_byte_array+0x52
    #

jantje, 2015-05-17:
The plugin comes with a librxtx so there is no need to install a additional one. I'm not sure which he will load if there is more than one.
The plugin first tries to load the arduino delivered rxtx dll which fails. That is the error you are seeing.
The rxtx test not only shows success or not but also the dll that has been found. Which dll is found?

braindevices, 2015-05-19:
Eclipse plugin delivered rxtx library (rxtxSerialx86_64) has been loaded successfully
Os =linux Os Architecture =x86_64

braindevices, 2015-05-19:
Can you tell me the path of the plugin delivered lib?

jantje, 2015-05-19:
I'm not sure where eclipse puts the rxtx dll.
Note that the nightly build no longer uses rxtx and has lots of improvements.

braindevices, 2015-05-27:
nightly build fixed this, thanks!

jantje, 2015-05-27:
Txs for testing
closing the issue

closed 2015-05-27 by jantje