Sloeber/arduino-eclipse-plugin #258: random crash after upload complete

braindevices, 2015-05-19:
This is a very helpful plugin. However, it often crashed after upload. Here is the error message:

    eclipse.buildId=M20130204-1200
    java.version=1.7.0_79
    java.vendor=Oracle Corporation
    BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_US
    Framework arguments:  -product org.eclipse.epp.package.cpp.product
    Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.cpp.product

    Error
    Mon May 18 04:53:17 PDT 2015
    Conflicting handlers for it.baeyens.arduino.actions.ArduinoBuildAction: {it.baeyens.arduino.actions.BuildHandler} vs {it.baeyens.arduino.actions.BuildHandler}

jantje, 2015-05-19:
you are probably working with the latest stable version. 
The nightly build has several fixes in this area.

jantje, 2015-05-26:
can you provide feedback?

braindevices, 2015-05-26:
the nightly build requires Luna. However, the Luna keep crashing in my system due to some GTK bugs. So I haven't get a chance to try that yet. I probably get back to you next week.

closed 2015-06-21 by braindevices

jantje, 2015-06-21:
txs for the input