Sloeber/arduino-eclipse-plugin #260: uint8_t is not recognised by Eclipse

riban-bw, 2015-05-23:
When I create a new workspace, any arduino projects I create do not recognise uint8_t. I can't figure out where the path to stdint.h is defined.

riban-bw, 2015-05-23:
When I create a new Arduino project from the default workspace, include paths are added to the project for arduino-1.6.1/hardware/tools/avr/lib/gcc/avr/include (and others). These paths are not added to project created in other workspaces. I wonder whether there is some configuration that is not copied in to newly created workspaces that may have been set in the workspace that was current when the plugin was first installed. I observe this behaviour on Ubuntu 15.04 and Windows 7, both using the plugin 2.4.0.201505170214.

riban-bw, 2015-05-24:
Project settings (in non-working workspace) is missing Arduino Compiler Settings under "C/C++ General->Preprocessor Include Paths, Macros etc.->GNU C++".
Is there a problem copying the Arduino Compiler Settings when creating new workspaces?

jantje, 2015-05-24:
is it a .ino project? have you added #include "Arduino.h"?

riban-bw, 2015-05-24:
It is a cpp project with #include "Arduino.h" in the header file. It looks (to me) like the built-in include paths for Arduino Compiler Settings are not configured for any new projects created in other the workspaces, hence the includes are not found.

riban-bw, 2015-05-24:
I've seen something in eclipse problems tab that may help.
```Program "-E" not found in PATH Preferences, C++/Build/Settings/Discovery, [Arduino Compiler Settings] options	C/C++ Scanner Discovery Problem```

It seems ```${COMMAND} -E -P -v -dD ${INPUTS}``` isn't seeing any ${COMMAND}. I don't know where that command is passed in.

jantje, 2015-05-24:
If I'm not mistaken the command is formed in getCompilerCommand in the class ArduinoLanguageProvider in package it.baeyens.arduino.toolchain

riban-bw, 2015-05-25:
getCompilerCommand may be a red herring.
I note that the language.settings.xml file for the project differs by missing ```<language id="org.eclipse.cdt.core.gcc">``` from the faulty project. This is where the missing configuration is stored. I haven't figured out where in the code that is added.

jantje, 2015-05-25:
I really don't know what is going on. The functionality create/swap workspace is a basic eclipse functionality and there is no code from the arduino eclipse plugin involved.
Have you configured the new workspaces? http://eclipse.baeyens.it/how_to.shtml#/e
Have you created the sketch as described here? http://eclipse.baeyens.it/how_to.shtml#/f
Can you provide the content of the console after a build?

riban-bw, 2015-05-26:
Workspaces are configured as described. I don't know how useful a post-build console output will be - the problem manifests earlier than that with no errors in output unless using the files in the missing includes. I just tried uninstalling and reinstalling plugin when in a non-working workspace and this did not fix it so i am quite confused how it works with one (and only one) workspace. Is anyone else able to replicate:
1. Switch workspace.
2. Create new Arduino project called "test".
3. Project Explorer Includes only shows test/arduino/core and test/arduino/variant.
>Missing the 3 include paths [arduino install]/hardware/tools/avr/avr/include,  [arduino install]/hardware/tools/avr/lib/gcc/avr/4.8.1/include,  [arduino install]/hardware/tools/avr/lib/gcc/avr/4.8.1/include-fixed.

riban-bw, 2015-05-26:
Arrgg! I think I have figured it! It seems to fail when the path to the workspace has spaces in it. On my Ubuntu machine I started using workspaces stored in a directory called, "Eclipse Workspaces" which does not work - so I know how to work around this problem. I haven't seen anything written down saying this is not acceptable - I wonder whether something needs to be quoted or escaped? I will check on Windows later where it may prove more likely to happen due to users home directories generally having spaces in the path.

jantje, 2015-05-26:
spaces is documented at several locations
http://eclipse.baeyens.it/installAdvice.shtml
http://eclipse.baeyens.it/faq.shtml#/troubleshooting
https://github.com/jantje/arduino-eclipse-plugin/issues/207

Txs for the fix. There have been several space fixes, lets hope this is the last one needed :-)

commit 2ccde7f9 (2015-05-26): Merge pull request #265 from riban-bw/master

Fixes #260 - handle spaces in workspace when language adds includes.
commit ec84cc0d (2015-05-26): Fixes #260 - handle spaces in workspace when language adds includes.
closed 2015-05-26 by jantje