Sloeber/arduino-eclipse-plugin #263: Windows error '-DUSB_PRODUCT=Arduino Due': No such file or directory

real-artswan, 2015-05-25:
Got error when build
arm-none-eabi-g++: error: '-DUSB_PRODUCT=Arduino Due': No such file or directory
To workaround either fix the boards.txt file by changing

real-artswan, 2015-05-26:
forgot to say: to make workaround working platform.txt must be fixed too. Need to remove quotes around define parameters so line
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} -DUSBCON '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'
must be replaced with
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} -DUSBCON -DUSB_MANUFACTURER={build.usb_manufacturer} -DUSB_PRODUCT={build.usb_product}

Issue confirmed with nightly 2015-05-26 and arduino 1.6.1

jantje, 2015-05-30:
on my system the command states
'-DUSB_PRODUCT="Arduino Due"'

this is in my platform.txt
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} -DUSBCON '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'

I checked my boards.txt 
build.usb_product="Arduino Due"
So the quotes go missing when converting platform.txt to environment variables on windows.
Most likely the problem is in https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/tools/ArduinoBoards.java
when triggered from the setTheEnvironmentVariablesAddtheBoardsTxt in
https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/tools/ArduinoHelpers.java

Is there a volunteer with a windows machine to fix this?

jantje, 2015-05-30:
looking a bit closer the place to start investigating is https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/tools/ArduinoHelpers.java#L679
currentPair.getValue() may return a value without double quotes
or MakeEnvironmentString removes the quotes
or something after that

jantje, 2015-09-16:
More investigation shows this is a problem with a different command line interpreter in the arduino IDE and the eclipse plugin.
I know a way to fix this and will fix it when the framework will be added.

closed 2015-09-29 by jantje