Sloeber/arduino-eclipse-plugin #270: MEGA2560 needs --relax option when linking

FredericG-BE, 2015-06-03:
See https://code.google.com/p/arduino/issues/detail?id=729
http://www.atmel.com/webdoc/AVRLibcReferenceManual/FAQ_1faq_funcptr_gt128kib.html

When compiling with Arduino IDE:
C:\ProgramFiles\Arduino\hardware\tools\avr\bin\avr-gcc -Os -Wl,--gc-sections,--relax -mmcu=atmega2560 -o ....

When compiling with Eclipse:
"C:/ProgramFiles/Arduino/hardware/tools/avr/bin/avr-gcc" -Os -Wl,--gc-sections -mmcu=atmega2560 -o ...

I changed the line in platform.txt to
compiler.c.elf.flags=-Os -Wl,--gc-sections,--relax    
and all seems fine...

jantje, 2015-06-04:
Wierd. The word "relax" is nowhere found in the platform nor the boards.txt file in arduino ide 1.6.1.
It looks like a hardcoded thing.
Note that your solution is only ok if you actually use a mega. I assume other boards will make bigger elf files with this option.
I guess it is the right time to think about adding an option to extend the linker and archiver command line.
Next to opening an issue at arduino's

FredericG-BE, 2015-06-04:
Indeed, the option that I added to platform.txt is just to keep me going and to demonstrate that it indeed fixes the issue I was facing; removing it and it comes back. I also suspect that this option comes at a price.

jantje, 2015-06-27:
I had an issue with a mega and I wanted to be sure it was not related to this --relax option so I thought about a good workaround.
What I did was add these lines to the boards.txt containing the mega.
mega.compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--gc-sections,--relax
megaADK.compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--gc-sections,--relax
These lines are ignored by the arduino IDE so it is a safe workaround.

jantje, 2015-09-16:
I build a additional framework in the plugin that allows me to add 
mega.compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--gc-sections,--relax
megaADK.compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--gc-sections,--relax
in the plugin.
I tested it on my system and it works fine
I'm not sure we will make the framework part of the release @rlogiacco is working on.

closed 2015-09-29 by jantje