Sloeber/arduino-eclipse-plugin #271: Class redefinitions

lordkev, 2015-06-03:
I just starting using the plugin and am having trouble getting my sketch to compile when including libraries. I'm receiving the following error:

```
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR     -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/eightanaloginputs" -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI" -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire" -I"/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/Wire/utility" -I"/Users/kevinlord/Documents/Arduino/libraries/Adafruit-GFX-Library" -I"/Users/kevinlord/Documents/Arduino/libraries/Adafruit_SSD1306" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall   -DSSD1306_SEGMENTS=4
In file included from ../LostModel_RX.ino:4:0,
                 from ../.ino.cpp:18:
/Users/kevinlord/Documents/Arduino/libraries/Adafruit_SSD1306/Adafruit_SSD1306.h:160:7: error: redefinition of 'class Adafruit_SSD1306'
 class Adafruit_SSD1306 : public Adafruit_GFX {
       ^
In file included from ../.ino.cpp:10:0:
/Users/kevinlord/Documents/Arduino/libraries/Adafruit_SSD1306/Adafruit_SSD1306.h:160:7: error: previous definition of 'class Adafruit_SSD1306'
 class Adafruit_SSD1306 : public Adafruit_GFX {
       ^
```

I have these includes at the top of my sketch (LostModel_RX.ino):

```
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
```

When looking at the generated cpp file, it seems that it is both including the above includes as well as my `.ino` sketch. Wouldn't this double include cause the issues I'm seeing? Did I set something up wrong to cause the file to be generated in this way?

```
//This is a automatic generated file
//Please do not modify this file
//If you touch this file your change will be overwritten during the next build
//This file has been generated on 2015-06-03 15:22:30

#include "Arduino.h"
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
void setup() ;
void loop() ;
void standardMode() ;
void displayLatLon() ;
void displayChars(char *chars, uint16_t textSize, uint16_t color) ;

#include "LostModel_RX.ino"
```

jantje, 2015-06-04:
It looks like Adafruit_SSD1306.h does not use "include guards" http://en.wikipedia.org/wiki/Include_guard
The plugin requires all header files to use "include guards" when using the ino file format.
I advice to add the include guards yourself and tell adafruit.

closed 2015-06-04 by jantje

lordkev, 2015-06-04:
Great, thanks for your help. Sorry, I'm coming from an Objective-C background where import takes care of it for us. Appreciate the assistance.

jantje, 2015-06-04:
No problem.
Please tell adafruit about the issue.

lordkev, 2015-06-04:
Will do! I'll submit a PR.

jantje, 2018-03-05:
Note that though this is still a issue sloeber (by default) adds `#pragma once` to library header files which makes this issue less annoying.

jantje, 2018-04-04:
Workaround add:
`#pragma once`
to the header files having this issue

raistlinmolina, 2018-07-29:
Hi, I have also found this problem.
In my case is with the MPU6050 library, it has the include guards and I have tried with the #pragma once with no luck.

I copy the first error (every method from the class appears with the same problem).

./sloeber.ino.cpp.o: In function `MPU6050::dmpInitialize()':
/home/xxx/Arduino/libraries/MPU6050/MPU6050_6Axis_MotionApps20.h:329: multiple definition of `MPU6050::dmpInitialize()'
./MPU6050Helper.cpp.o:/home/xxx/Arduino/libraries/MPU6050/MPU6050_6Axis_MotionApps20.h:329: first defined here

Any idea?
As I have told the #pragma once did not work for me (I even included it in every file, including the MPU6050_6Axis_MotionApps20.h, that as I also told has include guards).

It seems that the IDE generates both object files with those definitions and then it fails.

This MPU605 library is the most popular one:
https://github.com/jrowberg/i2cdevlib/blob/master/Arduino/MPU6050/MPU6050_6Axis_MotionApps20.h

jantje, 2018-07-29:
@raistlinmolina 
You are in the linking stage which has nothing to do with include guards.

raistlinmolina, 2018-07-29:
Forget it, moved the project to the arduino IDE, put the MPU6050Helper class in the libraries folder of Arduino and got the same result, so definitively is not the IDE.

Thanks for taking the time anyway.

jantje, 2018-07-29:
Great you solved the issue yourself :-)