Sloeber/arduino-eclipse-plugin #272: Add support for boardsManager

jantje, 2015-06-12:
Arduino IDE 1.6.2 came with a boards manager. This came with changes in how/where boards are stored.
Therefore the plugin needed changes and data from the arduino IDE that was not available in a file.
This resulted in issue #236 and in the arduino repository https://github.com/arduino/Arduino/issues/2982
The arduino core team changed the Arduino IDE so the info is now available.
This will be part of the arduino IDE release 1.6.5
This defect is to try to fix the whole problem before Arduino IDE 1.6.5 is released so I can declare arduino IDE 1.6.5 and the latest build of the plugin "compattible"

jantje, 2015-06-15:
first issue was reported by @ranm #273

jipp, 2015-06-18:
tried the latest plugin using the boards manager support on OSX. I get this error message:

------------
Failed to extract environment info from arduino ide 'arduino --get-pref'. The setup will not work properly
"/Applications/Arduino_1.6.5.app/Contents/Java/arduino" --board arduino:avr:uno --get-pref
Cannot run program "/Applications/Arduino_1.6.5.app/Contents/Java/arduino" (in directory "/Applications/Arduino_1.6.5.app/Contents/Java"): error=2, No such file or directory
------------

Manually I can start the command and get the right output:
/Applications/Arduino_1.6.5.app/Contents/MacOS/Arduino --board arduino:avr:uno --get-pref

I assume that the path for the Arduino executable on OSX is not correct. 
The manpage (https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) for the CLI states: 
"Note that on MacOS X, the main executable is Arduino.app/Contents/MacOS/JavaApplicationStub instead of arduino."

jantje, 2015-06-18:
@jipp 
I hope this fix fixes it

lakidd, 2015-06-28:
I've got the same issue as above (jipp). I'm running the nightly build AFAIK (2.4.0.201506260213) according to eclipse, which is slightly different from the latest nightly on your website.
I've tried to set the preferences to use the Arduino 1.6.5 and I get the error above.
Going into the project properties doesn't throw any errors but when you try to compile non of the arduino specific build instructions appear.
 Trying from a new workspace results in the same problem, but when I create my first .ino sketch I get the same error.
This is on OSX 10.10.3

jantje, 2015-06-28:
@lakidd 
Are you getting exactly the same as this? 
>Failed to extract environment info from arduino ide 'arduino --get-pref'. The setup will not work properly
"/Applications/Arduino_1.6.5.app/Contents/Java/arduino" --board arduino:avr:uno --get-pref
Cannot run program "/Applications/Arduino_1.6.5.app/Contents/Java/arduino" (in directory "/Applications/Arduino_1.6.5.app/Contents/Java"): error=2, No such file or directory

please dump the error here.

lakidd, 2015-06-28:
Thanks @jantje 
Here you go
eclipse.buildId=unknown
java.version=1.8.0_25
java.vendor=Oracle Corporation
BootLoader constants: OS=macosx, ARCH=x86_64, WS=cocoa, NL=en_US
Framework arguments:  -perspective it.baeyens.arduino.application.perspective -keyring /Users/lakidd/.eclipse_keyring -showlocation
Command-line arguments:  -os macosx -ws cocoa -arch x86_64 -perspective it.baeyens.arduino.application.perspective -keyring /Users/lakidd/.eclipse_keyring -showlocation

it.baeyens.core
Error
Sun Jun 28 18:49:48 EST 2015
Failed to extract environment info from arduino ide 'arduino --get-pref'. The setup will not work properly
"/Applications/Arduino.app/Contents/Java/arduino" --board arduino:avr:pro --get-pref

java.io.IOException: Cannot run program "/Applications/Arduino.app/Contents/Java/arduino" (in directory "/Applications/Arduino.app/Contents/Java"): error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at it.baeyens.arduino.tools.ExternalCommandLauncher.launch(ExternalCommandLauncher.java:280)
	at it.baeyens.arduino.ide.connector.ArduinoGetPreferences.generateDumpFileForBoard(ArduinoGetPreferences.java:115)
	at it.baeyens.arduino.ide.connector.ArduinoGetPreferences.generateDumpFiles(ArduinoGetPreferences.java:94)
	at it.baeyens.arduino.ide.connector.ArduinoGetPreferences.updateArduinoEnvironmentVariablesForAllProjectsIfNeeded(ArduinoGetPreferences.java:154)
	at it.baeyens.arduino.ui.ArduinoPreferencePage.performOk(ArduinoPreferencePage.java:200)
	at org.eclipse.jface.preference.PreferencePage.performApply(PreferencePage.java:438)
	at it.baeyens.arduino.ui.ArduinoPreferencePage.performApply(ArduinoPreferencePage.java:339)
	at org.eclipse.jface.preference.PreferencePage$2.widgetSelected(PreferencePage.java:284)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:248)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4199)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1467)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1490)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1475)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1279)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4042)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3669)
	at org.eclipse.jface.window.Window.runEventLoop(Window.java:832)
	at org.eclipse.jface.window.Window.open(Window.java:808)
	at org.eclipse.ui.internal.dialogs.WorkbenchPreferenceDialog.open(WorkbenchPreferenceDialog.java:221)
	at org.eclipse.ui.internal.handlers.ShowPreferencePageHandler.execute(ShowPreferencePageHandler.java:59)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:294)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:90)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:483)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:247)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:229)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:132)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:149)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:499)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:508)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:210)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:196)
	at org.eclipse.e4.ui.workbench.renderers.swt.cocoa.CocoaUIHandler.runCommand(CocoaUIHandler.java:715)
	at org.eclipse.e4.ui.workbench.renderers.swt.cocoa.CocoaUIHandler.access$6(CocoaUIHandler.java:703)
	at org.eclipse.e4.ui.workbench.renderers.swt.cocoa.CocoaUIHandler$8.widgetSelected(CocoaUIHandler.java:568)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:248)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4199)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1467)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1490)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1475)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1279)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4042)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3669)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$9.run(PartRenderingEngine.java:1151)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:332)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1032)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:148)
	at org.eclipse.ui.internal.Workbench$5.run(Workbench.java:636)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:332)
[... cut ...]
	at java.lang.reflect.Method.invoke(Method.java:483)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:648)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:603)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1465)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:185)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 69 more

jantje, 2015-06-28:
Seems I missed to connect a dot.
Please test the nightly build tomorrow.

lakidd, 2015-06-28:
No worries, will do...thanks.

jantje, 2015-06-30:
@lakidd 
How did it work?

jipp, 2015-06-30:
Hi

Idid some tests using the latest version: 2.4.0.201506290213

-started the arduino edi 1.6.5 and selected the UNO
-closed it

-now starting eclipse
-selecting the Arduino 1.6.5 as IE path.

I get the following error message:

Failed to extract environment info from arduino ide 'arduino --get-pref'. The setup will not work properly
"/Applications/Arduino_1.6.5.app/Contents/MacOS/JavaApplicationStub"  --board arduino:avr:uno --get-pref
Cannot run program "/Applications/Arduino_1.6.5.app/Contents/MacOS/JavaApplicationStub" (in directory "/Applications/Arduino_1.6.5.app/Contents/Java"): error=2, No such file or directory

the correct path should be "/Applications/Arduino_1.6.5.app/Contents/MacOS"
in the error message its show as: "Applications/Arduino_1.6.5.app/Contents/Java"
used is show: "/Applications/Arduino_1.6.5.app/Contents/MacOS"

program should be: "Arduino"
used: "JavaApplicationStub"

cheers
Jipp

jantje, 2015-06-30:
Hi jipp above you stated:
>I assume that the path for the Arduino executable on OSX is not correct.
The manpage (https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) for the CLI states:
"Note that on MacOS X, the main executable is Arduino.app/Contents/MacOS/JavaApplicationStub instead of arduino."

Is there such a program as JavaApplicationStub?

jantje, 2015-06-30:
I think I mixed up some things ....
The program is started in the arduino folder based on what you selected as arduino folder in the user preferences. As I start a full qualified program that should be OK.
But indeed JavaApplicationStub should be Arduino

It is amazing how mutch harder it becomes when you do not have the os at your fingertips.

lakidd, 2015-07-01:
So even with that change, it seems to be turning the path into /Application/Arduino.app/Contents/Java/MacOS/arduino (which breaks it w/ the same error.)
The problem is that /Java/
I am not a C programmer by any stretch of the imagination...but it looks like 
IPath ArduinoIDEPath = ArduinoInstancePreferences.getArduinoPath();
needs to have the last /Java/ removed (/Java/ and /MacOS/ are at the same level in the dir structure

jantje, 2015-07-01:
good. Seems we got MacOS finally going :-)

lakidd, 2015-07-01:
no worries and thanks again...so without being a non-contributing
ungrateful user :-)...how hard would it be to get the other boards (e.g.
esp8266) that the Arduino IDE now supports working in this environment ?
Love the project BTW, it's taught me so much about a decent dev environment.

jantje, 2015-07-01:
>.how hard would it be to get the other boards (e.g. esp8266) that the Arduino IDE now supports working in this environment ?

All boards that do not use the arduino:arduino reference should work now. This is related to #158.
Mostly there is a easy way around, but not always.

As I said before I tried it again now the boardsmanager is available but it is and remains a lot of work.

jantje, 2015-07-07:
Note I found a way to support build.core and build.variant. For upload there are easy way arounds.
Should be available from tomorrow onwards

lakidd, 2015-07-08:
thanks Jantje. So...if I have the esp8266 plugin installed in the arduino IDE, how do I see that as a board option in the eclipseArduino IDE ? Or are we not there yet ?

jantje, 2015-07-08:
@lakidd 
Yesterday I noticed that the boards installed by the boardsmanager are not visible in in the combobox.
As you can see they are on my linux system.
[image]
This is a very similar change as we did before for finding the arduino executable :-)
But This time I took care of having it at one location so maybe you could try to change and test this yourself? (and create a pull request or share the outcome)
The code that needs to change is in the package /it.baeyens.arduino.common/src/it/baeyens/arduino/common/ArduinoInstancePreferences.java

>    public static IPath getArduinoBoardsManagerPackagesPath() {
	IPath homPath = new Path(System.getProperty("user.home"));
	return homPath.append(".arduino15").append("packages");
    }

I just made a change based on https://github.com/arduino/Arduino/wiki/Boards-Manager-FAQ and it looks like
>    public static IPath getArduinoBoardsManagerPackagesPath() {
	switch (Platform.getOS()) {
	case Platform.OS_MACOSX: 
	    return new Path(System.getProperty("user.home")).append("Library").append("Arduino15").append("packages");
	case Platform.OS_WIN32:
	    return  new Path(System.getenv("APPDATA")).append("Arduino15").append("packages");
	default:
	    return new Path(System.getProperty("user.home")).append(".arduino15").append("packages");
	}
    }

please report progress

lakidd, 2015-07-09:
sure, will do tonight...looks good tho !

lakidd, 2015-07-09:
ok, some good some bad...
1. I can compile a completely empty *.ino sketch as in 
void setup {}
void loop {}
against a esp8266 and it compiles. I haven't actually uploaded but i assume that works..so thats a great start.
The (minor) bad..
So...with the same code, reverting back back to an arduino board...sometimes I get the splash screen sometime not. I've tried Uno, nano and a few other boards in the list and they all seem to work (i.e. i can compile the empty program above)...where it gets strange is that the main board i use is a pro-mini and it doesn't work...it doesn't build the command line properly. I haven't tested every arduino variant but the only busted one appears to be the pro-mini which is odd

jantje, 2015-07-09:
The build succeeded last night so an upgrade should give the same results.
I just build the pro mini here on my system and all seems good.
>Building file: ../ChargerBrains.cpp
Starting C++ compile
"/home/jan/programs/Arduino/arduino-1.6.5/hardware/tools/avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR     -I"/home/jan/programs/Arduino/arduino-1.6.5/hardware/arduino/avr/variants/eightanaloginputs" -I"/home/jan/programs/Arduino/arduino-1.6.5/libraries/Servo" -I"/home/jan/workspaces/arduino/Robot/libraries/VoltMeter" -I"/home/jan/programs/Arduino/arduino-1.6.5/libraries/Servo/src" -I"/home/jan/workspaces/arduino/Robot/libraries/I2CLiquidCrystal" -I"/home/jan/programs/Arduino/arduino-1.6.5/hardware/arduino/avr/libraries/Wire" -I"/home/jan/workspaces/arduino/Robot/libraries/SerialDataInterface" -I"/home/jan/workspaces/arduino/Robot/libraries/JanServo" -I"/home/jan/workspaces/arduino/Robot/libraries/SerialCommunicator" -I"/home/jan/workspaces/arduino/Robot/libraries/SerialStringReader" -I"/home/jan/workspaces/arduino/Robot/libraries/gps_Library" -I"/home/jan/workspaces/arduino/Robot/libraries/DataTypes" -I"/home/jan/workspaces/arduino/Robot/libraries/AltSoftSerial" -I"/home/jan/programs/Arduino/arduino-1.6.5/hardware/arduino/avr/cores/arduino" -I"/home/jan/programs/Arduino/arduino-1.6.5/hardware/arduino/avr/cores/arduino" -I"/home/jan/programs/Arduino/arduino-1.6.5/hardware/arduino/avr/libraries/SoftwareSerial" -I"/home/jan/workspaces/arduino/Robot/libraries/CurrentSensor" -I"/home/jan/workspaces/arduino/Robot/libraries/SerialBridgeCommunicator" -I"/home/jan/workspaces/arduino/Robot/libraries/FastRunningMedian" -I"/home/jan/programs/Arduino/arduino-1.6.5/hardware/arduino/avr/libraries/Wire/utility" -MMD -MP -MF"ChargerBrains.cpp.d" -MT"ChargerBrains.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../ChargerBrains.cpp"  -o  "ChargerBrains.cpp.o"   -Wall  -DI_USE_COMMUNICATOR  -DI_USE_DATETIME -DI_USE_processGPRMCMessage -DI_USE_SERIAL_REGISTER -DI_USE_GPS -DI_USE_ALTSOFTSERIAL -DVOLTAGE_MULTIPLE_READ
Finished building: ../ChargerBrains.cpp

It looks to me that your environment variable A.COMPILER.PATH is not set properly.
Have you modified the boards.txt or platform.txt?

Something to keep in the back of my mind .... there may be a problem with the generation of the dump files. It may be needed that they must be project dependent due to the options specified. On the other hand I process the boards.txt file so it should be ok.

lakidd, 2015-07-09:
OK...so I started a new workspace and the pro-mini works fine. It's something to do with that being the board i was using before I upgraded. Is there a way to blow away the config for an existing workspace and start from scratch ? I haven't modified boards.txt or platform.txt

jantje, 2015-07-09:
> Is there a way to blow away the config for an existing workspace and start from scratch ? 

do the upgrade procedure
http://eclipse.baeyens.it/how_to.shtml#/n
to remove the dump file 
change arduino version to before 1.6.5
select apply
change arduino version to 1.6.5
All settings are in the environment environment. So you should be able to see what is wrong there.

Best regards
Jantje

lakidd, 2015-07-09:
Thanks Jante...yes it's seems that a couple of my workspaces are a bit
busted from testing etc. Seems good so far from an esp8266
perspective...thanks very much !

jantje, 2015-07-10:
so we can say mac works fine?

lakidd, 2015-07-10:
It looks like it but need to test with actual code as opposed to empty projects. Great stuff, thanks !

Sent from my phone so please excuse brevity and typos

jantje, 2015-07-27:
I guess we can close this one now

closed 2015-07-27 by jantje

lakidd, 2015-07-28:
Yep, it works really well...great stuff ! Thanks