Sloeber/arduino-eclipse-plugin #273: updated to V2.4 and can't compile

ranm, 2015-06-14:
Hey , 
I updated the plugin to the new nightly to get the fix for #209 but now I can't even compile.

I keep getting a pop-up window saying:
Error parsing C:/Users/Ran/workspace/DUMP____ArduinoIDE.tmp file does not exist. 
C:\Users\Ran\workspace\DUMP____ArduinoIDE.tmp (The system cannot find the file specified)
and as for the console window I get : 
Building target: Blink
Printing size:
"/tools/avr/bin/avr-size" -A "C:/Users/Ran/workspace/Blink/Release/Blink.elf"
/usr/bin/sh: /tools/avr/bin/avr-size: No such file or directory
make: *** [Blink] Error 127

Any ideas ? I understand this is a nightly and not stabe yet but I can't work with my Arduino micro due to the issue I mentioned above.

b3ndo, 2015-06-15:
I have the same problem...
My investigating: 
1. Serial monitor - something wrong...
[image]
2. Preprocesor... - something wrong...
[image]

To ranm: My solution:
download eclipse C++, use Arduino eclipse plugin v.2.2, not v.2.4

ranm, 2015-06-15:
:-) but then I'm stuck with #209 ....

jantje, 2015-06-15:
Seems you guys (non-volunteer) took part of my test request 
http://blog.baeyens.it/#post21
First question did you do the upgrade procedure http://eclipse.baeyens.it/how_to.shtml#/n
both preferences and project must be done with the current nightly

ranm, 2015-06-15:
I updated from the eclipse , didn't re-install the Arduino IDE.
Didn't check the project settings - will do it in the evening.

jantje, 2015-06-15:
@ranm 
Do you use a hourly arduino IDE?

@b3ndo 
The serial monitor issue is a completely different issue and should have it's own issue.
Reverting to V2.2 is really bad advise. 
First of all there is no 2.4 the latest official release is 2.3 from July 2014. Since then lots of things have changed in the arduino IDE and have been adopted/fixed in the Arduino eclipse plugin.  (compares to ARDUINO 1.5.6-r2 BETA - 2014.02.21 and 1.0.5)
2.2 is from January 2014. that compares to Arduino IDE 1.0.5-r2 and 1.5.5-r2 BETA.

ranm, 2015-06-15:
No , using 1.5.6 or 1.5.7 , I don't remember ... 
I can check at the evening

jantje, 2015-06-15:
no problem.
Thanks for being the first bug reported on the support for the boards manager.
Seems there is an issue with "older arduino versions" Fixing it now. Will be available for download as of tomorrow.

ranm, 2015-06-15:
Will it work if I update my arduino IDE ? If so , which version should I Install?

jantje, 2015-06-15:
Removed the fixed in nightly label as that is only tomorrow :-)
@ranm 
Can you test the nightly tomorrow? If so and it works fine please close the issue.

ranm, 2015-06-15:
Ok, I can do it. When you say tomorrow I guess it's US timezone?

jantje, 2015-06-15:
The build process is mostly finished around 04:00 GMT. 
I'm not a US citizen. I'm from Belgium the country of chocolate, Brussels sprouts, French fries and ..... **the Arduino eclipse plugin** :-)

b3ndo, 2015-06-15:
OK, here is my process:
PC ===WinXP SP3, Arduino IDE 1.5.5-r2 in c:\ArduinoPF 
1. unpack  eclipse-cpp-luna-SR2-win32.zip  to c:\eclipse1
2. set workspace to C:\Arduino1
3. update Eclipse via Help -> Check for updates
4. instal plugin via Help -> install new software  http://www.baeyens.it/eclipse/nightly
   and restart eclipse
5. again update Eclipse via Help -> Check for updates
6. set plugin: path to ArduinoIDE: C:\ArduinoPF,
               path to ArdPersLib: D:\Documents and Settings\Branko\My Documents\Arduino\libraries
               pat to Private...  :C:\ArduinoPF\hardware
               File type *.ino to C++ etc...
7. click New -> Project ----- First problem?
[image]
8. When click build
[image]
And when I clik "Apply" in Window -Preference-Arduino I get this:
[image]

jantje, 2015-06-15:
If you install from the nightly step 5 should not be necessary.
I cannot reproduce step 7 here. It is probably related to language settings. It is harmless.
Why do you not just press the new sketch button in the toolbar?

Step 8 is a consequence of the dialog box that stopped all processing.
In short current nightly version is only usefull if you use the nightly arduino IDE.
Tomorrow should be better.

b3ndo, 2015-06-15:
Step 7: No matter where you click...result is the same :)
And one more question: what version ArduinoIDE only recommended?

jantje, 2015-06-15:
I'm sorry I don' understand what you are trying to say and ask.

b3ndo, 2015-06-16:
Plugin updated, still no working for me... :)
Parsing error mising, but..when I create new project, I get this
[image]
with this error:
[image]
PS:which version of ArduinoIDE (1.5.5, 1.5.6, ...,1.6.0, 1.6.1,1.6.2,1.6.3) you recommend?
sry, google translate...:)

jantje, 2015-06-16:
When you select your arduino version in the preferences it tells you each time I advice 1.6.1.
I can make leonardo projects without problems.
You probably found an issue related to windows.
Can you dump the error log in this issue?

b3ndo, 2015-06-16:
maybe help this?
[image]
errorlog: http://www.bendik.name/errorlog.log

jantje, 2015-06-16:
please delete the project also from disk before trying to recreate the project
from errorlog.log
>org.eclipse.core.internal.resources.ResourceException: Resource '/popo' already exists.

b3ndo, 2015-06-16:
org.eclipse.core.internal.resources.ResourceException: Resource '/popo' already exists.
This happened because the window is created the project remained open after a mistake, and I clicked again to finish.
Here is errorlog from begining, instal eclipse, to end, build project
http://www.bendik.name/errorlog1.log
see last line of errorlog...

ranm, 2015-06-16:
Updated to recent nightly - the warning dialog from before disappeared but I still  can't compile ... 
using IDE 1.5.6.-r2 , didn't update (as you asked). 
I'm getting this on the console window now:
Starting C++ compile
"/tools/avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=156-r2 -DARDUINO_AVR_UNO -DARDUINO_ARCH_avr    -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\variants\standard" -I"C:\Arduino\IDE\Arduino\libraries\Servo" -I"C:\Arduino\IDE\Arduino\libraries\Servo\src" -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\libraries\Wire" -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\libraries\Wire\utility" -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\libraries\EEPROM" -I"C:\Arduino\IDE\Arduino\libraries\RF24" -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\libraries\SPI" -I"C:\Users\Ran\workspace\CableCam\CommonSw" -I"C:\Arduino\IDE\Arduino\libraries\FlexiTimer2" -I"C:\Arduino\IDE\Arduino\libraries\ArduinoMemoryFree" -MMD -MP -MF"CableCamControl.cpp.d" -MT"CableCamControl.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../CableCamControl.cpp"  -o  "CableCamControl.cpp.o"   -w
/usr/bin/sh: /tools/avr/bin/avr-g++: No such file or directory
make: *** [CableCamControl.cpp.o] Error 127

Something to do with the path settings ?

ranm, 2015-06-16:
I think the plugin is looking for the compiler file at  /tools/avr/bin/avr-g++ , but in my case (maybe because of the version of the IDE) it should be looking at:
hardware\tools\avr\bin

jantje, 2015-06-16:
@b3ndo 
It looks like your plugin is not properly configured
>	at org.eclipse.core.filesystem.URIUtil.toPath(URIUtil.java:60)
	at it.baeyens.arduino.tools.ArduinoLibraries.findAllUserLibraries(ArduinoLibraries.java:70)

at line 70 you find
```C++
	URI PrivateLibraryURI = pathMan.resolveURI(pathMan.getURIValue(ArduinoConst.WORKSPACE_PATH_VARIABLE_NAME_PRIVATE_LIB));
	return findAllSubFolders(URIUtil.toPath(PrivateLibraryURI));```

>..\arduino-eclipse-plugin-master\it.baeyens.arduino.core\src\it\baeyens\arduino\tools\ArduinoHelpers.java in line 1042
what is: HardwareFolder2 ???

related to the boards manager Arduino puts hardware info at 2 locations now.

jantje, 2015-06-16:
@ranm 
Have you executed the update process?

ranm, 2015-06-16:
I believe so:
1. updated through "check for updates"
2. verified arduino IDE path is correct 
3. opened project , settings o.k. , apply . (even tries creating a new sketch - didn't help) 

Did I miss anything?

jantje, 2015-06-16:
2: select apply?

ranm, 2015-06-16:
Yes , got the warning about 1.6.1 because I have an older version , chose yes to ignore and continue.

jantje, 2015-06-16:
ranm
I found the issue
Compiler path is defined as 
>${A.HARDWARE.PATH}/tools/avr/bin/

but A.HARDWARE.PATH is not defined

I can reproduce the problem here. I think it is related to the oldness of the arduino ide.
I'm trying to fix it now.

I'm also confused when I look at your output. It looks like Linux but part of your output looks like windows 
> -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Arduino\IDE\Arduino\hardware\arduino\avr\variants\standard" -I"C:\Arduino\IDE\Arduino\libraries\Servo" [...]

ranm, 2015-06-16:
I also saw that part and I tried to add the env variable myself , but for some reason it wouldn't let me add it to the environment - tried 3 times without luck.... 
Would it work with IDE 1.6.5 ?

jantje, 2015-06-16:
I think I fixed it. I'm doing a test right now

jantje, 2015-06-16:
Fixed it.
Good catch from you :-)
As you can see below 1.5.6 builds like a charm with the fix (will be in the nightly tomorrow)

>make all 
Building file: ../.ino.cpp
Starting C++ compile
"/home/jan/programs/Arduino/arduino-1.5.6-r2/hardware/tools/avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=156-r2 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR    -I"/home/jan/programs/Arduino/arduino-1.5.6-r2/hardware/arduino/avr/cores/arduino" -I"/home/jan/programs/Arduino/arduino-1.5.6-r2/hardware/arduino/avr/variants/standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
Finished building: ../.ino.cpp

ranm, 2015-06-17:
looks o.k. now :-)
Closing issue .
Thanks for the help !

closed 2015-06-17 by ranm