Sloeber/arduino-eclipse-plugin #274: Linux x64 Nightly: Unable to find full path for -D__IN_ECLIPSE__=1

DerTomm, 2015-06-16:
Hi,

I just downloaded and installed the latest Linux x64 Nightly [http://eclipse.baeyens.it/download/product/linux64.2015-06-16_02-06-03.tar.gz].

When I try to create the first new Arduino project in a fresh workspace I get the following error dialog:
[image]

At the same time the Eclipse error log shows:
Unable to find full path for "-D__IN_ECLIPSE__=1"

Any ideas what this could cause?

Additional information:
- OS: Ubuntu 14.04.02 x64
- JDK: 1.7.0_67
- Arduino IDE: v1.6.1

jantje, 2015-06-16:
It is probably related to #272 and the project is not properly configured.
I fixed some more issues today in regards to pre 1.6.5 arduino IDE's
Can you debug?

DerTomm, 2015-06-16:
No problem. Sorry for the question: What do you mean with "Can you debug" in this context?

jantje, 2015-06-16:
>What do you mean with "Can you debug" in this context?
whether you can download the code of the plugin and debug the plugin to see where the bug is coming from.
It may be fixed with the things I changed today, it may be not.

DerTomm, 2015-06-16:
Hmm - I'm afraid debugging the whole plugin is one level too high for me. But I will try again with the next nightly of course.

jantje, 2015-06-16:
You're never to old to learn ;-)
sorry for the extra work.

b3ndo, 2015-06-17:
WinXP: I got this message too
jantje, can You check these two things?
1.
[image]
On the LEFT side: Eclipse LUNA + Arduino IDE1.6.1 + plugin nightly build  --- NOT working(missing folder it.baeyens.arduino.core.nl1_2.2.0.1 )
On the RIGHT side: Eclipse LUNA + Arduino IDE1.5.5 + plugin V.2 ----- working good
2.
 In your repo in file: ..\arduino-eclipse-plugin-master\it.baeyens.arduino.core\src\it\baeyens\arduino\tools\ArduinoHelpers.java in line 1045, you compose this folder: C:\Documents and Settings\Branko\.arduino15\packages\arduino\hardware, but in my WinXP I find only one folder with name arduino15, "c:\Documents and Settings\Branko\Application Data\Arduino15" with preferces.txt

reworks, 2015-06-17:
I get a similar error as well: 
Error: Cannot run program "-D__IN_ECLIPSE__=1"
Here is the stack trace:
```
java.io.IOException: Cannot run program "-D__IN_ECLIPSE__=1": Unknown reason
	at org.eclipse.cdt.utils.spawner.Spawner.exec(Spawner.java:349)
	at org.eclipse.cdt.utils.spawner.Spawner.<init>(Spawner.java:91)
	at org.eclipse.cdt.utils.spawner.ProcessFactory.exec(ProcessFactory.java:91)
	at org.eclipse.cdt.core.CommandLauncher.execute(CommandLauncher.java:194)
	at org.eclipse.cdt.internal.core.BuildRunnerHelper.build(BuildRunnerHelper.java:263)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runProgramForLanguage(AbstractBuiltinSpecsDetector.java:759)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForLanguage(AbstractBuiltinSpecsDetector.java:689)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector.runForEachLanguage(AbstractBuiltinSpecsDetector.java:562)
	at org.eclipse.cdt.managedbuilder.language.settings.providers.AbstractBuiltinSpecsDetector$1.runInWorkspace(AbstractBuiltinSpecsDetector.java:496)
	at org.eclipse.core.internal.resources.InternalWorkspaceJob.run(InternalWorkspaceJob.java:38)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)

```

DerTomm, 2015-06-17:
>You're never to old to learn ;-)

Of course not :-)
If it is not too time intensive: Could you maybe give me a rude hint how to debug the whole plugin? Then I'd like to try it. And it's good to know for future analyses...

jantje, 2015-06-17:
@DerTomm 
read the readme.md here https://github.com/jantje/arduino-eclipse-plugin

jantje, 2015-06-17:
@DerTomm 
Do you still have the issue with the build from last night?

DerTomm, 2015-06-17:
Obviously in latest nightly the problem is solved - but project creation can still not be completed. This time I get the following error message:

"AVR: Unknown architecture"

Failed to extract environment info from arduino ide 'arduino --get-pref'. The setup will not work properly.
"/opt/arduino-1.6.5/arduino" --board arduino:AVR:nano --get-pref

Shall I close this issue and create a new one for the error above?

DerTomm, 2015-06-17:
I guess mixing things up is not a good idea. So the latest test with IDE 1.6.5 gives the error above. With 1.6.1 there is still the "Unable to find full path for "-D__IN_ECLIPSE__=1"" error.

jantje, 2015-06-17:
Seems the uppercase fix broke 1.6.5 pffffffff
But 1.6.1 works fine with me.

jantje, 2015-06-17:
I didn't even know You can close a issue with a pull request.
If it isn't fixed tomorrow please reopen.

jantje, 2015-06-18:
just tested with the latest nightly. I changed my workspace from 1.6.5 to 1.5.6.r2 to 1.6.1 to 1.6.5
in each version I opened 3 projects properties selected apply and compiled.
All went well.

DerTomm, 2015-06-18:
I'm afraid I can't confirm the fix. With linux64.2015-06-18 nightly the problem is still existing - with both IDE 1.6.1 and 1.6.5. So I'll try to learn how to debug the plugin execution...

jantje, 2015-06-18:
ok txs

DerTomm, 2015-06-18:
That's really weird.

Thanks to your excellent guide how to checkout, import and run the plugin within a default eclipse environment I managed to run/debug the latest build from sources. And guess what? Everything works fine - no errors at all.

But as soon as I download the latest nightly, extract and run the pre-built EclipseArduino I cannot complete project creation due to the reported error.
To make sure that there are no hidden external problems I flushed the workspace and the $USER_HOME/Arduino folder before starting.

Do you have any idea which differences there are between "run from sources" and "run nightly build"?

jantje, 2015-06-18:
Great progress :-)

>Do you have any idea which differences there are between "run from sources" and "run nightly build"?

None

Did you use the same workspace in both cases?

DerTomm, 2015-06-18:
>Did you use the same workspace in both cases?

Yes. It's the same.

So I just cloned the sources from GIT locally and build the ArduinoEclipse IDE using maven now. The build also works fine.
When I try to create a new project in the same workspace using the nightly build the error occurs again.
So obviously there is a difference in the builds...

jantje, 2015-06-18:
that is really weird
the build is done by a build machine extracting all the data from git just like you did

b3ndo, 2015-06-18:
bro, try this settings>
[image]
FIRST line: Arduino IDE ver1.6.5
SECOND line: path to private lib
THIRD line: path to hardware folder in Arduino IDE 1.5.5-r2
and here>
[image]
board.txt point to Arduino v1.5.5-r2
for me its working....

jantje, 2015-06-19:
@b3ndo 
This is a usage of the plugin I hadn't thought of yet. Not sure what the pro's and cons are. I need to think about that.

jantje, 2015-06-19:
@DerTomm 
I had a similar wierd issue on my system so maybe this can help you out as well.
It is related to the fix for #153 
I project properties->resources->linked resources->path variables I had removed the path variable
ArduinoPivateLibPath
When I did Ardduino->reattach libraries I got a unhandled event loop exception and when pressing OK in the arduino project properties it gives a dialog box.
Simply adding the path variable ArduinoPivateLibPath back fixes the issue.
I guess this is a version problem (old versions lying around somewhere in eclipse, or bad version numbering in the plugin)
Can you confirm this is what you see?

DerTomm, 2015-06-20:
I just tried it with the 2015-06-20 nightly.

After clicking "Finish" in the project creation wizard there was the "null argument" error again. Then I removed the ArduinoPivateLibPath as described from the broken project.
The Arduino->Reattach command got executed without exceptions but when I tried to add the path under Project Properties again I got an "Internal Error" exception. Project is still corrupt afterwards.

jantje, 2015-06-27:
@DerTomm 
Does it work now?
Can this issue be closed?

DerTomm, 2015-06-27:
Unfortunately at the moment I cannot test the Linux build as I have only a Windows machine available. The Windows nightly works fine and the last self-built Linux release (tested last Tuesday) did also work properly. So I'd say you can close this issue.

jantje, 2015-06-27:
cool :-)

closed 2015-06-27 by jantje

SodaqMoja, 2015-06-28:
The instructions in readme.md talk about the plugin fragments to import. The items are listed in an image
 `images_plugin_dev_setup/plugins_select.png`.
I think that the png  must be updated. Right now it only has three items to import, but I'm fairly sure that you need to import `it.baeyens.arduino.application` too (and maybe more). Without it you don't have the Arduino perspective.

jantje, 2015-06-28:
@SodaqMoja 
If you have an issue with the readme.md please make a new issue.
Best regards
Jantje

commit 90231f5a (2015-06-17): fix for #272 should fix #274