Sloeber/arduino-eclipse-plugin #275: Project in workspace as library

marciodel, 2015-06-17:
It would be nice to be able to use a project in the same Workspace as library. Imagine you have several related projects that share some source code. These libraries would be only visible to the projects in the same workspace.

jantje, 2015-06-17:
Sorry I don't see what the benefit/difference is between what is possible now and what you propose.
Have you read http://eclipse.baeyens.it/how_to.shtml#/j
Combime this with the possibility to point your private library path to anywhere you want.

jantje, 2015-06-27:
@marciodel 
Can you please provide feedback on my response?

marciodel, 2015-06-30:
Hi, Jantje, sorry for the late answer. My idea was to have an "All projects" library path and another "All projects in the workspace" library path. But if the private library configuration is saved in the workspace configuration, your idea is very close to my needs.

jantje, 2015-06-30:
>But if the private library configuration is saved in the workspace configuration, your idea is very close to my needs. 

This is how I have been working for a long time now. This because I want all libraries I use under version control.

jantje, 2015-07-27:
due to acceptable alternative I'm closing his change request

closed 2015-07-27 by jantje