Sloeber/arduino-eclipse-plugin #278: Serial Monitor doesn't work on nightly

ranm, 2015-06-20:
Hey , 
Using arduino IDE 1.5.6-r2 , latest nightly . 
Serial monitor view is not usable ? 
Attached screenshot:
[image]

jantje, 2015-06-20:
I have this problem in  one of my workspaces as well.
When I tried a fresh install all works well though.
Can you try with a fresh install?

ranm, 2015-06-21:
I'm trying to uninstall the plugin but the uninstall option is greyed out . 
It's only greyed out for the plugin , for other plugins I have - it's o.k. 
It's a complete mess now - I tried updating but then the setup crashes when eclipse restarts - it says : 
"Failed to create the path variable variables. The setup will not work properly
Path variable value must be valid and absolute."
so now I'm totally stuck.... 
It's actually un-usable now how do I uninstall ?

ranm, 2015-06-21:
Actually I can still compile and work but I keep getting the error warning about the path. 
This is the exception trace :
org.eclipse.core.runtime.CoreException: Path variable value must be valid and absolute.
	at org.eclipse.core.internal.resources.PathVariableManager.checkIsValidValue(PathVariableManager.java:78)
	at org.eclipse.core.internal.resources.PathVariableManager.setValue(PathVariableManager.java:224)
	at org.eclipse.core.internal.resources.PathVariableManager.setURIValue(PathVariableManager.java:322)
	at org.eclipse.core.internal.resources.ProjectPathVariableManager.setURIValue(ProjectPathVariableManager.java:364)
	at it.baeyens.arduino.tools.ArduinoHelpers.setProjectPathVariables(ArduinoHelpers.java:373)
	at it.baeyens.arduino.listeners.ConfigurationChangeListener.handleEvent(ConfigurationChangeListener.java:35)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.notifyListeners(CProjectDescriptionManager.java:2212)
	at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.fireAboutToApplyEvent(AbstractCProjectDescriptionStorage.java:284)
	at org.eclipse.cdt.internal.core.settings.model.SetCProjectDescriptionOperation.executeOperation(SetCProjectDescriptionOperation.java:66)
	at org.eclipse.cdt.internal.core.model.CModelOperation.execute(CModelOperation.java:341)
	at org.eclipse.cdt.internal.core.model.CModelOperation.run(CModelOperation.java:607)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
	at org.eclipse.cdt.internal.core.model.CModelOperation.runOperation(CModelOperation.java:638)
	at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.setProjectDescription(AbstractCProjectDescriptionStorage.java:203)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionStorageManager.setProjectDescription(CProjectDescriptionStorageManager.java:149)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.setProjectDescription(CProjectDescriptionManager.java:885)
	at org.eclipse.cdt.managedbuilder.internal.core.BuildSettingsUtil.checkApplyDescription(BuildSettingsUtil.java:211)
	at org.eclipse.cdt.managedbuilder.core.ManagedBuildManager.updateCoreSettings(ManagedBuildManager.java:1574)
	at org.eclipse.cdt.build.internal.core.scannerconfig.CfgDiscoveredPathManager$1.run(CfgDiscoveredPathManager.java:154)
	at org.eclipse.cdt.internal.core.model.BatchOperation.executeOperation(BatchOperation.java:36)
	at org.eclipse.cdt.internal.core.model.CModelOperation.execute(CModelOperation.java:341)
	at org.eclipse.cdt.internal.core.model.CModelOperation.run(CModelOperation.java:607)
	at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
	at org.eclipse.cdt.core.model.CoreModel.run(CoreModel.java:1220)
	at org.eclipse.cdt.core.model.CoreModel.run(CoreModel.java:1181)
	at org.eclipse.cdt.build.internal.core.scannerconfig.CfgDiscoveredPathManager.updateCoreSettings(CfgDiscoveredPathManager.java:157)
	at org.eclipse.cdt.build.core.scannerconfig.ScannerConfigBuilder.build(ScannerConfigBuilder.java:125)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:734)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:206)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:246)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:299)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:302)
	at org.eclipse.core.internal.events.BuildManager.basicBuildLoop(BuildManager.java:358)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:381)
	at org.eclipse.core.internal.resources.Workspace.buildInternal(Workspace.java:496)
	at org.eclipse.core.internal.resources.Workspace.build(Workspace.java:405)
	at org.eclipse.ui.actions.GlobalBuildAction$1.run(GlobalBuildAction.java:180)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)

Serial monitor is still not usable . and I still can't un-install.

jantje, 2015-06-21:
If you started from the product it is normal you can not uninstall the plugin as it is an integral part of it.
I just upgraded to the latest version and all seems well on my system.
Can you provide a dump of the project properties->resources->linked resources->path variables

ranm, 2015-06-21:
As far as I can remember - I didn't start from the product , I installed the plugin on a given eclipse system . 
 Dump of path variables : 
[image]

jantje, 2015-06-21:
Based on what I see here;
or you are not on the latest version or you have not applied the upgrade actions.
Besides: uninstalling and reinstalling didn't fix the serial monitor issue. I really needed a fresh install.

ranm, 2015-06-22:
ok - I'll be happy to do that - I wanted to do a fresh install , but I can't uninstall the plugin because the un-install action in greyed out. 
And about the upgrade - I think the update process stopped in the middle due to the path issue I mentioned before . 
If I check fr updates now - It doesn't find any.
Is it possibe to uninstall the plugin manually?

ranm, 2015-06-23:
So path issue from #279 is fixed , but serial monitor is still unusable. 
How can I reinstall the plugin if the un-install option is greyed out ?

jantje, 2015-06-23:
Download the arduino eclipse project and unpack somewhere else on your system?

ranm, 2015-06-23:
Yea I know I can download the product again , but the thing is I have other SDKs installed on the one I already have + settings and stuff generated in over a year of work, so that's kind of a turn off . 
Is that really the only option ?

jantje, 2015-06-23:
I'm in the same situation. Fortunately fro me I'm using the serial monitor less and less.
If I find a solution I'll fix it but first I'll have to know what goes wrong. 
The only error message I found untill now is about listLineEndings that doesn't exist while is does.
https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.common/src/it/baeyens/arduino/common/Common.java#L465
It is really wierd as it seems very simular to the pivateLibPath problem (using pivateLibPath while the code only has privateLibPath)
I just can't get my finger on it.

jantje, 2015-06-23:
@wimjongman 
Hi Wim sorry for calling you but I think you can help me out here.

I found the root cause of 1 (and I think 2) weird bugs I have been fighting with the plugin over the last days.
The serial monitor no longer works in the latest builds. However in local builds it works fine. I have seen an error link that points to https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.common/src/it/baeyens/arduino/common/Common.java#L465 as if listLineEndings does not exists. Remote debugging confirmed this.
After some search in the history I found that on june 7th I changed
public static Object listLineEndings() 
to
public static String[] listLineEndings() 
This change did not require a code change in the monitor.java file.
I checked the content of the file it.baeyens.arduino.monitor_2.4.0.201506230213.jar on my system and I see the date of the serial monitor class is from 26 feb. Way before the change.
I assume that (due to name mangling) there is no need for a code change but there is a need to build the code. As the code has not been rebuild the JVM looks for the wrong listLineEndings and fails.

I have 2 questions here.
Can you confirm my thinking?
What is the standard solution here?

The other problem is the rename of a "public static final String" in the common plugin. Here the old name still seems to be used in plugins that use the common plugin. I assume that here the String has been copied for performance reasons and as such a rebuild should be triggered but is not.

Your input is appreciated

Best regards
Jantje

jantje, 2015-06-27:
Seems the wipe and rebuild fixed the issue (at least on my system)
Please reopen if the issue still exists on your system after upgrade to nightly

closed 2015-06-27 by jantje

wimjongman, 2015-06-27:
Hi,

I'm sorry I missed this. Please ping me privately if you need my help. I am not following all threads.

Cheers?

Wim

wimjongman, 2015-06-27:
Yes wiping is occasionally the fix. In fact, consider wiping the workspace before each build.

ranm, 2015-06-28:
updated and it works , thanks!