Sloeber/arduino-eclipse-plugin #279: nightly build install error: failed to create the path variable variables

braindevices, 2015-06-21:
eclipse.buildId=4.4.2.M20150204-1700
java.version=1.7.0_79
java.vendor=Oracle Corporation
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.cpp.product

it.baeyens.core
Error
Sun Jun 21 14:21:07 PDT 2015
Failed to create the path variable variables. The setup will not work properly

org.eclipse.core.runtime.CoreException: Path variable value must be valid and absolute.
    at org.eclipse.core.internal.resources.PathVariableManager.checkIsValidValue(PathVariableManager.java:78)
    at org.eclipse.core.internal.resources.PathVariableManager.setValue(PathVariableManager.java:224)
    at org.eclipse.core.internal.resources.PathVariableManager.setURIValue(PathVariableManager.java:322)
    at org.eclipse.core.internal.resources.ProjectPathVariableManager.setURIValue(ProjectPathVariableManager.java:364)
    at it.baeyens.arduino.tools.ArduinoHelpers.setProjectPathVariables(ArduinoHelpers.java:373)
    at it.baeyens.arduino.listeners.ConfigurationChangeListener.handleEvent(ConfigurationChangeListener.java:35)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.notifyListeners(CProjectDescriptionManager.java:2212)
    at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.fireAboutToApplyEvent(AbstractCProjectDescriptionStorage.java:284)
    at org.eclipse.cdt.internal.core.settings.model.SetCProjectDescriptionOperation.executeOperation(SetCProjectDescriptionOperation.java:66)
    at org.eclipse.cdt.internal.core.model.CModelOperation.execute(CModelOperation.java:341)
    at org.eclipse.cdt.internal.core.model.CModelOperation.run(CModelOperation.java:607)
    at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
    at org.eclipse.cdt.internal.core.model.CModelOperation.runOperation(CModelOperation.java:638)
    at org.eclipse.cdt.internal.core.settings.model.AbstractCProjectDescriptionStorage.setProjectDescription(AbstractCProjectDescriptionStorage.java:203)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionStorageManager.setProjectDescription(CProjectDescriptionStorageManager.java:149)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.setProjectDescription(CProjectDescriptionManager.java:885)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.setProjectDescription(CProjectDescriptionManager.java:846)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager$6.run(CProjectDescriptionManager.java:573)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager$5.run(CProjectDescriptionManager.java:523)
    at org.eclipse.core.internal.resources.Workspace.run(Workspace.java:2313)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.runAtomic(CProjectDescriptionManager.java:519)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.access$2(CProjectDescriptionManager.java:517)
    at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager$4.run(CProjectDescriptionManager.java:498)
    at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)

jantje, 2015-06-21:
this seems to be the same issue as currently discussed in #278 
Can you provide a dump of the project properties->resources->linked resources->path variables

braindevices, 2015-06-22:
[image]

Some misspelled variables in it.

jantje, 2015-06-22:
The misspelled is the root cause of the current problems.
It used to be ArduinoPivateLibPath and I changed it to ArduinoPrivateLibPath for #153 
Apparently this didn't work properly so I made another fix #274 that creates 
ArduinoPivateLibPath=${ArduinoPrivateLibPath}. It is this fix that pops up in the logs.

On your screendump I see ArduinoPivateLibPath has a different value from ${ArduinoPrivateLibPath. 

As you can see here https://github.com/jantje/arduino-eclipse-plugin/blob/master/it.baeyens.arduino.core/src/it/baeyens/arduino/tools/ArduinoHelpers.java#L373
this line is trying to add ArduinoPivateLibPath=${ArduinoPrivateLibPath}
Then the catch directly after that is activated.
The configuration is not stopped; so if ArduinoPivateLibPath is properly configured (as is in the screendump above) the project is properly configured (if nothing else went wrong)

There are some possible root causes:
1) you did not upgrade to the latest version of the plugin (unlikely given the log)
2) you did not do the upgrade process properly (note that if it fails a second try may fix it) (Part one is done as ArduinoPrivateLibPath exists and part 2 fails=> not the root cause)
3) Something goes wrong during the upgrade process on your os (as it works on mine)
Things that may go wrong:
1)The assignment as I use is not correct for your os.
2)your os does not support this redirection.
....

What you could do to pinpoint the issue:
1)Validate my assumption "the project is properly configured" by checking whether the project works fine. 
2) Use the screen to change ArduinoPivateLibPath to ArduinoPivateLibPath=${ArduinoPrivateLibPath}.
This should rule out OS syntax problems.
3)Debug and try to find out more what is going on.
For debugging I would add a validateValue statement as described here http://www.cs.utep.edu/cheon/download/jml4c/javadocs/org/eclipse/core/resources/IPathVariableManager.html#validateValue%28org.eclipse.core.runtime.IPath%29
And hope it makes sense.

Otherwise I'm out of ideas.

Jantje

@ranm for issue #278 this comment is equally valid.

ranm, 2015-06-22:
I'm not near my PC at the moment but I can already tell you that #1 is correct - the project is configured correctly because I was able to compile and upload sketches with the current issue i the background , I just keep getting the pop up window saying there is an error with the path . 
I will try #2 in the evening . 
about the OS - I'm using WIN8 as you can see , went over the upgrade process a number of times already ( arduino settings , apply , project settings , apply) . So for now I don't feel there was a "user flow error :-)) , should be something else.

braindevices, 2015-06-22:
@jantje "On your screendump I see ArduinoPivateLibPath has a different value from ${ArduinoPrivateLibPath". They seem exactly the same to me.

jantje, 2015-06-22:
not sure which screenshot you are referring to.
This is a screenshot just taken
[image]

braindevices, 2015-06-22:
@jantje 
"2) Use the screen to change ArduinoPivateLibPath to ArduinoPivateLibPath=${ArduinoPrivateLibPath}.
This should rule out OS syntax problems."

I used "edit variable" dialog to change the ArduinoPivateLibPath to "${ArduinoPrivateLibPath}", it can resolve the absolute path correctly. However, if I click on "apply" It give out an error: "the path must be absolute". So I guess Juno does not allow you to set path variables to some string contain environment variables.

"3)Debug and try to find out more what is going on."

I will try this later. To me it seems toURI may not convert "${ArduinoPrivateLibPath}" to absolute path, which result this error.

jantje, 2015-06-22:
are you still on juno? That is old :-(
@ranm are you also on juno?

braindevices, 2015-06-22:
@jantje Sorry I type it wrong. I am on Luna.

ranm, 2015-06-22:
same here - it won't let me apply the new value . 
it says "Path variable value must be valid and absolute"
and - like @braindevices  - working with Luna.

jantje, 2015-06-22:
I'm on luna to so I guess it is a os related issue and not a eclipse version issue.
I'll change the code so it doesn't create  ArduinoPivateLibPath=${ArduinoPrivateLibPath}
but uses the actual value of ArduinoPrivateLibPath

braindevices, 2015-06-22:
@jantje Thanks!

jantje, 2015-06-22:
For people who had one of the versions on their system (a linux or a mac?) which did not generate this issue:
If your pivateLibPath diverges from PrivateLibPath simply delete the pivateLibPath in the project properties.
When you reopen it should reappear and have the proper value.
The reason for this is that you can create path variables at the level of the project and the level of the workspace. The project overrules the workspace. The old solution worked at the project level. The new at the workspace level.

ranm, 2015-06-23:
will try to update to latest nightly in the evening and update - Thanks!

ranm, 2015-06-23:
Path issue is fixed with the new nightly

jantje, 2015-06-23:
Txs for the feedback
So we can close this one :-)

closed 2015-06-23 by jantje