Sloeber/arduino-eclipse-plugin #282: The instructions for setting up Eclipse to develop and debug the plugin need to be updated

SodaqMoja, 2015-06-29:
The instructions to setup the development environment in readme.md talk about the plugin fragments to import. The items are listed in an image `images_plugin_dev_setup/plugins_select.png`.
I think that the png must be updated. Right now it only has three items to import, but I'm fairly sure that you need to import `it.baeyens.arduino.application` too (and maybe more). Without it you don't have the Arduino perspective.

jantje, 2015-06-29:
yes. Indeed. you should select all found plugins.
You are aware that -as described in the top of the readme- if you have maven and git installed that the 3 following commands are all you need?
>git clone https://github.com/jantje/arduino-eclipse-plugin
cd arduino-eclipse-plugin
mvn verify

after that you can open a eclipse for java (add pde) and use the workspace arduino-eclipse-plugin?

jantje, 2015-07-27:
changed the description

closed 2015-07-27 by jantje