Sloeber/arduino-eclipse-plugin #283: Missing environment variable - A.RUNTIME.TOOLS.AVR-GCC.PATH

nkolban, 2015-07-03:
When running the nightly against Eclipse Mars ... I find that the variable A.RUNTIME.TOOLS.AVR-GCC.PATH is not set in the project properties->C/C++ Build->Environment settings.  This causes commands not to be found.  The resulting error is:

    10:39:55 **** Incremental Build of configuration Release for project TestWS2812 ****
    make all 
    'Building file: ../TestWS2812.cpp'
    'Starting C++ compile'
    "/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=165-r2 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"C:\Arduino\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\hardware\arduino\avr\cores\arduino" -I"C:\Arduino\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\hardware\arduino\avr\variants\standard" -I"C:\Users\IBM_ADMIN\Arduino\libraries\Adafruit_NeoPixel" -MMD -MP -MF"TestWS2812.cpp.d" -MT"TestWS2812.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../TestWS2812.cpp"  -o  "TestWS2812.cpp.o"
    The system cannot find the path specified.
    make: *** [TestWS2812.cpp.o] Error 1
    
    10:39:56 Build Finished (took 636ms)

In the current release build, I am able to manually enter this setting but in the nightlys, it seems to be "deleted" each time I try and compile.

nkolban, 2015-07-03:
As I debugged further, I have a suspicion that the arduino-eclipse-plugin is parsing <ArduinoInstall>/lib/version and expecting it to be "numeric.numeric.numeric".  However, this is not always true.  For example, the current Arduino release is "1.6.5-r2" and I believe that there is an attempt to turn "5-r2" into a numeric which fails.

jantje, 2015-07-03:
you say you use the nigtly. When have you last updated?
There is indeed a problem with the Arduino version number but that is unrelated to A.RUNTIME.TOOLS.AVR-GCC.PATH

nkolban, 2015-07-03:
I loaded the very latest of the nightly builds this morning.

When I edited the Arduino/lib/version and changed it from:

1.6.5-r2

to

1.6.5

The problem went away.  So I think the issue of variables is masking the underlying problem of version number parsing.  I suspect that in the code an exception is thrown during parsing of the environment which then fails to set the rest of the variables following that error.

jantje, 2015-07-03:
indeed the 1.6.5-r2 gives problems causing subsequent problems due to an exception.
I'll have a look what I can do.

jantje, 2015-07-03:
This should be fixed with the version build tonight.
Can you check and close if it works fine?

nkolban, 2015-07-03:
You betcha my friend.  Will download tomorrow and reset the version file back to 1.6.5-r2 and test again.

nkolban, 2015-07-05:
Unfortunately, I didn't get time to test and it may be a while.  However, I eyeballed your code fix and it all looks great to my eyes so will fully trust it has worked.

closed 2015-07-05 by nkolban

jantje, 2015-07-06:
I hope fortune comes back to you :-)
Txs for closing