Sloeber/arduino-eclipse-plugin #286: set di istruzioni nn risucisto

jkwatt, 2015-07-13:
Description	Resource	Path	Location	Type
set di istruzioni per l'obiettivo "arduino/WInterrupts.c.o" non riuscito	subdir.mk	/cancello_linux/Release/arduino/core	line 175	C/C++ Problem

jkwatt, 2015-07-13:
can anyone help me?

jantje, 2015-07-13:
Is this related to the plugin?
If so can you tell me why you think so?

jkwatt, 2015-07-13:
Yes is relatore the plug in, i tra to compile bit i have the issue shown onde step over this comment

jantje, 2015-07-13:
Can you provide the console output?

jkwatt, 2015-07-14:
ok 

10:52:20 **** Incremental Build of configuration Release for project cancello_linux ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"/home/tecnico/Scaricati/arduino-1.6.0/hardware/tools/avr/bin/avr-g++" -c -g -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"/home/tecnico/Scaricati/arduino-1.6.0/hardware/arduino/avr/cores/arduino" -I"/home/tecnico/Scaricati/arduino-1.6.0/hardware/arduino/avr/variants/standard" -I"/home/tecnico/Documenti/Arduino/libraries/RCSwitch" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
In file included from ../.ino.cpp:13:0:
../cancello_linux.ino:36:14: warning: overflow in implicit constant conversion [-Woverflow]
 int keycode =5592332;
              ^
../cancello_linux.ino: In function 'void loop()':
../cancello_linux.ino:160:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if ( muovi && (millis() - T) < deltaT ){
                                  ^
../cancello_linux.ino: In function 'boolean segnale()':
../cancello_linux.ino:316:65: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                             if ( mySwitch.getReceivedValue() == keycode ) return true;
                                                                 ^
Finished building: ../.ino.cpp
 
Building file: /home/tecnico/Scaricati/arduino-1.6.0/hardware/arduino/avr/cores/arduino/WInterrupts.c
Starting C compile
"/home/tecnico/Scaricati/arduino-1.6.0/hardware/tools/avr/bin/avr-gcc" -c -g -Os -ffunction-sections -fdata-sections -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10600 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR     -I"/home/tecnico/Scaricati/arduino-1.6.0/hardware/arduino/avr/cores/arduino" -I"/home/tecnico/Scaricati/arduino-1.6.0/hardware/arduino/avr/variants/standard" -I"/home/tecnico/Documenti/Arduino/libraries/RCSwitch" -MMD -MP -MF"arduino/WInterrupts.c.d" -MT"arduino/WInterrupts.c.o" -D__IN_ECLIPSE__=1-x c++ "/home/tecnico/Scaricati/arduino-1.6.0/hardware/arduino/avr/cores/arduino/WInterrupts.c"  -o  "arduino/WInterrupts.c.o"   -Wall
avr-gcc: error: c++: No such file or directory
arduino/core/subdir.mk:175: set di istruzioni per l'obiettivo "arduino/WInterrupts.c.o" non riuscito
make: *** [arduino/WInterrupts.c.o] Errore 1

10:52:21 Build Finished (took 413ms)

jantje, 2015-07-14:
something is really wrong in your setup.
The plugin is compiling a C source file "WInterrupts.c" partly as a c file (using avr-gcc) and partly as a c++ file "-x c++" but due to a missing space the -x is part of "-D__IN_ECLIPSE__=1" which makes that the gcc compiler is trying to compile the file "c++" which obviously does not exist.

things I think of to do to fix this
1) check your project properties->arduino whether -x c++ is part of the extra compiler options.
2) open your project properties->arduino and select apply

jkwatt, 2015-07-14:
I'm sorry I can not figure out which items have to change in the properties of the project, what you mean by "arduino whether".
excuse me but i'm a beginner with arduino+eclipse system

closed 2015-07-17 by jantje

jkwatt, 2015-07-24:
i solved the issue is only a .ino and .pde  setting problem like you see thanks for the help!!!

jkwatt, 2015-07-24:
now i like to debug  and test the program can you give me a liitle kick to a true direction?

jantje, 2015-07-24:
Debugging is advanced stuff. I need to do a video on zero debugging. But I don't get enough incentives. Here are some blog posts and a issues on local debugging.

http://iloapp.baeyens.it/blog/blog?Home&post=19
http://iloapp.baeyens.it/blog/blog?Home&post=20
http://blog.baeyens.it/#post23

https://github.com/jantje/hardware/issues/2
https://github.com/jantje/hardware/issues/1

jkwatt, 2015-07-24:
but it work done or not?

jantje, 2015-07-24:
it works